intel/intel-device-plugins-for-kubernetes #739: Unexpected Kubernetes node feature-labeling as sgx-capable when SGX feature is turned off in BIOS and host has no /dev/sgx_* devices

MustDie95, 2021-11-02:
At first I was mentioned this issue in NFD project ( https://github.com/kubernetes-sigs/node-feature-discovery/issues/638) but it looks like NFD itself is not responsible for this behavior.

What happened:

Kubernetes node has feature.node.kubernetes.io/custom-intel.sgx: 'true' even if SGX support was forcibly turned off in BIOS

What you expected to happen:

Kubernetes node is not feature-labeled with SGX or this label was removed when SGX was off in BIOS and there is no /dev/sgx_* devices on host (even if this feature supported by CPU and OS).

How to reproduce it (as minimally and precisely as possible):

Install server, setup Kubernetes and deploy Node Feature Discovery with 'kubectl apply -k https://github.com/kubernetes-sigs/node-feature-discovery/deployment/overlays/default?ref=v0.22.0'

All nodes are marked with feature.node.kubernetes.io/custom-intel.sgx: true' despite BIOS settings.

And Intel sgxdeviceplugin-sample constantly trying to start (because has nodeSelector based on this feature) on this node but failing.

Anything else we need to know?:

nfd-worker.conf  (from https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/deployments/sgx_nfd/nfd-worker.conf):

sources:
  custom:
  - name: "intel.sgx"
    matchOn:
      - kConfig: ["X86_SGX"]
        cpuId: ["SGX", "SGXLC"]

I think we should add more rules to make sure Intel SGX is really functional on the host. For example, examine the presence of /dev/sgx_* devices

But I cant find how to detect the presence of /dev/sgx_* devices with attributes/methods available in the Doc (https://github.com/kubernetes-sigs/node-feature-discovery/blob/master/docs/get-started/features.md#custom).

Environment:

Kubernetes version: 1.22.2
Cloud provider or hardware configuration: Intel M50CYP1UR212 server (2 x Intel Xeon Platinum 8352S @ 2.2GHz, 512GB RAM)
OS: Ubuntu Server 20.04.3 LTS
Kernel: 5.11.0-38-generic

mythi, 2021-11-02:
@MustDie95 thanks for the comprehensive report

> What you expected to happen:
>
> Kubernetes node is not feature-labeled with SGX or this label was removed when SGX was off in BIOS
> and there is no /dev/sgx_* devices on host (even if this feature supported by CPU and OS).

I'm not yet convinced this is the right goal. Applications are not expected to use that label alone but request `sgx.intel.com/epc` memory resources instead. Currently, getting the node resources registered (EPC size, by `epc-hook` and the device nodes by sgx-device-plugin) require the pod to run.

> And Intel sgxdeviceplugin-sample constantly trying to start (because has nodeSelector based on this feature) on
> this node but failing.

Can you provide to logs what is failing? The sgxdeviceplugin-sample has `HostPath` Volumes set as the pre-condition for the `/dev/sgx_` device nodes so the pod should not be able to run. See https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/pkg/controllers/sgx/controller.go#L176-L193

MustDie95, 2021-11-02:
@mythi, Pod with sgxdeviceplugin-sample trying to start repeatly on this node but failing with error "MountVolume.SetUp failed for volume "sgx-provision" : hostPath type check failed: /dev/sgx_provision is not a character device" (and this is true - there is not such device on this host). There are 2 pods from sgxdeviceplugin-sample DaemonSet and 2nd pod on SGX-BIOS-enabled host is running ok.

The question is "why this failed pod is schedulled to run on this node" and answer is "because of nodeSelector with feature.node.kubernetes.io/custom-intel.sgx: 'true'".

mythi, 2021-11-02:
> The question is "why this failed pod is schedulled to run on this node" and answer is "because of 
> nodeSelector with feature.node.kubernetes.io/custom-intel.sgx: 'true'".

Right now I cannot think of any other way around this than to improve the default NFD `cpuid` labeling for SGX.

mythi, 2021-11-03:
@MustDie95 as a workaround you could add a custom [local source](https://kubernetes-sigs.github.io/node-feature-discovery/v0.9/get-started/features.html#local--user-specific-features) and use that plus `intel-sgx: true` as the nodeSelectors in the `SgxDevicePlugin` CRs.

MustDie95, 2021-11-03:
@mythi, thanks a lot! this works well and the only pod is running on a properly SGX-configured node! :)

mythi, 2021-11-03:
@MustDie95 thanks for checking! I need to check can I follow the idea and just enhance `sgx_epchook` to print another label with `sgx.capable` when the EPC size is non-zero.

mythi, 2021-11-25:
@MustDie95 PR #766 fixes this. It can be easily tested using kustomize:

```
$ kubectl apply -k https://github.com/mythi/intel-device-plugins-for-kubernetes/deployments/sgx_nfd?ref=PR-2021-064
```

MustDie95, 2021-11-26:
@mythi, thanks, it works well, as desired!

host 01 = SGX enabled in BIOS, host 02 = disabled

root@pluo-k8s-sgx-01:/# kubectl get nodes -o json | jq .items[].metadata.labels | grep -i sgx
  "hostname": "pluo-k8s-sgx-01",
  "kubernetes.io/hostname": "pluo-k8s-sgx-01",
  "feature.node.kubernetes.io/custom-intel.sgx": "true"
  "hostname": "pluo-k8s-sgx-02",
  "kubernetes.io/hostname": "pluo-k8s-sgx-02"

And only 1 pod is scheduled and running on properly labeled node:

root@pluo-k8s-sgx-01:/etc# kubectl get pods -n inteldeviceplugins-system -o wide
NAME                                                    READY   STATUS    RESTARTS   AGE   IP              NODE              NOMINATED NODE   READINESS GATES
inteldeviceplugins-controller-manager-5bf97d77b-9kf7g   2/2     Running   0          14m   10.224.187.84   pluo-k8s-sgx-01   <none>           <none>

MustDie95, 2021-11-26:
Ops, sorry, I forgot to edit sgxdeviceplugins.deviceplugin.intel.com with my custom rule, so sgx-deviceplugin-sample was missed..

root@pluo-k8s-sgx-01:/etc# kubectl get pods -n inteldeviceplugins-system -o wide
NAME                                                    READY   STATUS    RESTARTS   AGE   IP              NODE              NOMINATED NODE   READINESS GATES
inteldeviceplugins-controller-manager-5bf97d77b-9kf7g   2/2     Running   0          25m   10.224.187.84   pluo-k8s-sgx-01   <none>           <none>
sgxdeviceplugin-sample-wgkck-s5slz                      1/1     Running   0          97s   10.224.187.80   pluo-k8s-sgx-01   <none>           <none>

mythi, 2021-11-26:
> `sgxdeviceplugin-sample-wgkck-s5slz 1/1 Running 0 97s 10.224.187.80 pluo-k8s-sgx-01`

nice, so only one plugin pod. thanks

PR #766 (2021-11-26): sgx: update NFD deployment
Fixes: #739
closed 2021-11-26 by rojkov