intel/intel-device-plugins-for-kubernetes #762: SCC and SeLinux issue on OpenShif-4.9 to run SGX on intel-device-plugins framework

Walnux, 2021-11-22:
### Comments
I'd like to open his issue to discuss and track the SCC and SeLinux setting issue on OpenShift-4.9 platform to enable intel-device-plugin framework and SGX plugin.
### The issue is:
If I enable SeLinux up as below on my work node
```
sh-4.4# sestatus
SELinux status:                 enabled
SELinuxfs mount:                /sys/fs/selinux
SELinux root directory:         /etc/selinux
Loaded policy name:             targeted
Current mode:                   enforcing
Mode from config file:          enforcing
Policy MLS status:              enabled
Policy deny_unknown status:     allowed
Memory protection checking:     actual (secure)
Max kernel policy version:      33
```
Then My initial container will run into  "permission access denied" issue on all the volume mounted in the pod
if I close the Selinux as below
```
sh-4.4# sestatus
SELinux status:                 enabled
SELinuxfs mount:                /sys/fs/selinux
SELinux root directory:         /etc/selinux
Loaded policy name:             targeted
Current mode:                   permissive
Mode from config file:          enforcing
Policy MLS status:              enabled
Policy deny_unknown status:     allowed
Memory protection checking:     actual (secure)
Max kernel policy version:      33

```
The operator can be up and running properly.
You can reproduce the issue using the below steps

### Reproduce Steps
Firstly I have to apply below patches to setup SCC according to documents:
[SCC in OCP-4.9](https://docs.openshift.com/container-platform/4.9/authentication/managing-security-context-constraints.html )
[Guide to UID, GID](https://cloud.redhat.com/blog/a-guide-to-openshift-and-uids)
```
Author: MartinXu <martin.xu@intel.com>
Date:   Thu Nov 18 22:29:51 2021 -0500

    Add SCC hostaccess to manager-role on OpenShift

    So the default SA (Service Account) can have the privilige to create pod to access to all
    host namespaces but still requires pods to be run with a UID and SELinux context that are
    allocated to the namespace.

    For detail
    see https://docs.openshift.com/container-platform/4.9/authentication/managing-security-context-constraints.html

diff --git a/deployments/operator/rbac/role.yaml b/deployments/operator/rbac/role.yaml
index 8d19b7a..dd93674 100644
--- a/deployments/operator/rbac/role.yaml
+++ b/deployments/operator/rbac/role.yaml
@@ -176,3 +176,11 @@ rules:
   - get
   - list
   - watch
+- apiGroups:
+  - security.openshift.io
+  resources:
+  - securitycontextconstraints
+  resourceNames:
+  -  hostmount-anyuid
+  verbs:
+  - use
```

```
commit 9e3106cef687a7f83ed7daed90575f7e16b16993
Author: Xu <jxu36@jfz1r09h07.otcdcslab.com>
Date:   Thu Nov 18 19:27:25 2021 -0500

    Dropoff securityContext from manager deployment

    OpenShift SCC (Security Context Constraints) is used to manage security
    context. See
    https://cloud.redhat.com/blog/a-guide-to-openshift-and-uids
    https://docs.openshift.com/container-platform/4.9/authentication/managing-security-context-constraints.html

    By default restricted SCC is used to Ensure that pods cannot be run as privileged.
    So this commit drops off the securityconttext to run as non-root user

diff --git a/deployments/operator/default/manager_auth_proxy_patch.yaml b/deployments/operator/default/manager_auth_proxy_patch.yaml
index 8ba668c..082782f 100644
--- a/deployments/operator/default/manager_auth_proxy_patch.yaml
+++ b/deployments/operator/default/manager_auth_proxy_patch.yaml
@@ -19,11 +19,11 @@ spec:
         ports:
         - containerPort: 8443
           name: https
-        securityContext:
-          runAsNonRoot: true
-          runAsUser: 1000
-          runAsGroup: 1000
-          readOnlyRootFilesystem: true
+          #securityContext:
+          #runAsNonRoot: true
+          #runAsUser: 1000
+          #runAsGroup: 1000
+          #readOnlyRootFilesystem: true
       - name: manager
         args:
         - "--metrics-addr=127.0.0.1:8080"
diff --git a/deployments/operator/manager/manager.yaml b/deployments/operator/manager/manager.yaml
index db335d3..9ee0a94 100644
--- a/deployments/operator/manager/manager.yaml
+++ b/deployments/operator/manager/manager.yaml
@@ -33,11 +33,11 @@ spec:
           requests:
             cpu: 100m
             memory: 20Mi
-        securityContext:
-          runAsNonRoot: true
-          runAsUser: 65532
-          runAsGroup: 65532
-          readOnlyRootFilesystem: true
+        #securityContext:
+        #runAsNonRoot: true
+        #runAsUser: 65532
+        #runAsGroup: 65532
+        #readOnlyRootFilesystem: true
         env:
           - name: DEVICEPLUGIN_NAMESPACE
             valueFrom:
```
```
commit fbf8bd8b120ab65fc456d4778fb156214230ffac
Author: MartinXu <martin.xu@intel.com>
Date:   Thu Nov 18 20:45:51 2021 -0500

    Backport https://github.com/intel/intel-device-plugins-for-kubernetes/pull/756

diff --git a/deployments/operator/rbac/role.yaml b/deployments/operator/rbac/role.yaml
index 3e490e5..8d19b7a 100644
--- a/deployments/operator/rbac/role.yaml
+++ b/deployments/operator/rbac/role.yaml
@@ -143,6 +143,12 @@ rules:
   - patch
   - update
   - watch
+- apiGroups:
+  - deviceplugin.intel.com
+  resources:
+  - sgxdeviceplugins/finalizers
+  verbs:
+  - update
 - apiGroups:
   - deviceplugin.intel.com
   resources:

```
Then start the intel device plugins framework using commd
`$ oc apply -k intel-device-plugins-for-kubernetes/deployments/operator/default/
`
and start SGX pluin DS as
`oc apply -f intel-device-plugins-for-kubernetes/deployments/operator/samples/deviceplugin_v1_sgxdeviceplugin.yaml`

The intel device plugins framework can up and running, and the SGX plugin DS also up and running.
But the init container in the pod run into the "permission access denied issue" when try to access directory 
/etc/kubernetes/node-feature-discovery/source.d/

this is the volume mounted in the pod
```
 nodeSelector:
    feature.node.kubernetes.io/custom-intel.sgx: 'true'
    kubernetes.io/arch: amd64
  restartPolicy: Always
  initContainers:
    - name: intel-sgx-initcontainer
      image: 'intel/intel-sgx-initcontainer:0.22.0'
      resources: {}
      volumeMounts:
        - name: nfd-source-hooks
          mountPath: /etc/kubernetes/node-feature-discovery/source.d/
        - name: kube-api-access-nkpq6
          readOnly: true
          mountPath: /var/run/secrets/kubernetes.io/serviceaccount
      terminationMessagePath: /dev/termination-log
      terminationMessagePolicy: File
      imagePullPolicy: IfNotPresent
      securityContext:
        capabilities:
          drop:
            - MKNOD
        readOnlyRootFilesystem: true
```
### Analysis:
You can see that I assigned the SCC as hostmount-anyuid.
And after I disabled the Selinux with command on worknode 1 with command
`$sudo setenforce 0`
Operator up and run on this node.
But I leave Selinux enable on worknode 0 
"The permission access denied issue still there"

After I set the SCC as hostaccess, no matter I disable or enable the SeLinux, The permission access denied issue always happens.

mythi, 2021-11-23:
@Walnux  thanks!

> Then start the intel device plugins framework using commd

we need to focus on the OLM path:

- the UID/GID changes are fine. Do we also need to drop `readOnlyRootFilesystem`?
- I'll test and merge #756 
- The SCC configs must be done in the `ClusterServiceVersion` metadata. Look for `clusterPermissions`.

Walnux, 2021-11-23:
@mythi readOnlyRootFilesystem is very useful to protect the rootfs and we should keep it.

Walnux, 2021-11-23:
And I agree with you @mythi after we can mannually start the operator and figure out the potential issues. We now should work on the bundle image and run the operator in OLM. That will apply another set of priviligy setting.

Walnux, 2021-11-23:
also file a same bug to track this on red hat buggerzilla https://bugzilla.redhat.com/show_bug.cgi?id=2026086

Walnux, 2021-11-29:
according to 
https://www.arhea.net/posts/2020-04-28-selinux-for-containers.html
try to add :z or :Z to Volume mount, but the issue is still there.

Not sure whether we also need to create the operator or plugin image with some SeLinux enabling.

Walnux, 2021-11-30:
According to https://www.redhat.com/sysadmin/user-namespaces-selinux-rootless-containers
The root cause might be:

The container engine, Podman, launches each container with a unique process SELinux label (usually container_t) and labels all of the container content with a single label (usually container_file_t). We have rules that state that container_t can read and write all content labeled container_file_t. This simple idea has blocked major file system exploits.

Everything works perfectly until the user attempts a volume mount. The problem with volumes is that they usually only bind mounts on the host. They bring in the labels from the host, which the SELinux policy does not allow the process label to interact with, and the container blows up.

Walnux, 2021-12-03:
https://github.com/intel/intel-device-plugins-for-kubernetes/pull/787 is sent for review

Walnux, 2022-01-07:
according to the feed back from Peter Hunt <pehunt@redhat.com>, 
```
  securityContext:
    seLinuxOptions:
      type: "spc_t"
```
Can be used to allow Pod to access the host filesystem without running pod with the privileged rights.
It can be verified to work properly. Will submit PR to fix the issue.

mythi, 2022-01-07:
@Walnux is `spc_t` the only type label that works or can we go with the one that NFD source dir is labeled?

Walnux, 2022-01-10:
It also works on NFD source dir. I am submitting the PR

mythi, 2022-01-11:
`spc_t` probably works but are there other `*_t` types that'd fit better?

Walnux, 2022-01-12:
Could be.

haircommander, 2022-01-17:
I believe `spc_t` is the right option. container selinux module doesn't really give much granularity (other than special types for containers that are init processes and kata containers).

Also, I am wondering if anyone would mind helping me figure out why adding `:z` helps anything. From my testing, all I can see that it does is append `:z` to the destination path.
<details>
<summary>Here's the pod I used</summary>
<br>

```yaml
apiVersion: v1                                                                                                                                    
kind: Pod                                                                                                                                         
metadata:                                                                                                                                         
  creationTimestamp: "2022-01-17T19:53:15Z"                                                                                                       
  labels:                                                                                                                                         
    app: sleepytereshkova                                                                                                                         
  name: sleepytereshkova                                                                                                                          
spec:                                                                                                                                             
  containers:                                                                                                                                     
  - command:                                                                                                                                      
    - top                                                                                                                                         
    image: docker.io/library/alpine:latest                                                                                                        
    name: sleepytereshkova                                                                                                                        
    volumeMounts:                                                                                                                                 
     - name: nfd-source-hooks                                                                                                                     
       mountPath: '/etc/kubernetes/node-feature-discovery/source.d/:z'                                                                            
  volumes:                                                                                                                                        
    - name: nfd-source-hooks                                                                                                                      
      hostPath:                                                                                                                                   
        path: /etc/kubernetes/node-feature-discovery/source.d/                                                                                    
        type: DirectoryOrCreate
```
</details>

If someone would be willing, I would be interested in seeing the output of `/proc/mounts` for the pod that has `:z` in its mount

Walnux, 2022-01-17:
We also tested that with spc_t we don't have to use :z. 
You can see below  documents why we tried :z
https://www.redhat.com/sysadmin/user-namespaces-selinux-rootless-containers

haircommander, 2022-01-17:
> We also tested that with spc_t we don't have to use :z.
You can see below documents why we tried :z
https://www.redhat.com/sysadmin/user-namespaces-selinux-rootless-containers

yeah that makes sense. However, CRI-O should not be processing `:z`, whereas podman is expected to. I see why you'd try it, but I mostly want to figure out why it worked (and possibly stop it from working, if it looks like a bug)

haircommander, 2022-01-17:
for instance, if you change it to
`mountPath: '/dev/sgx_enclave:Z'`

or even differently:
`mountPath:'/dev/sgx_enclave_other'`

without spc_t, does it work?

Walnux, 2022-01-17:
The /dev directory with :z doesn't work.
only the other normal directory 
like mountPath: '/etc/kubernetes/node-feature-discovery/source.d/:z'
it works.

Walnux, 2022-01-17:
according to
https://developers.redhat.com/blog/2014/11/06/introducing-a-super-privileged-container-concept#
spc_t is still the a super priviledged conatiner which "only" applies the mount namespace. (Please correct me)
I feel this is still pretty privileged for us.
Firstly, for the init container we only need to copy the NFD hook from conatiner into /etc/kubernetes/node-feature-discovery/source.d/ on the host. I don't think we have to use spec_t. Personally, I like the :z solution.
Secondly, for the sgx plugin container. we only need to acess the /dev/sgx_x device interface on host from the container. I think  running as spc_t assinged too much privileges for the container. A more fine control might be needed instead of directly run as some pretty prioviledged container.
I actually like the idea of
https://github.com/kubernetes/kubernetes/issues/60748

Further more if we have to use spc_t, we have to carefully inspect the these two container images and make sure we didn't include any extra binaies that is not needed and increase the potential the security attack interfaces.

We have quickly go through the UBI base image, the smallest one we can find is UBI-micro which is ~30M after decompressed.
We are using
https://github.com/intel/intel-device-plugins-for-kubernetes/issues/852
to track the UBI based image task

haircommander, 2022-01-18:
> Firstly, for the init container we only need to copy the NFD hook from conatiner into /etc/kubernetes/node-feature-discovery/source.d/ on the host. I don't think we have to use spec_t. Personally, I like the :z solution.

my point is that `:z` solution shouldn't work, and if it does I want to stop it from working.
can you try 
```
      volumeMounts:
        - name: nfd-source-hooks
          mountPath: '/etc/kubernetes/node-feature-discovery/source.d/:Z'
```

also, what installs the device `/dev/sgx_x`? maybe a selinux rule could be added to allow containers access to it?

or better yet, do you have access to a node with this device that I can play around with? I would be happy to investigate a solution for y'all (ideally that doesn't give spc_t)

mythi, 2022-01-18:
> also, what installs the device /dev/sgx_x? maybe a selinux rule could be added to allow containers access to it?

it's an in-tree kernel driver, RHEL 8.4+ has it as a tech preview. would that already cover the rules part automatically? I'll work with @Walnux to check if was get you an access to a node with SGX.

Walnux, 2022-01-18:
volumeMounts:
        - name: nfd-source-hooks
          mountPath: '/etc/kubernetes/node-feature-discovery/source.d/:Z'
I have tied and it can work without spc_t. :)

haircommander, 2022-01-18:
> I have tied and it can work without spc_t. :)

what is `/proc/mounts` inside the container that has this mounted? I believe you have just mounted the literal directory `/etc/kubernetes/node-feature-discovery/source.d/:Z`. As a final piece of experimentation, can you try
```
  volumeMounts:
    - name: nfd-source-hooks
      mountPath: '/etc/kubernetes/node-feature-discovery/source.d/sources'
```
as I believe all it's doing is creating a new directory there

mythi, 2022-01-19:
@Walnux you can build toybox using cat support pretty easily:
```
$ git diff
diff --git a/build/docker/toybox-config b/build/docker/toybox-config
index df9e6d3..f415aa8 100644
--- a/build/docker/toybox-config
+++ b/build/docker/toybox-config
@@ -21,7 +21,7 @@ CONFIG_TOYBOX_GETRANDOM=y
 #
 # CONFIG_BASENAME is not set
 # CONFIG_CAL is not set
-# CONFIG_CAT is not set
+CONFIG_CAT=y
 # CONFIG_CAT_V is not set
 # CONFIG_CATV is not set
 # CONFIG_CHGRP is not set
$ make intel-sgx-initcontainer
...
$ docker run --entrypoint "" intel/intel-sgx-initcontainer:devel cat /proc/mounts
# (change the initContainer command to cat instead of the default entrypoint)
```

uMartinXu, 2022-01-20:
daemonset yaml without :z

```
      initContainers:
        - resources: {}
          terminationMessagePath: /dev/termination-log
          name: intel-sgx-initcontainer
          command:
            - sh
            - '-c'
            - >-
              cat /proc/mounts && cp -a /usr/local/bin/sgx-sw/intel-sgx-epchook
              /etc/kubernetes/node-feature-discovery/source.d/
          securityContext:
            readOnlyRootFilesystem: false
          imagePullPolicy: IfNotPresent
          volumeMounts:
            - name: nfd-source-hooks
              mountPath: /etc/kubernetes/node-feature-discovery/source.d/

```

Log: 
overlay / overlay rw,context="system_u:object_r:container_file_t:s0:c88,c734",relatime,lowerdir=/var/lib/containers/storage/overlay/l/CGEOKTEVBSWVW37STEBG7DSUZK:/var/lib/containers/storage/overlay/l/ZZ7PBK43SV6NRMUKXNTMLRJ2DG,upperdir=/var/lib/containers/storage/overlay/1ab5e921e8cba30560e8838dbb8b635553715681c5e743fe370045fe3b03e2ba/diff,workdir=/var/lib/containers/storage/overlay/1ab5e921e8cba30560e8838dbb8b635553715681c5e743fe370045fe3b03e2ba/work 0 0
proc /proc proc rw,nosuid,nodev,noexec,relatime 0 0
tmpfs /dev tmpfs rw,context="system_u:object_r:container_file_t:s0:c88,c734",nosuid,size=65536k,mode=755 0 0
devpts /dev/pts devpts rw,context="system_u:object_r:container_file_t:s0:c88,c734",nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=666 0 0
mqueue /dev/mqueue mqueue rw,seclabel,nosuid,nodev,noexec,relatime 0 0
sysfs /sys sysfs ro,seclabel,nosuid,nodev,noexec,relatime 0 0
tmpfs /sys/fs/cgroup tmpfs rw,context="system_u:object_r:container_file_t:s0:c88,c734",nosuid,nodev,noexec,relatime,mode=755 0 0
cgroup /sys/fs/cgroup/systemd cgroup ro,seclabel,nosuid,nodev,noexec,relatime,xattr,release_agent=/usr/lib/systemd/systemd-cgroups-agent,name=systemd 0 0
cgroup /sys/fs/cgroup/pids cgroup ro,seclabel,nosuid,nodev,noexec,relatime,pids 0 0
cgroup /sys/fs/cgroup/cpuset cgroup ro,seclabel,nosuid,nodev,noexec,relatime,cpuset 0 0
cgroup /sys/fs/cgroup/cpu,cpuacct cgroup ro,seclabel,nosuid,nodev,noexec,relatime,cpu,cpuacct 0 0
cgroup /sys/fs/cgroup/blkio cgroup ro,seclabel,nosuid,nodev,noexec,relatime,blkio 0 0
cgroup /sys/fs/cgroup/memory cgroup ro,seclabel,nosuid,nodev,noexec,relatime,memory 0 0
cgroup /sys/fs/cgroup/net_cls,net_prio cgroup ro,seclabel,nosuid,nodev,noexec,relatime,net_cls,net_prio 0 0
cgroup /sys/fs/cgroup/rdma cgroup ro,seclabel,nosuid,nodev,noexec,relatime,rdma 0 0
cgroup /sys/fs/cgroup/perf_event cgroup ro,seclabel,nosuid,nodev,noexec,relatime,perf_event 0 0
cgroup /sys/fs/cgroup/devices cgroup ro,seclabel,nosuid,nodev,noexec,relatime,devices 0 0
cgroup /sys/fs/cgroup/freezer cgroup ro,seclabel,nosuid,nodev,noexec,relatime,freezer 0 0
cgroup /sys/fs/cgroup/hugetlb cgroup ro,seclabel,nosuid,nodev,noexec,relatime,hugetlb 0 0
shm /dev/shm tmpfs rw,context="system_u:object_r:container_file_t:s0:c88,c734",nosuid,nodev,noexec,relatime,size=65536k 0 0
tmpfs /etc/resolv.conf tmpfs rw,seclabel,nosuid,nodev,noexec,mode=755 0 0
tmpfs /etc/hostname tmpfs rw,seclabel,nosuid,nodev,mode=755 0 0
/dev/sda4 /etc/hosts xfs rw,seclabel,relatime,attr2,inode64,logbufs=8,logbsize=32k,prjquota 0 0
/dev/sda4 /dev/termination-log xfs rw,seclabel,relatime,attr2,inode64,logbufs=8,logbsize=32k,prjquota 0 0
tmpfs /run/secrets tmpfs rw,seclabel,nosuid,nodev,mode=755 0 0
**/dev/sda4 /etc/kubernetes/node-feature-discovery/source.d xfs rw,seclabel,relatime,attr2,inode64,logbufs=8,logbsize=32k,prjquota 0 0**
tmpfs /var/run/secrets/kubernetes.io/serviceaccount tmpfs ro,seclabel,relatime,size=262629092k 0 0
proc /proc/bus proc ro,nosuid,nodev,noexec,relatime 0 0
proc /proc/fs proc ro,nosuid,nodev,noexec,relatime 0 0
proc /proc/irq proc ro,nosuid,nodev,noexec,relatime 0 0
proc /proc/sys proc ro,nosuid,nodev,noexec,relatime 0 0
proc /proc/sysrq-trigger proc ro,nosuid,nodev,noexec,relatime 0 0
tmpfs /proc/acpi tmpfs ro,context="system_u:object_r:container_file_t:s0:c88,c734",relatime 0 0
tmpfs /proc/kcore tmpfs rw,context="system_u:object_r:container_file_t:s0:c88,c734",nosuid,size=65536k,mode=755 0 0
tmpfs /proc/keys tmpfs rw,context="system_u:object_r:container_file_t:s0:c88,c734",nosuid,size=65536k,mode=755 0 0
tmpfs /proc/timer_list tmpfs rw,context="system_u:object_r:container_file_t:s0:c88,c734",nosuid,size=65536k,mode=755 0 0
tmpfs /proc/sched_debug tmpfs rw,context="system_u:object_r:container_file_t:s0:c88,c734",nosuid,size=65536k,mode=755 0 0
tmpfs /proc/scsi tmpfs ro,context="system_u:object_r:container_file_t:s0:c88,c734",relatime 0 0
tmpfs /sys/firmware tmpfs ro,context="system_u:object_r:container_file_t:s0:c88,c734",relatime 0 0
cp: /etc/kubernetes/node-feature-discovery/source.d//intel-sgx-epchook: Permission denied

daemonset yaml with :z
```
      initContainers:
        - resources: {}
          terminationMessagePath: /dev/termination-log
          name: intel-sgx-initcontainer
[... 29 lines cut ...]
cgroup /sys/fs/cgroup/rdma cgroup ro,seclabel,nosuid,nodev,noexec,relatime,rdma 0 0
cgroup /sys/fs/cgroup/perf_event cgroup ro,seclabel,nosuid,nodev,noexec,relatime,perf_event 0 0
cgroup /sys/fs/cgroup/devices cgroup ro,seclabel,nosuid,nodev,noexec,relatime,devices 0 0
cgroup /sys/fs/cgroup/freezer cgroup ro,seclabel,nosuid,nodev,noexec,relatime,freezer 0 0
cgroup /sys/fs/cgroup/hugetlb cgroup ro,seclabel,nosuid,nodev,noexec,relatime,hugetlb 0 0
shm /dev/shm tmpfs rw,context="system_u:object_r:container_file_t:s0:c235,c809",nosuid,nodev,noexec,relatime,size=65536k 0 0
[... cut ...]
tmpfs /proc/keys tmpfs rw,context="system_u:object_r:container_file_t:s0:c235,c809",nosuid,size=65536k,mode=755 0 0
tmpfs /proc/timer_list tmpfs rw,context="system_u:object_r:container_file_t:s0:c235,c809",nosuid,size=65536k,mode=755 0 0
tmpfs /proc/sched_debug tmpfs rw,context="system_u:object_r:container_file_t:s0:c235,c809",nosuid,size=65536k,mode=755 0 0
tmpfs /proc/scsi tmpfs ro,context="system_u:object_r:container_file_t:s0:c235,c809",relatime 0 0
tmpfs /sys/firmware tmpfs ro,context="system_u:object_r:container_file_t:s0:c235,c809",relatime 0 0

uMartinXu, 2022-01-20:
> it's an in-tree kernel driver, RHEL 8.4+ has it as a tech preview. would that already cover the rules part automatically? I'll work with @Walnux to check if was get you an access to a node with SGX.

we are trying to figure a way to let @haircommander access the node with SGX support. But that needs some time and efforts. Before that, I can just work as the proxy for @haircommander and try to figure out a proper way. :)

uMartinXu, 2022-01-24:
@haircommander Any updates?
Thanks!

haircommander, 2022-01-25:
what's `ls -lZd /etc/kubernetes/node-feature-discovery/source.d/` ? either we have to change the label of that directory so all containers can access it, or we need to make this plugin privileged.

uMartinXu, 2022-01-26:
I guess what you talking about is Host OS. 
I use below way to access the node:
[jxu36@jfz1r09h07 ~]$ oc debug node/worker-1
Starting pod/worker-1-debug ...
To use host binaries, run `chroot /host`
Pod IP: 172.16.9.2
If you don't see a command prompt, try pressing enter.
sh-4.4# chroot /host
sh-4.4# ls
bin  boot  dev  etc  home  lib  lib64  media  mnt  opt  ostree  proc  root  run  sbin  srv  sys  sysroot  tmp  usr  var
sh-4.4# ls -lZd /etc/kubernetes/node-feature-discovery/source.d/
drwxr-xr-x. 2 root root system_u:object_r:kubernetes_file_t:s0 31 Jan 21 20:40 /etc/kubernetes/node-feature-discovery/source.d/
sh-4.4# ls -lZd /dev
drwxr-xr-x. 20 root root system_u:object_r:device_t:s0 3360 Nov 30 06:09 /dev

rhatdan, 2022-01-27:
If you want this content to be read/write within a container it will need to be container_file_t.

uMartinXu, 2022-01-27:
@rhatdan it is from the node running RHCOS as HostOS.  How could the user set the proper label? I think the Label should be set by the OS and CRI-O before starting the Pod. Please correct me. Thanks!

haircommander, 2022-01-28:
what about instead of setting spc_t, we give the pod `container_runtime_t` ? that should be about to access the kubernetes dir. Then ni the SCC, we could set the seLinuxOption to MustRunAs

Walnux, 2022-01-28:
What does "ni" mean?
For Kubernetes dir accessing from Pod, looks like it is a better solution than spc_t. :)
How about the /dev/sgx_x accessing from Pod?

haircommander, 2022-01-28:
sorry, a typo, I meant to type "in" 

I believe access to devices need privilege unless they're going through a device plugin, so we otherwise have a circle and egg problem haha

Walnux, 2022-01-28:
@mythi what are your opinions? :)

mythi, 2022-02-07:
> I believe access to devices need privilege unless they're going through a device plugin

it does not sound right that we need privileged to just check the dev nodes are present. was there a device label (`device_t`? that could be used)

haircommander, 2022-02-07:
hm yeah it does look like there's a device_t, could you give it a try?

rhatdan, 2022-02-07:
device_t is generic, containers will not be allowed to use chr_file or block_file labeled device_t.

rhatdan, 2022-02-07:
```
sesearch -A -s container_t -c chr_file -p write
allow domain devtty_t:chr_file { append getattr ioctl lock open read write };
allow domain kmsg_device_t:chr_file { append getattr ioctl lock open write }; [ domain_can_write_kmsg ]:True
allow domain null_device_t:chr_file { append getattr ioctl lock open read write };
allow domain zero_device_t:chr_file { append getattr ioctl lock map open read write };
allow svirt_sandbox_domain sshd_devpts_t:chr_file { append getattr ioctl lock read write };
allow svirt_sandbox_domain user_devpts_t:chr_file { append getattr ioctl lock read write };
allow svirt_sandbox_domain user_tty_device_t:chr_file { append getattr ioctl lock read write };
allow syslog_client_type console_device_t:chr_file { append getattr ioctl lock open write };

```

Walnux, 2022-02-11:
@rhatdan @haircommander what is your suggestion then? Can we get a conclusion on this issue?
 thanks!

rhatdan, 2022-02-14:
I added a boolean to container-selinux policy to allow a container to have full access to all devices.

container_connect_any --> off
container_manage_cgroup --> off
container_use_cephfs --> off
**container_use_devices --> on**
logrotate_read_inside_containers --> off

$ sudo setsebool container_use_devices true

$ rpm -q container-selinux
container-selinux-2.176.0-2.fc36.noarch

The udica project is working on a fix for it, to be able to support more fine grained control.

Walnux, 2022-02-15:
@rhatdan Cool, this fine-grained control is the right thing to do. :)
Could you let us know when we can use it on an OCP release? Do you have a plan for it?

haircommander, 2022-03-04:
currently that version of containers-selinux isn't tareted at 4.10, so I am guessing at earliest it would make 4.11. We have to be careful about bumps to containers-selinux so close to GA, and the fix may not qualify for backport.

For the `/etc/kubernetes/node-feature-discovery/source.d/` piece, have you tried the solution in https://github.com/intel/intel-device-plugins-for-kubernetes/issues/762#issuecomment-1024259524

mregmi, 2022-03-07:
@haircommander I will try that solution. Also is there a way to test the up-coming fix for container-selinux in our cluster. is there a link to a branch or build for us to test? Thanks

Walnux, 2022-03-17:
Hi @haircommander @rhatdan,
Is it possible that we can define the Selinux policy/label to assign the permissions which are really needed by the container?
For our SGX device plugin it only access /dev/sgx_provision /dev/sgx_epc. So We can define a label called Intel_sgx_t to only assign the access permission to these two device files. And all the other device files access should be denied.
I think that is the best way to protect security. 
And I know we can define our own policy on Selinux in RHEL. But on OCP how to deploy the policy is a problem.

haircommander, 2022-03-17:
are you planning on using an operator to install on OCP? maybe it could install machine configs that enable that policy.

Walnux, 2022-03-20:
Yes, Operator should be the right way to install it. Can you point us some detail about the machine config? I think this should not a requirement from us. It should be a general request. :)

rhatdan, 2022-03-22:
If these devices get added to a container, then there is no need to label, the devices will get the label of the container.  If you are volume mounting them into the container, then they would not be allowed access.  @haircommander How do you add a device to a container with k8s?

haircommander, 2022-03-22:
that's the problem, this is a container that enables other containers to add devices. there's no way to do so without a device plugin, but we're putting together the device plugin...

rhatdan, 2022-03-22:
If you need to volume mount them in, and want the containers to have access then you could just 
`chcon -t container_file_t /dev/sgx*`

To make this permanent, you could execute something like:
```
semanage fcontext -a -t container_file_t '/dev/sgx.*'
restorecon -R -v /dev/sgx*
```
Then when the devices got created at reboot they would be labeled correctly.

mythi, 2022-03-23:
> To make this permanent, you could execute something like:
> 
> ```
> semanage fcontext -a -t container_file_t '/dev/sgx.*'
> restorecon -R -v /dev/sgx*
> ```

@rhatdan thanks! can these be managed by a `MachineConfig` or we need the Special Resource Operator to run them?

rhatdan, 2022-03-23:
Not my area of expertise, but I think MachineConfig should be able to do it.

haircommander, 2022-03-23:
typically the way to do it on rhcos is to create a machine config that creates a systemd unit file that runs the commands. Those kinds of state changes often don't persist across reboots otherwise. the sgx operator could create said machineconfig and trigger a reboot, then the device would be available on the next reboot

mythi, 2022-03-24:
> the sgx operator could create said machineconfig and trigger a reboot, then the device would be available on the next reboot

@haircommander OK, this sounds like a reasonable workaround until the problem gets fixed in the next/future release. However, I think we should try to leverage SRO+MCO operators for this and not add the functionality into the device plugins operator.

haircommander, 2022-03-24:
gotcha, then maybe SRO would be a good fit for this. Is there a registration of the SGX plugin in the SRO? ideally this unit would only be run when SGX device is enabled and installing.

Walnux, 2022-03-30:
@haircommander if we use SRO, should we install the policy from a container? If we can package the policy into a container and install it through a container, we can use the standard way to release and install policy on OCP. I am trying to do that. Do you guys know someone ever try it before?
And looks like the normal way to install the policy now is from rpm package.

haircommander, 2022-03-30:
not for policy but I do know privileged containers are used to configure things on the node. However, it's usually on startup from what I know. Something to think about: if we're using a privileged container to create a file on the host, is that much different from having the SGX plugin container being privileged?

haircommander, 2022-03-30:
good points, makes sense to me. @rhatdan if an selinux policy is configured, does the node need to be rebooted for it to take effect? (it's possible rhcos also behaves differently in this case, in which case we may need the reboot anyway)

rhatdan, 2022-03-30:
No SELinux does not require a reboot, as long as it was enabled in the first place.  Policy is instantly applied, and labels are placed on disk by restorecon.

mregmi, 2022-04-04:
@rhatdan One other issue we encountered is that it looks like the socket communication is not allowed between containers. our plugins use this to communicate. we had to manually create a selinux policy to allow it. Is there a way to allow this without deploying custom selinux policy. we used a policy something like:

#============= container_t ==============
allow container_t container_runtime_t:unix_stream_socket connectto;

rhatdan, 2022-04-04:
What is running as container_runtime_t?  The intel-device-plugin?

[6 more replies]

closed 2022-05-02 by Walnux