intel/intel-device-plugins-for-kubernetes #768: default nodeSelector labels

mythi, 2021-11-26:
currently, the device plugin CR samples set `nodeSelector` that no-one sets:

```
deployments/operator/samples/deviceplugin_v1_dlbdeviceplugin.yaml:  nodeSelector:
deployments/operator/samples/deviceplugin_v1_dlbdeviceplugin.yaml-    feature.node.kubernetes.io/dlb: 'true'
--
deployments/operator/samples/deviceplugin_v1_fpgadeviceplugin.yaml:  nodeSelector:
deployments/operator/samples/deviceplugin_v1_fpgadeviceplugin.yaml-    feature.node.kubernetes.io/fpga-arria10: 'true'
--
deployments/operator/samples/deviceplugin_v1_gpudeviceplugin.yaml:  nodeSelector:
deployments/operator/samples/deviceplugin_v1_gpudeviceplugin.yaml-    feature.node.kubernetes.io/gpu-i915: 'true'
--
deployments/operator/samples/deviceplugin_v1_qatdeviceplugin.yaml:  nodeSelector:
deployments/operator/samples/deviceplugin_v1_qatdeviceplugin.yaml-    feature.node.kubernetes.io/qat-c6xx: 'true'
--
deployments/operator/samples/deviceplugin_v1_sgxdeviceplugin.yaml:  nodeSelector:
deployments/operator/samples/deviceplugin_v1_sgxdeviceplugin.yaml-    feature.node.kubernetes.io/custom-intel.sgx: 'true'
```

(except SGX)

The next NFD release brings powerful labeling capabilities. We should have the samples defaults backed by NFD labeling rules and documentation how to deploy/use them.

This is to avoid end-user confusion that pods don't run when using the samples.

bart0sh, 2022-01-10:
@mythi
> The next NFD release brings powerful labeling capabilities.
Can you point me out on them? Are they described in the [customization guide](https://github.com/kubernetes-sigs/node-feature-discovery/pull/704)?

mythi, 2022-01-10:
@bart0sh that's the place for more info.

bart0sh, 2022-01-31:
@mythi would it be better to contribute to change NFD to support all our devices instead? So far it's done for sgx only.

bart0sh, 2022-01-31:
@mythi where would you suggest to put nodefeaturerules yaml[s]? They should be applied before plugins as nfd rescans them once per minute.

would it make sense to put all of them into one file and tell users to load all of them in advance with kubectl -f ?

PR #871 (2022-02-09): operator: implement NFD labelling rules
- added labeling rules for all supported devices
- updated operator installation instructions

Fixes: #768
commit 8626d47d (2022-02-09): operator: implement NFD labelling rules

- added labelling rules for all supported devices
- updated operator installation instructions

Fixes: #768
closed 2022-02-09 by mythi