intel/intel-device-plugins-for-kubernetes #773: deployments: add Helm charts

mythi, 2021-11-29:


maxirus, 2021-12-21:
I built a Chart [here](https://github.com/maxirus/helm-charts/tree/master/charts/intel-gpu). Would like to contribute this Chart to this project so that it can be easily maintained along side the device plugin itself.

A prerequisite to this contribution is the maintainers of this repo will need to set up either an [Artifact Hub](https://artifacthub.io/) acct/repo, GitHub Pages + CI workflow, or some other hosting solution. Once complete, I will open a PR and deprecate mine.

mythi, 2022-01-04:
@maxirus thanks!

We have Github Pages enabled so that can be used. However, I'd like to start with a Helm chart for the operator first. This is to reduce the maintenance burden of various deployment options.

msivosuo, 2022-05-18:
Fixed by #981

closed 2022-05-18 by msivosuo