intel/intel-device-plugins-for-kubernetes #775: OCP:  Operator Certification Step 1: Operator Container Image and plugin Container Image Cerificate

Walnux, 2021-11-29:
According to the OCP document:
https://redhat-connect.gitbook.io/partner-guide-for-red-hat-openshift-and-container/program-on-boarding/certification-workflow
we need to certificate the operator Container Image and plugin Container Image.

chaitanya1731, 2021-12-07:
Update 1 - 
Started working on certifying application container image. Started with intel-sgx-initcontainer image.. Tried scanning the image on redhat certification portal. It keeps loading for a few seconds and disappears.
We need to convert our current image to UBI base image..
Base image requirements -
- All images to be deployed in a cluster should be built on the same base image or images with the same Linux libraries.
- To ensure that the base image is compatible with the worker node’s operating system, the Linux libraries in the base image and the operating system running the worker nodes should be compatible versions of the same Linux distribution.
- The base image should be freely redistributable across public and private registries without requiring any paid licensing.
- The base image should run well in both Kubernetes and OpenShift.
- While a Kubernetes worker node can run a variety of Linux distributions, an OpenShift worker node requires a Linux distribution that is compatible with Red Hat Enterprise Linux.

Walnux, 2022-03-14:
The Image certification & release process is very clear. We can close this issue now.

closed 2022-03-14 by Walnux