intel/intel-device-plugins-for-kubernetes #776: OCP: work on Bundle Image running on OCP OLM (Operator Lifecycle Manager), doing Certification, testing and releasing

Walnux, 2021-11-29:
Build the proper operator Bundle Image for OCP OLM. 
1. Meets the Webhook requirements and removes the dependent on crt-management. 
2. Figure out the proper way to run NFD through NFD-operator: The NFD should run by this operator or by cluster user/adminstrator through NFD-operator? 
4. push the Bundle image to the proper repository
5. Do the Certification and testing on this bundle image
6. Add the proper release notes and readme for the release of the operator on OCP
7. more to add

Walnux, 2021-12-01:
Below task is done
2. Figure out the proper way to run NFD through NFD-operator: The NFD should run by this operator or by cluster user/adminstrator through NFD-operator?
for detail see task/issue https://github.com/intel/intel-device-plugins-for-kubernetes/issues/781

Walnux, 2021-12-01:
For task 1, operator-sdk can create the bundle image with CSV file using OLM webhook, please reference the related CSV file.
And The operator can start properly without crt-manager.
You can simply run operator using below command on OCP
`$ operator-sdk run bundle docker.io/walnuxdocker/intel-device-plugins-operator-bundle:0.22.0    `

Walnux, 2022-01-13:
the bundle image has been upgraded the v0.23.0
to run the image to install operator use below command
operator-sdk run bundle docker.io/walnuxdocker/intel-device-plugins-operator-bundle:0.23.0

closed 2022-01-13 by Walnux

mythi, 2022-01-13:
@Walnux can that be generated using `make bundle`?

Walnux, 2022-01-19:
@mythi yes. it is created from make bulndle.