intel/intel-device-plugins-for-kubernetes #780: OCP: Merge SGX HW feature detecting fixing into OCP

Walnux, 2021-11-30:
This SGX HW feature detecting issue is resolved in upstream see
https://github.com/intel/intel-device-plugins-for-kubernetes/issues/739
We need to make sure this fix can be merged into the NFD-Operator as well as the OCP

Walnux, 2021-11-30:
The NFD upstream fixing is:
https://github.com/kubernetes-sigs/node-feature-discovery/commit/8a394346598e2be3fc769ed29f38547784a212f1

Walnux, 2021-12-01:
According to https://github.com/openshift/cluster-nfd-operator/issues/230
@ArangoGutierrez  will merge this fix into OCP-4.9 and OCP-4.10

Walnux, 2022-01-18:
@ArangoGutierrez
NFD-0.10.0 upstream has been released. Could you let us know the statutes in OCP-4.9 and 4.10?
Looks like in OCP-4.9 the latest NFD-operator version is 4.9.0-202111171831.
Thanks!

Walnux, 2022-01-18:
@ArangoGutierrez since we are working on the integration testing, NFD and NFD operator is the most important part. Is it possible to share us the bundle image, so we can install it on 4.9 manually? So we can verify it as early as possible. Thanks!

ArangoGutierrez, 2022-02-08:
Backporting such commit into 4.9 will require porting too much, a workaround, tested, is to set `quay.io/openshift/origin-node-feature-discovery:4.10` on the image field of the CR 

```yaml
apiVersion: nfd.openshift.io/v1
kind: NodeFeatureDiscovery
metadata:
  name: nfd-instance
  namespace: openshift-nfd
spec:
  #instance: "" # instance is empty by default
  #labelWhiteList: ""
  #extraLabelNs:
  #  - "example.com"
  #resourceLabels:
  #  - "example.com/resource"
  operand:
    image: quay.io/openshift/origin-node-feature-discovery:4.10
    imagePullPolicy: Always
    servicePort: 12000
  workerConfig:
    configData: |
      core:
        sleepInterval: 60s
        pci:
          deviceClassWhitelist:
            - "0200"
            - "03"
            - "12"
          deviceLabelFields:
            - "vendor"
```

Walnux, 2022-02-08:
@ArangoGutierrez Do you know when 4.10 will be released?
We can use the workaround to do the development work. But, if NFD-operator can't be backported to 4.9. We can't officially claim that SGX is supported on 4.9.
And we have to do the test and certificate and release on 4.10.

mythi, 2022-02-08:
> But, if NFD-operator can't be backported to 4.9. We can't officially claim that SGX is supported on 4.9.

@Walnux as suggested by @ArangoGutierrez we can use the existing NFD-Operator in 4.9 and create a `NodeFeatureDiscovery` resource with the fixed NFD image. We'd be testing with the same NFD version on both 4.9 and 4.10.

ArangoGutierrez, 2022-02-08:
Yes, you can use the 4.9 Operator, just point to the 4.10 OPERAND on the CR, as seen in the yaml above

Walnux, 2022-02-08:
@mythi yes, we will use that workaround to develop and test on 4.9 before we have 4.10.
But for the final release and certificate we have to do it on 4.10 if NFD operator is not backported on 4.9.

ArangoGutierrez, 2022-02-08:
and the 4.10 image as you can see is from the official openshift quay repo

mythi, 2022-02-08:
> NFD operator is not backported on 4.9.

the backporting is not needed

Walnux, 2022-02-14:
I tried this workround. The result is: 
**worker1-node labels**
 feature.node.kubernetes.io/custom-intel.sgx=true
 feature.node.kubernetes.io/cpu-sgx.enabled=true
**worker0-node labels**
 no sgx related label

And at the nfd-master log files there has some err, but I think it doesn't impact SGX feature
@mythi @ArangoGutierrez please see below detail CR file and log files and check whether it is what we want. 

**NFD-operator CR file**
```
apiVersion: nfd.openshift.io/v1
kind: NodeFeatureDiscovery
metadata:
  creationTimestamp: '2022-02-14T02:09:12Z'
  finalizers:
    - foreground-deletion
  generation: 4
  managedFields:
    - apiVersion: nfd.openshift.io/v1
      fieldsType: FieldsV1
      fieldsV1:
        'f:metadata':
          'f:finalizers':
            .: {}
            'v:"foreground-deletion"': {}
        'f:spec':
          'f:instance': {}
      manager: node-feature-discovery-operator
      operation: Update
      time: '2022-02-14T02:09:12Z'
    - apiVersion: nfd.openshift.io/v1
      fieldsType: FieldsV1
      fieldsV1:
        'f:status':
          .: {}
          'f:conditions': {}
      manager: node-feature-discovery-operator
      operation: Update
      subresource: status
      time: '2022-02-14T02:09:12Z'
    - apiVersion: nfd.openshift.io/v1
      fieldsType: FieldsV1
      fieldsV1:
        'f:spec':
          .: {}
          'f:customConfig':
            .: {}
            'f:configData': {}
          'f:extraLabelNs': {}
          'f:operand':
            .: {}
            'f:image': {}
            'f:imagePullPolicy': {}
            'f:namespace': {}
            'f:servicePort': {}
          'f:resourceLabels': {}
          'f:workerConfig':
            .: {}
            'f:configData': {}
      manager: Mozilla
      operation: Update
      time: '2022-02-14T02:16:43Z'
  name: nfd-instance
  namespace: openshift-operators
  resourceVersion: '14201564'
  uid: 5d865ea8-5bbf-476d-a9a3-3e1aa53cdf3c
spec:
  customConfig:
[... 189 lines cut ...]
I0214 02:27:55.610848       1 component.go:36] [core]Subchannel Connectivity change to READY
I0214 02:27:55.610876       1 component.go:36] [core]Channel Connectivity change to READY
I0214 02:27:55.610893       1 nfd-worker.go:212] reloading configuration
I0214 02:27:55.611144       1 nfd-worker.go:423] configuration file "/etc/kubernetes/node-feature-discovery/nfd-worker.conf" parsed
I0214 02:27:55.611203       1 nfd-worker.go:461] worker (re-)configuration successfully completed
I0214 02:27:55.692384       1 nfd-worker.go:472] starting feature discovery...
I0214 02:27:55.693063       1 nfd-worker.go:484] feature discovery completed
I0214 02:27:55.693073       1 nfd-worker.go:565] sending labeling request to nfd-master
I0214 02:28:55.791945       1 nfd-worker.go:472] starting feature discovery...
I0214 02:28:55.792535       1 nfd-worker.go:484] feature discovery completed
I0214 02:28:55.792546       1 nfd-worker.go:565] sending labeling request to nfd-master
I0214 02:29:24.067321       1 nfd-worker.go:212] reloading configuration

**nfd-master log**
30 lines
W0214 02:27:53.337925       1 main.go:61] version not set! Set -ldflags "-X github.com/openshift/node-feature-discovery/pkg/version.version=`git describe --tags --dirty --always`" during build or run.
I0214 02:27:53.338162       1 nfd-master.go:170] Node Feature Discovery Master undefined
I0214 02:27:53.338168       1 nfd-master.go:174] NodeName: "master-1"
I0214 02:27:53.338172       1 nfd-master.go:185] starting nfd LabelRule controller
E0214 02:27:53.346981       1 reflector.go:138] pkg/generated/informers/externalversions/factory.go:117: Failed to watch *v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:openshift-operators:nfd-master" cannot list resource "nodefeaturerules" in API group "nfd.k8s-sigs.io" at the cluster scope
I0214 02:27:53.350327       1 nfd-master.go:226] gRPC server serving on port: 12000
E0214 02:27:54.633096       1 reflector.go:138] pkg/generated/informers/externalversions/factory.go:117: Failed to watch *v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:openshift-operators:nfd-master" cannot list resource "nodefeaturerules" in API group "nfd.k8s-sigs.io" at the cluster scope
I0214 02:27:55.702889       1 nfd-master.go:420] received labeling request for node "worker-1"
E0214 02:27:57.741749       1 reflector.go:138] pkg/generated/informers/externalversions/factory.go:117: Failed to watch *v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:openshift-operators:nfd-master" cannot list resource "nodefeaturerules" in API group "nfd.k8s-sigs.io" at the cluster scope
E0214 02:28:03.075046       1 reflector.go:138] pkg/generated/informers/externalversions/factory.go:117: Failed to watch *v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:openshift-operators:nfd-master" cannot list resource "nodefeaturerules" in API group "nfd.k8s-sigs.io" at the cluster scope

Walnux, 2022-02-14:
@mythi @ArangoGutierrez 
according to
https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/deployments/sgx_nfd/nfd-worker.conf

Looks like I should set the CR as below. But nfd-server runs into error, see the flowing log. 
```
apiVersion: nfd.openshift.io/v1
kind: NodeFeatureDiscovery
metadata:
  creationTimestamp: '2022-02-14T02:09:12Z'
  finalizers:
    - foreground-deletion
  generation: 3
  managedFields:
    - apiVersion: nfd.openshift.io/v1
      fieldsType: FieldsV1
      fieldsV1:
        'f:metadata':
          'f:finalizers':
            .: {}
            'v:"foreground-deletion"': {}
        'f:spec':
          'f:instance': {}
      manager: node-feature-discovery-operator
      operation: Update
      time: '2022-02-14T02:09:12Z'
    - apiVersion: nfd.openshift.io/v1
      fieldsType: FieldsV1
      fieldsV1:
        'f:status':
          .: {}
          'f:conditions': {}
      manager: node-feature-discovery-operator
      operation: Update
      subresource: status
      time: '2022-02-14T02:09:12Z'
    - apiVersion: nfd.openshift.io/v1
      fieldsType: FieldsV1
      fieldsV1:
        'f:spec':
          .: {}
          'f:customConfig':
            .: {}
            'f:configData': {}
          'f:extraLabelNs': {}
          'f:operand':
            .: {}
            'f:image': {}
            'f:imagePullPolicy': {}
            'f:namespace': {}
            'f:servicePort': {}
          'f:resourceLabels': {}
          'f:workerConfig':
            .: {}
            'f:configData': {}
      manager: Mozilla
      operation: Update
      time: '2022-02-14T02:16:43Z'
  name: nfd-instance
  namespace: openshift-operators
  resourceVersion: '14189951'
  uid: 5d865ea8-5bbf-476d-a9a3-3e1aa53cdf3c
spec:
  customConfig:
    configData: |
      #    - name: "more.kernel.features"
      #      matchOn:
      #      - loadedKMod: ["example_kmod3"]
      #    - name: "more.features.by.nodename"
      #      value: customValue
[... 153 lines cut ...]
4 lines
W0214 02:20:03.222462       1 main.go:51] version not set! Set -ldflags "-X github.com/openshift/node-feature-discovery/pkg/version.version=`git describe --tags --dirty --always`" during build or run.
I0214 02:20:03.222566       1 nfd-worker.go:155] Node Feature Discovery Worker undefined
I0214 02:20:03.222569       1 nfd-worker.go:156] NodeName: 'worker-1'
F0214 02:20:03.222707       1 main.go:64] failed to parse config file: error converting YAML to JSON: yaml: line 84: mapping values are not allowed in this context

log from nfd-master
17 lines
W0214 02:17:01.210758       1 main.go:61] version not set! Set -ldflags "-X github.com/openshift/node-feature-discovery/pkg/version.version=`git describe --tags --dirty --always`" during build or run.
I0214 02:17:01.210978       1 nfd-master.go:170] Node Feature Discovery Master undefined
I0214 02:17:01.210981       1 nfd-master.go:174] NodeName: "master-0"
I0214 02:17:01.210985       1 nfd-master.go:185] starting nfd LabelRule controller
E0214 02:17:01.219422       1 reflector.go:138] pkg/generated/informers/externalversions/factory.go:117: Failed to watch *v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:openshift-operators:nfd-master" cannot list resource "nodefeaturerules" in API group "nfd.k8s-sigs.io" at the cluster scope
I0214 02:17:01.220199       1 nfd-master.go:226] gRPC server serving on port: 12000
E0214 02:17:02.505835       1 reflector.go:138] pkg/generated/informers/externalversions/factory.go:117: Failed to watch *v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:openshift-operators:nfd-master" cannot list resource "nodefeaturerules" in API group "nfd.k8s-sigs.io" at the cluster scope
E0214 02:17:05.614283       1 reflector.go:138] pkg/generated/informers/externalversions/factory.go:117: Failed to watch *v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:openshift-operators:nfd-master" cannot list resource "nodefeaturerules" in API group "nfd.k8s-sigs.io" at the cluster scope
E0214 02:17:10.947487       1 reflector.go:138] pkg/generated/informers/externalversions/factory.go:117: Failed to watch *v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:openshift-operators:nfd-master" cannot list resource "nodefeaturerules" in API group "nfd.k8s-sigs.io" at the cluster scope
E0214 02:17:20.153138       1 reflector.go:138] pkg/generated/informers/externalversions/factory.go:117: Failed to watch *v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:openshift-operators:nfd-master" cannot list resource "nodefeaturerules" in API group "nfd.k8s-sigs.io" at the cluster scope
E0214 02:17:38.393163       1 reflector.go:138] pkg/generated/informers/externalversions/factory.go:117: Failed to watch *v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:openshift-operators:nfd-master" cannot list resource "nodefeaturerules" in API group "nfd.k8s-sigs.io" at the cluster scope
[... cut ...]
E0214 02:20:40.194114       1 reflector.go:138] pkg/generated/informers/externalversions/factory.go:117: Failed to watch *v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:openshift-operators:nfd-master" cannot list resource "nodefeaturerules" in API group "nfd.k8s-sigs.io" at the cluster scope
E0214 02:21:25.657791       1 reflector.go:138] pkg/generated/informers/externalversions/factory.go:117: Failed to watch *v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:open

mythi, 2022-02-14:
> @mythi @ArangoGutierrez please have a look at below detail CR file and log files and check whether it is what we want.

@Walnux Looks OK to me if `worker0-node` should not be labeled.

Walnux, 2022-02-14:
worker0 doesn't have SGX support. So it should not be labelled. 
But in order to make it work, looks like we also need below patch
```
diff --git a/deployments/operator/samples/deviceplugin_v1_sgxdeviceplugin.yaml b/deployments/operator/samples/deviceplugin_v1_sgxdeviceplugin.yaml
index df842d7..2f2a422 100644
--- a/deployments/operator/samples/deviceplugin_v1_sgxdeviceplugin.yaml
+++ b/deployments/operator/samples/deviceplugin_v1_sgxdeviceplugin.yaml
@@ -9,4 +9,4 @@ spec:
   provisionLimit: 110
   logLevel: 4
   nodeSelector:
-    feature.node.kubernetes.io/intel.sgx: 'true'
+    feature.node.kubernetes.io/custom-intel.sgx: 'true' 
```

mythi, 2022-02-15:
> `-    feature.node.kubernetes.io/intel.sgx: 'true'`
>`+    feature.node.kubernetes.io/custom-intel.sgx: 'true'`

@Walnux NFD v0.10 dropped `custom-` so something is not backported in OCP NFD 4.10. You don't have to follow this example though. It just needs to be in sync with the NFD configuration.

Walnux, 2022-02-15:
@ArangoGutierrez please have a look at comments from @mythi. I want to make sure in OCP NFD 4.10 it is backported properly.
Otherwise our plugin still can't work properly even on OCP-4.10.
Thanks!

ArangoGutierrez, 2022-02-15:
thanks for looking into this, will take a look

ArangoGutierrez, 2022-02-16:
https://github.com/openshift/node-feature-discovery/commit/8a394346598e2be3fc769ed29f38547784a212f1 commit is in
@mythi  I am a bit confused here, using the 4.10 image is not working for you?

ArangoGutierrez, 2022-02-16:
```bash
*v1alpha1.NodeFeatureRule: failed to list *v1alpha1.NodeFeatureRule: nodefeaturerules.nfd.k8s-sigs.io is forbidden: User "system:serviceaccount:open
```
I know what this error is, thanks for the catch!

ArangoGutierrez, 2022-02-16:
https://github.com/openshift/node-feature-discovery/pull/77

mythi, 2022-02-17:
> @mythi I am a bit confused here, using the 4.10 image is not working for you?

@ArangoGutierrez  @Walnux reported that 4.10 still adds `custom-` in the labels which would suggest that https://github.com/kubernetes-sigs/node-feature-discovery/pull/697 is missing

Walnux, 2022-02-22:
@mythi @ArangoGutierrez That is cool. 
@ArangoGutierrez I just want to double confirm it will be fixed in 4.10 GA. Right. :)

ArangoGutierrez, 2022-02-23:
https://github.com/openshift/node-feature-discovery/pull/79

ArangoGutierrez, 2022-02-24:
https://github.com/openshift/node-feature-discovery/pull/80

ArangoGutierrez, 2022-02-24:
https://github.com/kubernetes-sigs/node-feature-discovery/pull/697 Merged into 4.10 , let it roll out for a day before testing it

ArangoGutierrez, 2022-02-24:
@Walnux @mythi ^ let me know if extra moves are needed on my end

mythi, 2022-03-03:
@ArangoGutierrez sorry for the delay, I cannot think of anything else right now. If these will be available in 4.10 we are good. Can you confirm that?

Walnux, 2022-03-07:
@ArangoGutierrez now we are working on upgrading our Cluster to 4.10. And we will verify it on 4.10. And will let you know the result when it is done.
Thank you very much for your help. :)

mregmi, 2022-03-15:
Verified that this works. closing this.

Walnux, 2022-03-16:
@mregmi @ArangoGutierrez  @mythi thanks for your great help here. Now I am very happy to close this issue. :)

closed 2022-03-16 by Walnux