intel/intel-device-plugins-for-kubernetes #781: OCP: deploy NFD through NFD-operator

Walnux, 2021-12-01:
After the SGX detect fixing got merged into OCP see https://github.com/intel/intel-device-plugins-for-kubernetes/issues/780
We need work out the solution to deploy NFD through NFD-operator on OCP.

According to feed back from https://github.com/openshift/cluster-nfd-operator/issues/231 by @ArangoGutierrez

We need firstly add required nfd-operator CRD in CSV file. like what has been done in SRO project see 
https://github.com/openshift/special-resource-operator/blob/master/config/manifests/bases/openshift-special-resource-operator.clusterserviceversion.yaml#L28

And then we should provide the user (cluster-administrator) the YAML file to create the CR and deploy the NFD and enable the SGX detecting.

The document on how to configure CR is https://docs.openshift.com/container-platform/4.9/hardware_enablement/psap-node-feature-discovery-operator.html

mregmi, 2022-03-16:
@mythi do you think we can provide a sample NFD CR file for SGX and other operators like this in this repo? if yes we can create a pull request for it.

mythi, 2022-03-17:
> @mythi do you think we can provide a sample NFD CR file for SGX and other operators like this in this repo?

That works for me. See `deployments/nfd`. Try to move to use `NodeFeatureRule` since the NFD operator adds the CRD for it and we maintain them in this repo also.

Walnux, 2022-05-02:
https://github.com/intel/intel-device-plugins-for-kubernetes/pull/932 is used to fix this issue.

PR #932 (2022-05-06): sgx: add a sample nfd CR that can be used in openshift
This sample nfd cr can be used to make a new nfd instance with SGX support in openshift.

Fixes: #781
closed 2022-05-06 by mythi