intel/intel-device-plugins-for-kubernetes #791: intel-gpu-plugin is running, but i915 label is not being attached

eltomato89, 2021-12-07:
Host OS: **Ubuntu 20.04.3 LTS**
K8s version: **v1.22.4-3**+adc4115d990346 on **microk8s**
intel device plugin version: **v0.22.0** (also tried v0.21.0)
CPU/GPU: Intel(R) **Xeon(R) CPU E3-1226 v3** @ 3.30GHz integrated graphics

The plugin pod is running but does not show any logs (using kubectl logs ..)

When I run 
`kubectl get nodes -o=jsonpath="{range .items[*]}{.metadata.name}{'\n'}{' i915: '}{.status.allocatable.gpu\.intel\.com/i915}{'\n'}"`
output looks like this:

```
hades
 i915:

 i915:
```

Also my plex pod does not get scheduled using 
```
resources:
            limits:
              gpu.intel.com/i915: 1
```
kubectl describe pod/plex-... gives:
```
Events:
  Type     Reason            Age   From               Message
  ----     ------            ----  ----               -------
  Warning  FailedScheduling  43s   default-scheduler  0/1 nodes are available: 1 Insufficient gpu.intel.com/i915.
```

I was using the intel device plugin (v0.19.0) on microk8s before on this machine .. worked with no issues.

Any Ideas on whats the problem here?

uniemimu, 2021-12-08:
Most probably the scan() -function is not finding your GPU.

From that node, could you provide output of these three:
ls -la /sys/class/drm/card?
cat /sys/class/drm/card*/device/vendor
ls -la /dev/dri

Then the verbose version of the log of the plugin running in that node (kubectl logs -n namespacename podname) while the plugin is started with command line arg --v=5.

eltomato89, 2021-12-08:
Here we go!

**ls -la /sys/class/drm/card?**
```
koehler@hades:~$ ls -la /sys/class/drm/card?
ls: cannot access '/sys/class/drm/card?': No such file or directory
koehler@hades:~$ ls -la /sys/class/drm/card
ls: cannot access '/sys/class/drm/card': No such file or directory
```

**cat /sys/class/drm/card*/device/vendor**
```
koehler@hades:~$ cat /sys/class/drm/card*/device/vendor
cat: '/sys/class/drm/card*/device/vendor': No such file or directory
```

**ls -la /dev/dri**
```
koehler@hades:~$ ls -la /dev/dri
total 0
drwxr-xr-x  2 root root   40 Dez  8 15:18 .
drwxr-xr-x 21 root root 4480 Dez  8 15:19 ..
```

Then the verbose version of the log of the plugin running in that node (kubectl logs -n namespacename podname) while the plugin is started with command line arg --v=5.

```
╰─ kubectl logs -f pod/intel-gpu-plugin-6drcf
I1208 15:51:03.094842       1 gpu_plugin.go:175] GPU device plugin started
I1208 15:51:03.094957       1 gpu_plugin.go:112] Not compatible deviceversion
I1208 15:51:03.094963       1 gpu_plugin.go:87] GPU scan update: devices found: 0
I1208 15:51:08.095299       1 gpu_plugin.go:112] Not compatible deviceversion
I1208 15:51:13.095057       1 gpu_plugin.go:112] Not compatible deviceversion
I1208 15:51:18.095199       1 gpu_plugin.go:112] Not compatible deviceversion
I1208 15:51:23.095161       1 gpu_plugin.go:112] Not compatible deviceversion
```

eltomato89, 2021-12-08:
Okay .. seems like I am an idiot ..

A look into dmesg gave the answer:
```
[    0.079699] Kernel command line: BOOT_IMAGE=/vmlinuz-5.4.0-91-generic root=UUID=d5bc9cd5-0fb4-43bb-a821-af45cf3e6e2a ro nomodeset debian-installer/language=de keyboard-configuration/layoutcode?=de
[    0.079739] You have booted with nomodeset. This means your GPU drivers are DISABLED
[    0.079739] Any video related functionality will be severely degraded, and you may not even be able to suspend the system properly
[    0.079740] Unless you actually understand what nomodeset does, you should reboot without enabling it
```

I have indeed added the "nomodeset" as I was fighting with VGA/HDMI Adapter on PiKVM not getting a video stream...
removed it and now it works (even if my pikvm does not, grr)

thanks for your help!