intel/intel-device-plugins-for-kubernetes #811: failed calling webhook "webhook.cert-manager.io": failed to call webhook

belanasaikiran, 2021-12-17:
Iam trying to run fpga admission webhook, but Iam getting the following errors with the latest version of **Device Plugins of K8s**

```
server2@server2:~/Kubernetes/fpga-device-plugin$ kubectl apply -k intel-device-plugins-for-kubernetes/deployments/fpga_admissionwebhook/default/
namespace/intelfpgawebhook-system created
customresourcedefinition.apiextensions.k8s.io/acceleratorfunctions.fpga.intel.com created
customresourcedefinition.apiextensions.k8s.io/fpgaregions.fpga.intel.com created
clusterrole.rbac.authorization.k8s.io/intelfpgawebhook-manager-role created
clusterrolebinding.rbac.authorization.k8s.io/intelfpgawebhook-manager-rolebinding created
service/intelfpgawebhook-webhook-service created
deployment.apps/intelfpgawebhook-webhook created
mutatingwebhookconfiguration.admissionregistration.k8s.io/intelfpgawebhook-mutating-webhook-configuration created
Error from server (InternalError): error when creating "intel-device-plugins-for-kubernetes/deployments/fpga_admissionwebhook/default/": Internal error occurred: failed calling webhook "webhook.cert-manager.io": failed to call webhook: Post "https://cert-manager-webhook.cert-manager.svc:443/mutate?timeout=10s": context deadline exceeded
Error from server (InternalError): error when creating "intel-device-plugins-for-kubernetes/deployments/fpga_admissionwebhook/default/": Internal error occurred: failed calling webhook "webhook.cert-manager.io": failed to call webhook: Post "https://cert-manager-webhook.cert-manager.svc:443/mutate?timeout=10s": context deadline exceeded
```
Can someone help me in solving this issue please.

Also Iam trying the region mode of application deployment. The tutorial aren't explained properly, Is there any better tutorial or Documentation on using INTEL_FPGA_DEVICE_PLUGIN?.

Looking forward for respose @intel-k8s-bot @mythi

mythi, 2021-12-17:
@belanasaikiran are you behind proxies and you have proxy env variables set?

belanasaikiran, 2021-12-17:
@mythi  can you please provide commands to check if I am behind proxies ? Yes, I have deployed `cert-manager`

and here is the output I get when I check all pods and namespaces

```
server2@server2:~/Kubernetes/fpga-device-plugin/intel-device-plugins-for-kubernetes/deployments/fpga_admissionwebhook/default$ kubectl get all --all-namespaces
NAMESPACE                 NAME                                            READY   STATUS              RESTARTS       AGE
cert-manager              pod/cert-manager-cainjector-8c7796555-kjsl9     1/1     Running             0              96m
cert-manager              pod/cert-manager-cdc85d4c4-65fl6                1/1     Running             0              96m
cert-manager              pod/cert-manager-webhook-55f45df998-vbgp5       1/1     Running             0              96m
intelfpgawebhook-system   pod/intelfpgawebhook-webhook-7f9b997d8f-p7p4t   0/1     ContainerCreating   0              13m
kube-system               pod/calico-kube-controllers-647d84984b-pr2f8    1/1     Running             1 (102m ago)   103m
kube-system               pod/calico-node-ftjg9                           0/1     Running             0              103m
kube-system               pod/calico-node-lkpj5                           0/1     Running             0              103m
kube-system               pod/coredns-64897985d-djvd7                     1/1     Running             0              107m
kube-system               pod/coredns-64897985d-g7wnb                     1/1     Running             0              107m
kube-system               pod/etcd-server2                                1/1     Running             4              107m
kube-system               pod/kube-apiserver-server2                      1/1     Running             0              107m
kube-system               pod/kube-controller-manager-server2             1/1     Running             0              107m
kube-system               pod/kube-proxy-499mr                            1/1     Running             0              106m
kube-system               pod/kube-proxy-82s94                            1/1     Running             0              107m
kube-system               pod/kube-scheduler-server2                      1/1     Running             0              107m

NAMESPACE                 NAME                                       TYPE        CLUSTER-IP      EXTERNAL-IP   PORT(S)                  AGE
cert-manager              service/cert-manager                       ClusterIP   10.102.29.72    <none>        9402/TCP                 96m
cert-manager              service/cert-manager-webhook               ClusterIP   10.105.91.252   <none>        443/TCP                  96m
default                   service/kubernetes                         ClusterIP   10.96.0.1       <none>        443/TCP                  107m
intelfpgawebhook-system   service/intelfpgawebhook-webhook-service   ClusterIP   10.103.122.10   <none>        443/TCP                  41m
kube-system               service/kube-dns                           ClusterIP   10.96.0.10      <none>        53/UDP,53/TCP,9153/TCP   107m

NAMESPACE     NAME                         DESIRED   CURRENT   READY   UP-TO-DATE   AVAILABLE   NODE SELECTOR            AGE
kube-system   daemonset.apps/calico-node   2         2         0       2            0           kubernetes.io/os=linux   103m
kube-system   daemonset.apps/kube-proxy    2         2         2       2            2           kubernetes.io/os=linux   107m

NAMESPACE                 NAME                                       READY   UP-TO-DATE   AVAILABLE   AGE
cert-manager              deployment.apps/cert-manager               1/1     1            1           96m
cert-manager              deployment.apps/cert-manager-cainjector    1/1     1            1           96m
cert-manager              deployment.apps/cert-manager-webhook       1/1     1            1           96m
intelfpgawebhook-system   deployment.apps/intelfpgawebhook-webhook   0/1     1            0           41m
kube-system               deployment.apps/calico-kube-controllers    1/1     1            1           103m
kube-system               deployment.apps/coredns                    2/2     2            2           107m

NAMESPACE                 NAME                                                  DESIRED   CURRENT   READY   AGE
cert-manager              replicaset.apps/cert-manager-cainjector-8c7796555     1         1         1       96m
cert-manager              replicaset.apps/cert-manager-cdc85d4c4                1         1         1       96m
cert-manager              replicaset.apps/cert-manager-webhook-55f45df998       1         1         1       96m
intelfpgawebhook-system   replicaset.apps/intelfpgawebhook-webhook-7f9b997d8f   1         1         0       41m
kube-system               replicaset.apps/calico-kube-controllers-647d84984b    1         1         1       103m
kube-system               replicaset.apps/coredns-64897985d                     2         2         2       107m
```

mythi, 2021-12-17:
This doc talks about it: https://github.com/intel/intel-device-plugins-for-kubernetes/tree/main/cmd/operator#installation

We see that type of error when `no_proxy` is not correctly set.

belanasaikiran, 2021-12-17:
Thank you @mythi 

Yes, I have just followed the instructions from the above link you have shared, here's the output of api_server proxy

```
server2@server2:~/Kubernetes/fpga-device-plugin/intel-device-plugins-for-kubernetes/deployments/fpga_admissionwebhook/default$ kubectl describe pod kube-apiserver --namespace kube-system | grep -i no_proxy | grep "\.svc"
      no_proxy:     127.0.0.1,localhost,.example.com,10.0.0.0/8,.svc,.svc.cluster.local
```

and Here is the output when I tried to apply the `default operator`
```
server2@server2:~/Kubernetes/fpga-device-plugin/intel-device-plugins-for-kubernetes/deployments/fpga_admissionwebhook/default$  kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/operator/default
namespace/inteldeviceplugins-system created
customresourcedefinition.apiextensions.k8s.io/acceleratorfunctions.fpga.intel.com created
customresourcedefinition.apiextensions.k8s.io/dlbdeviceplugins.deviceplugin.intel.com created
customresourcedefinition.apiextensions.k8s.io/dsadeviceplugins.deviceplugin.intel.com created
customresourcedefinition.apiextensions.k8s.io/fpgadeviceplugins.deviceplugin.intel.com created
customresourcedefinition.apiextensions.k8s.io/fpgaregions.fpga.intel.com created
customresourcedefinition.apiextensions.k8s.io/gpudeviceplugins.deviceplugin.intel.com created
customresourcedefinition.apiextensions.k8s.io/qatdeviceplugins.deviceplugin.intel.com created
customresourcedefinition.apiextensions.k8s.io/sgxdeviceplugins.deviceplugin.intel.com created
role.rbac.authorization.k8s.io/inteldeviceplugins-leader-election-role created
clusterrole.rbac.authorization.k8s.io/inteldeviceplugins-gpu-manager-role created
clusterrole.rbac.authorization.k8s.io/inteldeviceplugins-manager-role created
clusterrole.rbac.authorization.k8s.io/inteldeviceplugins-metrics-reader created
clusterrole.rbac.authorization.k8s.io/inteldeviceplugins-proxy-role created
rolebinding.rbac.authorization.k8s.io/inteldeviceplugins-leader-election-rolebinding created
clusterrolebinding.rbac.authorization.k8s.io/inteldeviceplugins-manager-rolebinding created
clusterrolebinding.rbac.authorization.k8s.io/inteldeviceplugins-proxy-rolebinding created
service/inteldeviceplugins-controller-manager-metrics-service created
service/inteldeviceplugins-webhook-service created
deployment.apps/inteldeviceplugins-controller-manager created
mutatingwebhookconfiguration.admissionregistration.k8s.io/inteldeviceplugins-mutating-webhook-configuration created
validatingwebhookconfiguration.admissionregistration.k8s.io/inteldeviceplugins-validating-webhook-configuration created
Error from server (InternalError): error when creating "https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/operator/default": Internal error occurred: failed calling webhook "webhook.cert-manager.io": failed to call webhook: Post "https://cert-manager-webhook.cert-manager.svc:443/mutate?timeout=10s": context deadline exceeded
Error from server (InternalError): error when creating "https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/operator/default": Internal error occurred: failed calling webhook "webhook.cert-manager.io": failed to call webhook: Post "https://cert-manager-webhook.cert-manager.svc:443/mutate?timeout=10s": context deadline exceeded
```

mythi, 2021-12-17:
Looks like the fpga_admissionwebhook README talks about the same `$no_proxy` thing. Do you have the latest `cert-manager` version?

belanasaikiran, 2021-12-17:
I couldn't find any README file at `fpga_admissionwebhook`  folder, can you please share the link for README.

I have applied 1.6.1 version of `cert-manager`

`https://github.com/jetstack/cert-manager/releases/download/v1.6.1/cert-manager.yaml`

mythi, 2021-12-17:
It is there:
https://github.com/intel/intel-device-plugins-for-kubernetes/tree/main/cmd/fpga_admissionwebhook#installation

Yes `1.6.1` sounds OK.

belanasaikiran, 2021-12-17:
yes @mythi , I have just checked the documentation file and still while following the guide, the `inteldeviceplugin-controller-manager` is stuck at `ContainerCreating` state. 

Here are the logs.
```
server2@server2:~/Kubernetes/fpga-device-plugin/intel-device-plugins-for-kubernetes/deployments/fpga_admissionwebhook/crd/bases$ kubectl get all --all-namespaces
NAMESPACE                   NAME                                                         READY   STATUS              RESTARTS       AGE
cert-manager                pod/cert-manager-748f4d667d-5blvc                            1/1     Running             0              27m
cert-manager                pod/cert-manager-cainjector-6cd5988f45-s82vr                 1/1     Running             0              27m
cert-manager                pod/cert-manager-webhook-7498ddfd98-7st22                    1/1     Running             0              27m
inteldeviceplugins-system   pod/inteldeviceplugins-controller-manager-76dd86b7bd-qg6lq   0/2     ContainerCreating   0              21m
kube-system                 pod/calico-kube-controllers-647d84984b-pr2f8                 1/1     Running             1 (5h ago)     5h1m
kube-system                 pod/calico-node-ftjg9                                        0/1     Running             0              5h1m
kube-system                 pod/calico-node-lkpj5                                        0/1     Running             0              5h1m
kube-system                 pod/coredns-64897985d-djvd7                                  1/1     Running             0              5h5m
kube-system                 pod/coredns-64897985d-g7wnb                                  1/1     Running             0              5h5m
kube-system                 pod/etcd-server2                                             1/1     Running             4              5h5m
kube-system                 pod/kube-apiserver-server2                                   1/1     Running             0              3h4m
kube-system                 pod/kube-controller-manager-server2                          1/1     Running             1 (3h5m ago)   5h5m
kube-system                 pod/kube-proxy-499mr                                         1/1     Running             0              5h5m
kube-system                 pod/kube-proxy-82s94                                         1/1     Running             0              5h5m
kube-system                 pod/kube-scheduler-server2                                   1/1     Running             1 (3h5m ago)   5h5m

NAMESPACE                   NAME                                                            TYPE        CLUSTER-IP      EXTERNAL-IP   PORT(S)                  AGE
cert-manager                service/cert-manager                                            ClusterIP   10.100.131.0    <none>        9402/TCP                 27m
cert-manager                service/cert-manager-webhook                                    ClusterIP   10.96.122.87    <none>        443/TCP                  27m
default                     service/kubernetes                                              ClusterIP   10.96.0.1       <none>        443/TCP                  5h5m
inteldeviceplugins-system   service/inteldeviceplugins-controller-manager-metrics-service   ClusterIP   10.102.75.123   <none>        8443/TCP                 21m
inteldeviceplugins-system   service/inteldeviceplugins-webhook-service                      ClusterIP   10.102.173.0    <none>        443/TCP                  21m
kube-system                 service/kube-dns                                                ClusterIP   10.96.0.10      <none>        53/UDP,53/TCP,9153/TCP   5h5m

NAMESPACE     NAME                         DESIRED   CURRENT   READY   UP-TO-DATE   AVAILABLE   NODE SELECTOR            AGE
kube-system   daemonset.apps/calico-node   2         2         0       2            0           kubernetes.io/os=linux   5h1m
kube-system   daemonset.apps/kube-proxy    2         2         2       2            2           kubernetes.io/os=linux   5h5m

NAMESPACE                   NAME                                                    READY   UP-TO-DATE   AVAILABLE   AGE
cert-manager                deployment.apps/cert-manager                            1/1     1            1           27m
cert-manager                deployment.apps/cert-manager-cainjector                 1/1     1            1           27m
cert-manager                deployment.apps/cert-manager-webhook                    1/1     1            1           27m
inteldeviceplugins-system   deployment.apps/inteldeviceplugins-controller-manager   0/1     1            0           21m
kube-system                 deployment.apps/calico-kube-controllers                 1/1     1            1           5h1m
kube-system                 deployment.apps/coredns                                 2/2     2            2           5h5m

NAMESPACE                   NAME                                                               DESIRED   CURRENT   READY   AGE
cert-manager                replicaset.apps/cert-manager-748f4d667d                            1         1         1       27m
cert-manager                replicaset.apps/cert-manager-cainjector-6cd5988f45                 1         1         1       27m
cert-manager                replicaset.apps/cert-manager-webhook-7498ddfd98                    1         1         1       27m
[... cut ...]
  Normal   Scheduled    23m                 default-scheduler  Successfully assigned inteldeviceplugins-system/inteldeviceplugins-controller-manager-76dd86b7bd-qg6lq to server1
  Warning  FailedMount  62s (x10 over 21m)  kubelet            Unable to attach or mount volumes: unmounted volumes=[cert], unattached volumes=[cert kube-api-access-kxtw4]: timed out waiting for the condition
  Warning  FailedMount  56s (x19 over 23m)  kubelet            MountVolume.SetUp failed for volume "cert" : secret "webhook-server-cert" not found
```

mythi, 2021-12-17:
@belanasaikiran yes that is because of the error -> the pod needs the webhook certificates to run but that fails

belanasaikiran, 2021-12-17:
Is there any fix in solving this issue ? 

I am following the same documentation steps provided or do i setup the cluster from scratch again.

mythi, 2021-12-17:
> or do i setup the cluster from scratch again.. ?

the deployment is OK for me so this might be a good thing to try. perhaps with some other CNI?

belanasaikiran, 2021-12-18:
sure, I've been using Calico as my CNI. Can you suggest the best alternative for CNI? May be the one which supports/works with Intel FPGA Device Plugin.

Thank you

mythi, 2021-12-20:
> Maybe the one that supports/works with Intel FPGA Device Plugin.

there's no dependency. For some reason `cert-manager` in your cluster is not responding correctly and the FPGA deployment just needs its services.

belanasaikiran, 2021-12-20:
okay, I'll try with different versions of `cert-manager` and also with different `network-plugins`. I'll comment back if I face any issues. Thank you @mythi

belanasaikiran, 2021-12-23:
Hi @mythi, I have re-created my K8s Cluster from cluster and changed the Master to a different Machine. I am still facing the same issue. Can you help me out with this ?

Here are the logs.

```
lsai_saikiran@server1:~/Kubernetes/fpga-device-plugin/intel-device-plugins-for-kubernetes/deployments/fpga_plugin/overlays/af$ k apply -k .
customresourcedefinition.apiextensions.k8s.io/acceleratorfunctions.fpga.intel.com created
customresourcedefinition.apiextensions.k8s.io/fpgaregions.fpga.intel.com created
clusterrole.rbac.authorization.k8s.io/intelfpgaplugin-manager-role created
clusterrole.rbac.authorization.k8s.io/intelfpgaplugin-node-getter created
clusterrolebinding.rbac.authorization.k8s.io/intelfpgaplugin-get-nodes created
clusterrolebinding.rbac.authorization.k8s.io/intelfpgaplugin-manager-rolebinding created
service/intelfpgaplugin-webhook-service created
deployment.apps/intelfpgaplugin-webhook created
daemonset.apps/intelfpgaplugin-fpgadeviceplugin created
mutatingwebhookconfiguration.admissionregistration.k8s.io/intelfpgaplugin-mutating-webhook-configuration created
Error from server (InternalError): error when creating ".": Internal error occurred: failed calling webhook "webhook.cert-manager.io": failed to call webhook: Post "https://cert-manager-webhook.cert-manager.svc:443/mutate?timeout=10s": context deadline exceeded
Error from server (InternalError): error when creating ".": Internal error occurred: failed calling webhook "webhook.cert-manager.io": failed to call webhook: Post "https://cert-manager-webhook.cert-manager.svc:443/mutate?timeout=10s": context deadline exceeded
```
here's the PODS Status

```
lsai_saikiran@server1:~$ kubectl get pod -o wide --all-namespaces
NAMESPACE                 NAME                                        READY   STATUS              RESTARTS        AGE     IP               NODE      NOMINATED NODE   READINESS GATES
cert-manager              cert-manager-55658cdf68-l5sz8               1/1     Running             1 (3h50m ago)   4h42m   10.244.179.68    server2   <none>           <none>
cert-manager              cert-manager-cainjector-967788869-fl2kg     1/1     Running             6 (3h47m ago)   4h42m   10.244.179.66    server2   <none>           <none>
cert-manager              cert-manager-webhook-5d9c79595f-sl2d9       1/1     Running             0               17m     10.244.179.72    server2   <none>           <none>
intelfpgaplugin-system    intelfpgaplugin-fpgadeviceplugin-pkqr7      1/1     Running             0               16m     10.244.179.73    server2   <none>           <none>
intelfpgaplugin-system    intelfpgaplugin-fpgadeviceplugin-wxn8z      1/1     Running             0               16m     10.244.116.135   server1   <none>           <none>
intelfpgaplugin-system    intelfpgaplugin-webhook-5d9fcb76f9-sg752    0/1     ContainerCreating   0               16m     <none>           server2   <none>           <none>
intelfpgawebhook-system   intelfpgawebhook-webhook-5d9fcb76f9-xp8gn   1/1     Running             0               3h36m   10.244.179.69    server2   <none>           <none>
kube-system               calico-kube-controllers-6b9fbfff44-62jzq    1/1     Running             7 (3h46m ago)   4h43m   10.244.179.65    server2   <none>           <none>
kube-system               calico-node-dl2zs                           0/1     Running             0               4h43m   192.168.0.112    server2   <none>           <none>
kube-system               calico-node-thjwc                           0/1     Running             1 (3h47m ago)   4h43m   192.168.0.124    server1   <none>           <none>
kube-system               coredns-78fcd69978-7cnr4                    1/1     Running             1 (3h47m ago)   5h1m    10.244.116.131   server1   <none>           <none>
kube-system               coredns-78fcd69978-g92qc                    1/1     Running             1 (3h47m ago)   5h1m    10.244.116.132   server1   <none>           <none>
kube-system               etcd-server1                                1/1     Running             5 (3h47m ago)   5h1m    192.168.0.124    server1   <none>           <none>
kube-system               kube-apiserver-server1                      1/1     Running             2 (3h47m ago)   4h14m   192.168.0.124    server1   <none>           <none>
kube-system               kube-controller-manager-server1             1/1     Running             8 (3h47m ago)   5h1m    192.168.0.124    server1   <none>           <none>
kube-system               kube-proxy-d2bmp                            1/1     Running             2 (4h53m ago)   4h56m   192.168.0.112    server2   <none>           <none>
kube-system               kube-proxy-ppvzx                            1/1     Running             1 (3h47m ago)   5h1m    192.168.0.124    server1   <none>           <none>
kube-system               kube-scheduler-server1                      1/1     Running             7 (3h47m ago)   5h1m    192.168.0.124    server1   <none>           <none>
```

Describing the POD which is stuck at `ContainerCreating` State

```
lsai_saikiran@server1:~/Kubernetes/fpga-device-plugin/intel-device-plugins-for-kubernetes/deployments/fpga_plugin/overlays/af$ kubectl describe po  intelfpgaplugin-webhook-5d9fcb76f9-sg752   -n intelfpgaplugin-system 
Name:           intelfpgaplugin-webhook-5d9fcb76f9-sg752
Namespace:      intelfpgaplugin-system
Priority:       0
Node:           server2/192.168.0.112
Start Time:     Thu, 23 Dec 2021 09:39:22 +0000
[... cut ...]
  Warning  FailedMount  2m58s (x2 over 5m16s)  kubelet            Unable to attach or mount volumes: unmounted volumes=[cert], unattached volumes=[kube-api-access-f9h2m cert]: timed out waiting for the condition
  Warning  FailedMount  42s (x6 over 16m)      kubelet            Unable to attach or mount volumes: unmounted volumes=[cert], unattached volumes=[cert kube-api-access-f9h2m]: timed out waiting for the condition
  Warning  FailedMount  18s (x17 over 18m)     kubelet            MountVolume.SetUp failed for volume "cert" : secret "webhook-server-cert" not found
```

Is the problem with the `Cert-Manager` or `Webhook` ?

mythi, 2021-12-23:
@belanasaikiran The FPGA webhook certificate request from `cert-manager` fails and that's why it stays in pending state. Our FPGA validation is OK so there's probably an error in your cluster setup. Is this still with v1.6.1 or some earlier version?

mythi, 2021-12-23:
@belanasaikiran can you try:
`kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/fpga_plugin/overlays/af`

belanasaikiran, 2021-12-24:
I have just tried the above command you sent. I'm still unable to validate the `cert-manager`. The following are the logs.

```
lsai_saikiran@server1:~/Kubernetes/fpga-device-plugin/intel-device-plugins-for-kubernetes$ kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/fpga_plugin/overlays/af
customresourcedefinition.apiextensions.k8s.io/acceleratorfunctions.fpga.intel.com configured
customresourcedefinition.apiextensions.k8s.io/fpgaregions.fpga.intel.com configured
clusterrole.rbac.authorization.k8s.io/intelfpgaplugin-manager-role configured
clusterrole.rbac.authorization.k8s.io/intelfpgaplugin-node-getter unchanged
clusterrolebinding.rbac.authorization.k8s.io/intelfpgaplugin-get-nodes unchanged
clusterrolebinding.rbac.authorization.k8s.io/intelfpgaplugin-manager-rolebinding unchanged
service/intelfpgaplugin-webhook-service unchanged
deployment.apps/intelfpgaplugin-webhook unchanged
daemonset.apps/intelfpgaplugin-fpgadeviceplugin unchanged
mutatingwebhookconfiguration.admissionregistration.k8s.io/intelfpgaplugin-mutating-webhook-configuration configured
Error from server (InternalError): error when creating "https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/fpga_plugin/overlays/af": Internal error occurred: failed calling webhook "webhook.cert-manager.io": failed to call webhook: Post "https://cert-manager-webhook.cert-manager.svc:443/mutate?timeout=10s": dial tcp 10.99.72.26:443: connect: connection refused
Error from server (InternalError): error when creating "https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/fpga_plugin/overlays/af": Internal error occurred: failed calling webhook "webhook.cert-manager.io": failed to call webhook: Post "https://cert-manager-webhook.cert-manager.svc:443/mutate?timeout=10s": dial tcp 10.99.72.26:443: connect: connection refused

```

```
lsai_saikiran@server1:~$ kubectl get po -o wide -n intelfpgaplugin-system
NAME                                       READY   STATUS              RESTARTS      AGE   IP               NODE      NOMINATED NODE   READINESS GATES
intelfpgaplugin-fpgadeviceplugin-pkqr7     1/1     Running             1 (17h ago)   20h   10.244.179.81    server2   <none>           <none>
intelfpgaplugin-fpgadeviceplugin-wxn8z     1/1     Running             1 (17h ago)   20h   10.244.116.140   server1   <none>           <none>
intelfpgaplugin-webhook-5d9fcb76f9-sg752   0/1     ContainerCreating   0             20h   <none>           server2   <none>           <none>
```

I've tried both `af` and `region` modes. Are there any alternative cert-manager available for K8s. 
Could it be anything related to IPs and Ports?

belanasaikiran, 2021-12-31:
Hi @mythi , 

Some how the `cert-manager` is working correctly after recreating the cluster from scratch and deploying all the plugins on a `single node` and then joining the `worker nodes` to it.

However, as a `starter/learner`, I want to built `Init containers`  with my own **bitstreams** files. Can you please provide some documentation regarding building the containers from scratch **or** modifying the source  `Makefile` and specifying the bitstreams and `mapping collection.yaml` to it. 

Thank you.

mythi, 2022-01-03:
> Some how the cert-manager is working correctly after recreating the cluster from scratch and deploying all the
> plugins on a single node and then joining the worker nodes to it.

OK, I don't know what might be going on but it smells like some networking mis-config in your cluster.

> I want to built Init containers with my own bitstreams files

this is not how it works. I'd suggest you start with our `opae-nlb-demo` and move from there.

belanasaikiran, 2022-01-04:
Thank you @mythi.

I will start with `opae-nlb-demo` and see if I can build anything with custom `bitstreams`.

closed 2022-01-04 by belanasaikiran