intel/intel-device-plugins-for-kubernetes #852: OCP: UBI Based images for SGX plugin and operator

chaitanya1731, 2022-01-13:
According to [link](https://redhat-connect.gitbook.io/partner-guide-for-red-hat-openshift-and-container/program-on-boarding/containers-with-red-hat-universal-base-image-ubi) we need to release the UBI based SGX plugin and operator container images on the OCP platform.

- [ ] 1. Find proper UBI base image similar to Google's - gcr.io/distroless/static and know about potential security risks
- [ ] 2. How to maintain the related dockerfiles for UBI based images along with other dockerfiles and where to release these UBI based images?
- [ ] 3. Check and understand proper Licenses for UBI
- [ ] 4. Work out the UBI dockerfiles and images and test them
- [ ] 5. Start Certification process

mythi, 2022-01-14:
> 1. Find proper UBI base image similar to Google's - gcr.io/distroless/static and know about potential security risks

the smallest possible because we don't depend on the base image. The builder image needs provide Go and toybox build capabilities for the initcontainer(s). 

> 2. How to maintain the related dockerfiles for UBI based images along with other dockerfiles and where to release these UBI based images?

I guess we want the "build service" to deal with the publishing and we just provide the Dockerfile(s). Have you tried are the existing Dockerfiles suitable for that build service?

chaitanya1731, 2022-01-14:
As per RedHat - 
- The ubi-micro is the smallest possible UBI image, obtained by excluding a package manager and all of its dependencies
- This minimizes the attack surface of container images based on the ubi-micro image and is suitable for minimal applications
- Constructing a container image without the Linux distro’s packaging tools is sometimes referred to as distroless. By this criteria, UBI can be called a distroless container image.
- Size is 12Mb compressed, 38Mb uncompressed. This is the smallest possible image. So need to use the 'ubi-micro' image

chaitanya1731, 2022-01-26:
> I guess we want the "build service" to deal with the publishing and we just provide the Dockerfile(s). Have you tried are the existing Dockerfiles suitable for that build service?

Tried UBI-micro as base images and keeping builder images as it is.. This surely works as tried testing on the RedHat portal for certification. Passes all the test cases. We can just replace the final base image gcr.io/distroless with RedHat ubi-micro and add the required labels and it is good to go

chaitanya1731, 2022-01-26:
> * 3. Check and understand proper Licenses for UBI
>   Notes - UBI is release under under the terms of the UBI End User License Agreement (EULA)
>   Refer https://developers.redhat.com/articles/ubi-faq#introduction?source=sso

As per the discussion, we keep the same licenses for upstream and downstream. Hence using what upstream team already uses. Apache 2.0

PR #897 (2022-02-24): Images: Updated base image to use 'ubi8-micro' image from RedHat
Updated base image to use 'ubi8-micro' image from RedHat for all plugins.

Closes: #852
closed 2022-02-24 by bart0sh