intel/intel-device-plugins-for-kubernetes #863: admission webhook "fpga.mutator.webhooks.intel.com" denied the request: environment variable 'FPGA_AFU_1' is not allowed

belanasaikiran, 2022-01-27:
I was trying to run one of the example deployment ( `test-fpga-region.yaml` ) on my cluster with FPGAs.
Iam getting the following error when running the app after programming the devices:

```
saikiran@server1:~/Kubernetes/fpga-device-plugin/intel-device-plugins-for-kubernetes/demo$ k apply -f test-fpga-orchestrated.yaml
Error from server: error when creating "test-fpga-orchestrated.yaml": admission webhook "fpga.mutator.webhooks.intel.com" denied the request: environment variable 'FPGA_AFU_1' is not allowed
```

I have followed the following guide:
[https://asciinema.org/a/sUnLNwpazbUXKdpC66g09W5w1](https://asciinema.org/a/sUnLNwpazbUXKdpC66g09W5w1)

here's the result of programmed devices:
```
> cat /sys/class/*/*/*/afu_id                                                                                                                                                     
d8424dc4a4a3c413f89e433683f9040b                                                                                                                                                  
d8424dc4a4a3c413f89e433683f9040b 
```

please can you help me resolve this issue/errors

mythi, 2022-02-08:
@bart0sh any ideas?

rojkov, 2022-02-08:
IIRC the webhook denies new pods if they set FPGA_AFU_* (and FPGA_REGION_*) env variables. These variables are supposed to be set by the webhook itself, not a user. It sounds like `test-fpga-orchestrated.yaml` for some reason sets the variable. Though in my copy of the repo this file doesn't touch ENV at all.

What's the content of the failing YAML?

bart0sh, 2022-02-08:
@belanasaikiran Have you modified test-fpga-orchestraded.yaml somehow? Which project branch did you use for your experiments? Did you build images locally or downloaded them from the dockerhub?

belanasaikiran, 2022-02-08:
Thank you @mythi @rojkov @bart0sh  for replying,

@rojkov, I haven't set any variables with `user`, I have followed the user guide. Here is the content of the file `test-fpga-orchestrated.yaml`

```
apiVersion: v1
kind: Pod
metadata:
  name: test-fpga-orchestrated
spec:
  containers:
  - name: test-container
    image: intel/opae-nlb-demo:devel
    imagePullPolicy: IfNotPresent
    securityContext:
      capabilities:
        add:
          [IPC_LOCK]
    resources:
      limits:
        fpga.intel.com/arria10.dcp1.2-nlb3-orchestrated: 1
        cpu: 1
        hugepages-2Mi: 20Mi

  restartPolicy: Never
```
Is there anything I need to change in this file ?

@bart0sh, I haven't modified the contents of `test-fpga-orchestrated.yaml`.
I am using the current main branch for testing and I have build images locally as per the instructions provided in Documentation (haven't modified).

Please can you help me resolving the issue so.

bart0sh, 2022-02-16:
@rojkov @mythi Similar or the same issue just happened in our CI: https://github.com/intel/intel-device-plugins-for-kubernetes/runs/5206716986?check_suite_focus=true

Any thoughts on what the reason could be?

mythi, 2022-02-16:
> Any thoughts on what the reason could be?

@bart0sh the pod get mutated twice and we error because of an earlier mutation? Looking at the code I wonder if FPGA patcher is idempotent...

bart0sh, 2022-02-16:
interesting. Should it be idempotent? @rojkov WDYT?

mythi, 2022-02-16:
@bart0sh yes: https://kubernetes.io/docs/reference/access-authn-authz/extensible-admission-controllers/#idempotence

mythi, 2022-02-16:
@rojkov there were also other mutation errors in the e2e logs, not just this one.

`admission webhook "fpga.mutator.webhooks.intel.com" denied the request: no such resource: "fpga.intel.com/region-ce48969398f05f33946d560708be108a"`

rojkov, 2022-02-17:
@mythi The "no such resource" error is expected when mappings haven't been populated at the moment of mutation. Could be a result of improper test env setup.

mythi, 2022-02-21:
@belanasaikiran can you verify using the latest `:devel` image?

belanasaikiran, 2022-02-22:
I have re-tested with the latest changes to Git repo. 

The environment variable errors are gone but now I am facing the following error while deploying the `intel-device-plugins-for-kuberenetes/demo/test-fpga-orchestrated.yaml`  file

```
server2@server2:~/intel-device-plugins-for-kubernetes/demo$ k apply -f test-fpga-orchestrated.yaml 
Error from server: error when creating "test-fpga-orchestrated.yaml": admission webhook "fpga.mutator.webhooks.intel.com  denied the request: no such resource: "fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6"
```

I have check the `mapping-collection.yaml` and verified the Interface ID exists. 

It denies with `no such resource: "fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6"`

rojkov, 2022-02-22:
Hm... So, it's the same thing here again - idempotency of the webhook. It used to be not a problem at the time the webhook was developed, but it is a problem now.

I think this particular error can be solved with adding an identity mapping resource ("fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6" -> "fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6").

mythi, 2022-05-02:
@belanasaikiran can you confirm everything is working now?

belanasaikiran, 2022-05-02:
Hi @mythi, I have checked and verified everything in the cluster  is working as expected with the latest changes to plugin. 

Thank you for you and the team :)

PR #892 (2022-02-18): fpga_webhook: make patcher idempotent
Webhook's patcher currently can't be re-applied to its output because it
rejects containers with the env vars it adds (e.i. FPGA_AFU_* and
FPGA_REGION_*).

Instead of rejecting container with disallowed env vars just reset them
first upon patching.

Closes #863