intel/intel-device-plugins-for-kubernetes #868: intelgpu-job image intel/intel-gpu-initcontainer accessed denied

vmcclurem, 2022-01-31:
The intel/intel-gpu-initcontainer referenced in intelgpu-job.yaml fails to deploy container.

"reading manifest latest in quay.io/intel/intel-gpu-initcontainer: unauthorized: access to the requested resource is not authorized"

Reproduction steps:

1) Base cluster 2x Ubuntu 20.04 (1 VM (control plane), 1 Coffee Lake NUC (worker node))
2) Setup Kubernetes 0.23.0, CRI-O 1.23.0, Docker 20.10.7
3) Setup worker node

  worker node # apt install intel-opencl-icd clinfo

4) Clone the repo

  control plane $ git clone https://github.com/intel/intel-device-plugins-for-kubernetes intel

5) Create the intelgpu job

   control plane $ cd intel
   control plane $ kubectl create -f demo/intelgpu-job.yaml

Setting the image name in intelgpu-job.yaml to "intel/intel-gpu-initcontainer:0.23.0" fixes the issue

mythi, 2022-01-31:
> "reading manifest latest in quay.io/intel/intel-gpu-initcontainer: unauthorized: access to the requested resource is not authorized"

`intelgpu-job.yaml` does not have a reference to initcontainer. How did it go there for you? Getting quay.io used instead of docker.io is probably a miss in the CRI-O config (no "docker.io" in unqualified-search-registries or something)

vmcclurem, 2022-01-31:
User-error. Sorry for the inconvenience.

closed 2022-01-31 by vmcclurem