intel/intel-device-plugins-for-kubernetes #870: Unable to deploy container for intelgpu-job

vmcclurem, 2022-02-01:
The "intelopencl/intel-opencl:ubuntu-20.10-ppa" referenced in intelgpu-job.yaml fails to deploy container. Output is as follows:

  reading manifest ubuntu-20.10-ppa in docker.io/intelopencl/intel-opencl: errors:
  denied: requested access to the resource is denied
  unauthorized: authentication required"

Reproduction steps:

1) Base cluster 2x Ubuntu 20.04 (1 VM (control plane), 1 Coffee Lake NUC (worker node))
2) Setup Kubernetes 0.23.0, CRI-O 1.23.0, Docker 20.10.7
3) Setup worker node

  worker node # apt install intel-opencl-icd clinfo

4) # docker login docker.io
5) Edit /etc/containers/registries.conf to remove "quay.io" from the unqualified-search-registries
6) # systemctl restart crio
7) Clone the repo

  control plane $ git clone https://github.com/intel/intel-device-plugins-for-kubernetes intel

8) Create the intelgpu job

  control plane $ cd intel
  control plane $ kubectl create -f demo/intelgpu-job.yaml

9) Get the pods and describe the intelgpu-demo-job pod

   control plane $ kubectl get pods
   control plane $ kubectl describe pod intelgpu-demo-job...

mythi, 2022-02-07:
> reading manifest ubuntu-20.10-ppa in docker.io/intelopencl/intel-opencl: errors:

looks like the image does not exist anymore. we'll need to fix the yaml to use some other image.

hj-johannes-lee, 2022-02-07:
@bart0sh 

Minimize admission of root containers (https://docs.bridgecrew.io/docs/bc_k8s_22)
- /demo/test-fpga-preprogrammed.yaml
- /demo/test-fpga-orchestrated.yaml
- /demo/intelfpga-job.yaml
- /demo/intelvpu-job.yaml
- ... total 81 files

Ensure containers do not share host process ID namespace (https://docs.bridgecrew.io/docs/bc_k8s_16)
Ensure containers do not share host IPC namespace (https://docs.bridgecrew.io/docs/bc_k8s_17)
- /deployments/gpu_plugin/overlays/namespace_kube-system/add-namespace-kube-system.yaml
- /deployments/dsa_plugin/overlays/namespace_kube-system/add-namespace-kube-system.yaml
- /deployments/vpu_plugin/overlays/namespace_kube-system/add-namespace-kube-system.yaml

Ensure containers do not share the host network namespace (https://docs.bridgecrew.io/docs/bc_k8s_18)
- /deployments/gpu_plugin/overlays/namespace_kube-system/add-namespace-kube-system.yaml
- /deployments/dsa_plugin/overlays/namespace_kube-system/add-namespace-kube-system.yaml
- /deployments/vpu_plugin/overlays/namespace_kube-system/add-namespace-kube-system.yaml
- /deployments/sgx_enclave_apps/overlays/sgx_ecdsa_inproc_quote/add_hostnetwork.yaml
- /deployments/sgx_aesmd/base/kustomization.yaml
- /deployments/sgx_enclave_apps/overlays/sgx_ecdsa_inproc_quote/kustomization.yaml
- /deployments/sgx_aesmd/base/intel-sgx-aesmd.yaml

Ensure containers do not run with AllowPrivilegeEscalation (https://docs.bridgecrew.io/docs/bc_k8s_19)
- /deployments/sgx_admissionwebhook/manager/manager.yaml
- /deployments/sgx_admissionwebhook/base/manager_webhook_patch.yaml
- /deployments/sgx_plugin/overlays/epc-nfd/add-epc-nfd-initcontainer.yaml
- ... total 102 files

bart0sh, 2022-02-07:
@hj-johannes-lee thanks. From the first look all findings make sense and worth looking at.
@mythi wdyt?

mythi, 2022-02-07:
@bart0sh @hj-johannes-lee wrong issue to have this conversation?

PR #903 (2022-02-24): demo: add intel-opencl-icd
Fixes: #870
closed 2022-02-24 by bart0sh