intel/intel-device-plugins-for-kubernetes #872: Add devices to NFD

bart0sh, 2022-02-01:
Instead of providing [NFD labelling rules](https://github.com/intel/intel-device-plugins-for-kubernetes/pull/871) we can add all our devices to NFD, so NFD would support them out of the box. 
This would simplify user experience as user would just need to install NFD to perform node labelling.

mythi, 2022-02-07:
NFD does not maintain user configurations. We can have NFD installation and the rules creation in the same place so the user experience is not that different.

/cc @marquiz

marquiz, 2022-02-07:
Yeah, we've been trying to keep vendor and use-case specific logic and all kinds of "device databases" etc out of NFD. I believe deployment of NFD labeling rules should now be easy enough as that can be done with just dropping one CR. So you could include e.g. the CR as part of the device plugin deployment(?)

One thing that is a bit cumbersome is enabling custom label namespaces (like `intel.com/<gpu-label>`). NFD now allows sub-namespaces of the default ns so you could use something like `intel.feature.node.kubernetes.io/<device-label>` in order to avoid customized NFD deployment.

mythi, 2022-02-07:
> One thing that is a bit cumbersome is enabling custom label namespaces

Good that you mentioned it. We need a customized deployment ([SGX](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/deployments/sgx_nfd/master-args.yaml#L3) so we could also look into clean labels like `sgx.intel.com: true` or something like that.

bart0sh, 2022-02-07:
@marquiz thanks for the suggestion to use intel.feature.node.kubernetes.io/<device-label> format. I'll update #871 to use it.

closed 2022-02-17 by bart0sh