intel/intel-device-plugins-for-kubernetes #886: doc: missing vmlinux while building dlb2 kernel module

ttzeng, 2022-02-14:
Fail to build the `dlb2.ko` kernel module by following [this paragraph](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/dlb_plugin/README.md#dlb2-driver-configuration-for-pfs) in the DLB plugin README.md. The error message says it requires `vmlinux` to build the kernel module, so perhaps some prerequisites or instructions are missing in the documentation?

ttzeng, 2022-02-15:
@bart0sh thank you for looking into this. I tried to build the kernel module natively on an Eagle Stream system installed with Ubuntu 21.10 distro. Here are the build log:
```
$ make
make -C /lib/modules/`uname -r`/build M=/home/sdp/Workspace/dlb/driver/dlb2 modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-28-generic'
  CC [M]  /home/sdp/Workspace/dlb/driver/dlb2/dlb2_main.o
  CC [M]  /home/sdp/Workspace/dlb/driver/dlb2/dlb2_file.o
  CC [M]  /home/sdp/Workspace/dlb/driver/dlb2/dlb2_ioctl.o
  CC [M]  /home/sdp/Workspace/dlb/driver/dlb2/base/dlb2_resource.o
  CC [M]  /home/sdp/Workspace/dlb/driver/dlb2/dlb2_pf_ops.o
  CC [M]  /home/sdp/Workspace/dlb/driver/dlb2/dlb2_intr.o
  CC [M]  /home/sdp/Workspace/dlb/driver/dlb2/dlb2_dp.o
  CC [M]  /home/sdp/Workspace/dlb/driver/dlb2/dlb2_sriov.o
  CC [M]  /home/sdp/Workspace/dlb/driver/dlb2/dlb2_vf_ops.o
  CC [M]  /home/sdp/Workspace/dlb/driver/dlb2/dlb2_vdcm.o
  LD [M]  /home/sdp/Workspace/dlb/driver/dlb2/dlb2.o
  MODPOST /home/sdp/Workspace/dlb/driver/dlb2/Module.symvers
  CC [M]  /home/sdp/Workspace/dlb/driver/dlb2/dlb2.mod.o
  LD [M]  /home/sdp/Workspace/dlb/driver/dlb2/dlb2.ko
  BTF [M] /home/sdp/Workspace/dlb/driver/dlb2/dlb2.ko
Skipping BTF generation for /home/sdp/Workspace/dlb/driver/dlb2/dlb2.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-28-generic'
sdp@b49691a766f0:~/Workspace/dlb/driver/dlb2

$ sudo insmod dlb2.ko
insmod: ERROR: could not insert module dlb2.ko: Unknown symbol in module
```

mythi, 2022-02-15:
@ttzeng 

> `$ sudo insmod dlb2.ko`
> `insmod: ERROR: could not insert module dlb2.ko: Unknown symbol in module`

check with `dmesg` what symbols are missing. it's probably enough to just `modprobe` the dependent symbols.

ttzeng, 2022-02-15:
@mythi Thanks for the tips, here are the missing symbols:
```
[  203.625497] dlb2: module verification failed: signature and/or required key missing - tainting kernel
[  203.625946] dlb2: Unknown symbol mdev_bus_type (err -2)
[  203.626015] dlb2: Unknown symbol mdev_parent_dev (err -2)
[  203.626130] dlb2: Unknown symbol mdev_register_device (err -2)
[  203.626166] dlb2: Unknown symbol mdev_unregister_device (err -2)
[  284.343989] dlb2: Unknown symbol mdev_bus_type (err -2)
[  284.344067] dlb2: Unknown symbol mdev_parent_dev (err -2)
[  284.344177] dlb2: Unknown symbol mdev_register_device (err -2)
[  284.344210] dlb2: Unknown symbol mdev_unregister_device (err -2)
[  526.967095] dlb2: Unknown symbol mdev_bus_type (err -2)
[  526.967175] dlb2: Unknown symbol mdev_parent_dev (err -2)
[  526.967287] dlb2: Unknown symbol mdev_register_device (err -2)
[  526.967320] dlb2: Unknown symbol mdev_unregister_device (err -2)
[  546.502246] dlb2: Unknown symbol mdev_bus_type (err -2)
[  546.502325] dlb2: Unknown symbol mdev_parent_dev (err -2)
[  546.502432] dlb2: Unknown symbol mdev_register_device (err -2)
[  546.502465] dlb2: Unknown symbol mdev_unregister_device (err -2)
```
After loading the `mdev` kernel module, though the `dlb2.ko`  can be inserted, no `/dev/dlb*` nodes are registered, and the `dmesg`  indicates the following errors:
```
[10173.501432] dlb2 0000:6d:00.0: [dlb2_probe()] AER is not supported
[10173.607977] dlb2 0000:6d:00.0: [dlb2_vdcm_init()] Failed to enable aux domains
[10173.608857] dlb2: probe of 0000:6d:00.0 failed with error -22
[10173.608921] dlb2 0000:72:00.0: [dlb2_probe()] AER is not supported
[10173.715527] dlb2 0000:72:00.0: [dlb2_vdcm_init()] Failed to enable aux domains
[10173.716271] dlb2: probe of 0000:72:00.0 failed with error -22
[10173.716329] dlb2 0000:77:00.0: [dlb2_probe()] AER is not supported
[10173.823506] dlb2 0000:77:00.0: [dlb2_vdcm_init()] Failed to enable aux domains
[10173.824236] dlb2: probe of 0000:77:00.0 failed with error -22
[10173.824291] dlb2 0000:7c:00.0: [dlb2_probe()] AER is not supported
[10173.931504] dlb2 0000:7c:00.0: [dlb2_vdcm_init()] Failed to enable aux domains
[10173.932246] dlb2: probe of 0000:7c:00.0 failed with error -22
[10173.932356] dlb2 0000:ea:00.0: [dlb2_probe()] AER is not supported
[10174.040082] dlb2 0000:ea:00.0: [dlb2_vdcm_init()] Failed to enable aux domains
[10174.040961] dlb2: probe of 0000:ea:00.0 failed with error -22
[10174.041035] dlb2 0000:ef:00.0: [dlb2_probe()] AER is not supported
[10174.147523] dlb2 0000:ef:00.0: [dlb2_vdcm_init()] Failed to enable aux domains
[10174.148298] dlb2: probe of 0000:ef:00.0 failed with error -22
[10174.148363] dlb2 0000:f4:00.0: [dlb2_probe()] AER is not supported
[10174.255495] dlb2 0000:f4:00.0: [dlb2_vdcm_init()] Failed to enable aux domains
[10174.256265] dlb2: probe of 0000:f4:00.0 failed with error -22
[10174.256332] dlb2 0000:f9:00.0: [dlb2_probe()] AER is not supported
[10174.363506] dlb2 0000:f9:00.0: [dlb2_vdcm_init()] Failed to enable aux domains
[10174.364255] dlb2: probe of 0000:f9:00.0 failed with error -22
```

mythi, 2022-02-15:
>  though the `dlb2.ko` can be inserted, no `/dev/dlb*` nodes are registered, 

@hj-johannes-lee do you remember seeing these errors?

hj-johannes-lee, 2022-02-16:
This issue seems not what we need to fix.
We will just fix README soon.

closed 2022-02-16 by hj-johannes-lee