intel/intel-device-plugins-for-kubernetes #891: intel xe integrated gpu also supported?

MichelBr, 2022-02-16:
Hello,

I'm currently running the 0.19-version, as that is what my kubernetes distribution provides by default. However, I can't see the GPU. For machine learning workloads, it would be nice to use the integrated Xe GPU of the Tigerlake (NUC Pro) processor.

I saw support for io-srv ..., but is this supported or going to be supported.

Please :), while I wait on the discrete gpu's :).

Thanks,

Michel

MichelBr, 2022-02-16:
My plex-server would also appreciate it now and then. And yes, plex on kubernets is a thing :) (Truenas)

mythi, 2022-03-09:
> > However, I can't see the GPU.
> 
> @MichelBr you can't see the `gpu.intel.com` resources or your GPU does not show up on your host?
> 
> Please post the `intel-gpu-plugin` logs with debug log level.

@MichelBr Closing due to inactivity. Please re-open if the problem is still valid.

closed 2022-03-09 by mythi