intel/intel-device-plugins-for-kubernetes #904: opae-nlb-demo image contains vulnerable json-c 0.13.1

okartau, 2022-02-24:
Vulnerability details in [CVE-2020-12762](https://nvd.nist.gov/vuln/detail/CVE-2020-12762)

msivosuo, 2022-03-03:
@bart0sh - can you take a look.

bart0sh, 2022-03-03:
@msivosuo please schedule it to the milestone according to its importance

bart0sh, 2022-04-19:
@okartau @mythi It looks like the fix was actually not needed as [CVE-2020-12762](https://nvd.nist.gov/vuln/detail/CVE-2020-12762) was fixed in [json-c 0.13.1-12fc32](https://lists.fedoraproject.org/archives/list/package-announce@lists.fedoraproject.org/message/CQQRRGBQCAWNCCJ2HN3W5SSCZ4QGMXQI/) and the image included [json-c 0.13.1-13.fc32](https://github.com/intel/intel-device-plugins-for-kubernetes/runs/6071804514?check_suite_focus=true)

mythi, 2022-04-19:
Hmm. The CVE and the fix are from 2020 but the tool "found" it in 2022...

bart0sh, 2022-04-19:
yep. should we close this?

mythi, 2022-04-19:
I'd propose we close this via #971 to get the scanning tool happy. Keeping the existing version would require continuous manual acks in the tool.

closed 2022-04-19 by mythi

okartau, 2022-04-20:
scanning tool is still not happy.
Does merged status mean scanner should see updated/fixed image now? The tool still reports libjson-c-4 in the image. I am not sure which version exactly is tested, but at least how it refers to image has changed: was (about 1 day ago):
opae-nlb-demo-devel-sha256:7b6582766c8fc20b5501414b5796ba1b5f2fd01a2f65f1198909a788d8548b9e-local-content.tar
The latest job today:
opae-nlb-demo-0.23.0-sha256:bf1e81d7c33edae1f7da2e2e4e52d0a19ad3439b0d32707c1f421d1b7bd2f715-local-content.tar
Both scans find this in the image:
./local/lib/libjson-c.so.4.0.0
Is this expected and something else is needed (release or something) so that scanner can see updated image?

mythi, 2022-04-20:
we have scan results for `local/lib/libjson-c.so.5.0.0` too but looks like 0.14 in Fedora is not enough and the tool is not able to detect patching

okartau, 2022-04-20:
I just realized the diff in images names comes from job params, tag was "devel" in 1st one and in other case tag was 0.23.0, so nothing interesting in that finding

okartau, 2022-04-20:
I also located a scan job of image from "devel" with libjson.5. In that the tool claims to find json-c 0.14

mythi, 2022-04-20:
> I also located a scan job of image from "devel" with libjson.5. In that the tool claims to find json-c 0.14

yes and the CVE is in "up to 0.14 (including)" but the tool does not know it's a patched version

bart0sh, 2022-04-21:
I'll try to switch this image to just released Ubuntu 22.04. It has json-c 0.15: https://packages.ubuntu.com/jammy/libjson-c5

commit 6bac1200 (2022-04-19): opae-nlb-demo: use libjson-c 5.0

Getting rid of vulnerable libjson version should fix
vulnerability CVE-2020-12762

Fixes: #904