intel/intel-device-plugins-for-kubernetes #919: DSA container failed due to operation not permitted

yengliong, 2022-03-15:
Hi, I am trying to run DSA plugin on my Ubuntu using kubectl command below. 
`kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/dsa_plugin?ref=main
`

The command deploys the daemonset but the pod is in **CreateContainerError** status. 

NAME                           READY   STATUS                 RESTARTS        AGE
intel-dsa-plugin-d4qgj   0/1     CreateContainerError   0               4s

By checking on the pod message, there is an error mentioning about operation not permitted. 

[image]

Any ideas?

mythi, 2022-03-15:
> Any ideas?

Is the `idxd` module loaded?

yengliong, 2022-03-15:
Thank you for pointing that out, I didn't load idxd module.
Can I know how to get it?

mythi, 2022-03-15:
@yengliong it's part of mainline kernel, we have been using 5.15+ kernels.

yengliong, 2022-03-15:
After loading idxd driver, the container is running.
I want to verify the plugin with command below. 
`kubectl get nodes -o go-template='{{range .items}}{{.metadata.name}}{{"\n"}}{{range $k,$v:=.status.allocatable}}{{"  "}}{{$k}}{{": "}}{{$v}}{{"\n"}}{{end}}{{end}}' | grep '^\([^ ]\)\|\(  dsa\)'`

The output only shows me **node1**.  
I didn't see the information like _dsa.intel.com/wq-user-dedicated_ or _dsa.intel.com/wq-user-shared_.
Btw, my kernel is 5.11.0-16. Maybe the kernel version causes the issue?

mythi, 2022-03-15:
> I didn't see the information like dsa.intel.com/wq-user-dedicated or dsa.intel.com/wq-user-shared.

`idxd` does not create any of these automatically but they must be configured using `accel-config`. We provide an initcontainer that provisions a default configuration. DSA README documents it.

mythi, 2022-03-15:
@yengliong OK to close?

yengliong, 2022-03-15:
Hi @mythi Yes, please go ahead to close it. I will go through the DSA README, thank you :)

mythi, 2022-03-15:
@yengliong Thanks! Please re-open or submit another issue if you run into problems.

closed 2022-03-15 by mythi