intel/intel-device-plugins-for-kubernetes #955: build: make intel-idxd-config-initcontainer failure

ttzeng, 2022-04-03:
Fail to build `intel-idxd-config-initcontainer` with the following error:
```
Step 1/18 : ARG FINAL_BASE
Step 2/18 : FROM debian:unstable-slim AS builder
unstable-slim: Pulling from library/debian
Digest: sha256:4194b0c0134addd46bea437c8094a53131da2b17cb79d05c0fe9f365e5338288
Status: Image is up to date for debian:unstable-slim
 ---> d39e7782c1b2
Step 3/18 : RUN echo "deb-src http://deb.debian.org/debian unstable main" >>         /etc/apt/sources.list.d/deb-src.list &&     apt update && apt install -y --no-install-recommends         gcc make patch autoconf automake libtool pkg-config         libjson-c-dev uuid-dev curl ca-certificates
 ---> Running in 77b7f3670c4a

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Get:1 http://deb.debian.org/debian unstable InRelease [165 kB]
Err:1 http://deb.debian.org/debian unstable InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9
Reading package lists...
W: http://deb.debian.org/debian/dists/unstable/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/debian-archive-bullseye-automatic.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://deb.debian.org/debian/dists/unstable/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/debian-archive-bullseye-security-automatic.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://deb.debian.org/debian/dists/unstable/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/debian-archive-bullseye-stable.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://deb.debian.org/debian/dists/unstable/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/debian-archive-buster-automatic.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://deb.debian.org/debian/dists/unstable/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/debian-archive-buster-security-automatic.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://deb.debian.org/debian/dists/unstable/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/debian-archive-buster-stable.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://deb.debian.org/debian/dists/unstable/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/debian-archive-stretch-automatic.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://deb.debian.org/debian/dists/unstable/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/debian-archive-stretch-security-automatic.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://deb.debian.org/debian/dists/unstable/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/debian-archive-stretch-stable.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: GPG error: http://deb.debian.org/debian unstable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9
E: The repository 'http://deb.debian.org/debian unstable InRelease' is not signed.
The command '/bin/sh -c echo "deb-src http://deb.debian.org/debian unstable main" >>         /etc/apt/sources.list.d/deb-src.list &&     apt update && apt install -y --no-install-recommends         gcc make patch autoconf automake libtool pkg-config         libjson-c-dev uuid-dev curl ca-certificates' returned a non-zero code: 100
Makefile:176: recipe for target 'intel-idxd-config-initcontainer' failed
make: *** [intel-idxd-config-initcontainer] Error 100
```

mythi, 2022-04-04:
@ttzeng this looks like a temporary glitch. `intel-vpu-plugin` built OK for me this morning. Are you able to reproduce today?

ttzeng, 2022-04-04:
> @ttzeng this looks like a temporary glitch. `intel-vpu-plugin` built OK for me this morning. Are you able to reproduce today?

@mythi , I am still able to reproduce the error while building `intel-vpu-plugin` on my Ubuntu 16.04 with Docker version 18.09.7.

mythi, 2022-04-05:
@ttzeng I tested again and both build OK for me right now.

ttzeng, 2022-04-06:
It looks like this issue only observed on my Ubuntu 16.04 system, I can build all plugin images successfully on a Ubuntu 20.04 system, so I'm closing this for now.

closed 2022-04-06 by ttzeng