intel/intel-device-plugins-for-kubernetes #961: doc: [fpga] info on the Demo3 OpenCL demo are missing

ttzeng, 2022-04-09:
The demo [readme.md](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/demo/readme.md) mentioned 3 demos for the FPGA plugin, but there is no screencast file nor the associated resource files in the repo for replicating the Demo3 (OpenCL).

mythi, 2022-04-11:
The video was contributed without the script so we don't have it. Is this issue about removing the video or about creating the script?

ttzeng, 2022-04-11:
> The video was contributed without the script so we don't have it. Is this issue about removing the video or about creating the script?

It's up to your team to decide whether to make it consistent as we have provided screencast files for some of demos, and whether we'd like users to be able to replicate the demo using the provided screencast files as few files used in the screencast may be missing?

bart0sh, 2022-04-13:
@linjiach Can you provide screencast script and image building instructions for the OpenCL demo or just point us out to where to get those, please?

bart0sh, 2022-04-13:
@linjiach I was only able to find [demo image on the dockerhub](https://hub.docker.com/r/linjiach/fpga-opencl-centos) and [your set of repos on the github](https://github.com/linjiach?tab=repositories) However, there is no mentioning of fpga-opencl-centos there. Can you help?

linjiach, 2022-04-13:
hey @bart0sh hope you're doing well! It's been a while but I was able to see that the image is on my dockerhub: https://hub.docker.com/repository/docker/linjiach/fpga-opencl-centos lmk if you can access it.

linjiach, 2022-04-13:
I don't work at Intel anymore so I don't have access to the internal git lab. But if they're still using the SSE gitlab, then it should be in that repo. @eadamsintel should be able to point you to it if you have trouble!

eadamsintel, 2022-04-13:
@linjiach I had to migrate the demo to a new site and will send the link separately.

belanasaikiran, 2022-04-29:
hi @eadamsintel, any update on where I can find the `OpenCL demo video` and `Image Building` Instructions?

eadamsintel, 2022-04-29:
@belanasaikiran It is on an internal Intel repo.  I don't know anything about this and whether it can be made public or not. I gave access to @ttzeng so I would follow up with him on whether it can be posted here.

belanasaikiran, 2022-04-30:
@eadamsintel  @ttzeng, please do let us know if that can be made public. Any basic `hello-world` OpenCL based Container Building Instructions would really help us to start creating custom OpenCL based Containers.

Thank you.

closed 2024-02-07 by mythi