intel/intel-device-plugins-for-kubernetes #968: doc: [sgx] theoretically, the screencast of the SGX demo is not replicable

ttzeng, 2022-04-16:
Though I don't have the hardware platform to try out, but it looks like some files are missing which prevents `screencast-sgx.sh` from running successfully. Specifically:
* In screen6, I know users can save the docker images in local registry to files, but this must be done in advance, otherwise there will be no `sgx-aesmd.tar` and `sgx-demo.tar` to be loaded.

mythi, 2022-05-05:
@dbolshak thanks for the detailed feedback. I haven't paid too much attention to the script(s) it but looks like they are useful to many. I'll get at least this issue fixed at some point.

Is there something blocking you to set things up?

mythi, 2022-05-06:
> Unfortunately, I didn't notice at the beginning that it requires CPU with Flexible Launch Control support (or it's not mentioned in
>  docs), so now I am awaiting another server to repeat all steps.

Yeah, this is the requirement in the upstream driver. Our plugin only recognizes the device nodes provided by that driver. FLC is mentioned in the very first sentence of our README.

> But I have a question which is not directly related to my issues in first message. Could you please point me to documentation
> which covers remote attestation of an application running inside a pod on SGX aware k8s cluster using sgx-caching-service, 
> or it's not kubernetes specific?

It's not kubernetes specific as such but we have some helpers that could be used. "PCCS" normally runs somewhere in the datacenter and serves connections from the sgx quote provider lib ("default" is provided by Intel, Azure has "dcap client" for example). This lib needs to be configured to use the network address of that PCCS. The screencast and our sample deployments give examples how to configure the provider lib in case it's used by "aesmd" (for "out-of-proc" quote generation) and the app itself (for "in-proc" quote generation).

In the screencast I have PCCS (sgx-caching-service) in a single-node k8s cluster serving localhost connections so the pods just use `HostNetwork: true` but normally you'd have PCCS running somewhere else. 

DCAP attestation overview doc is [here](https://download.01.org/intel-sgx/sgx-dcap/1.13/linux/docs/DCAP_ECDSA_Orientation.pdf).

dbolshak, 2022-05-11:
`lsmod | grep sgx` gives the following
```
intel_sgx              32768  0
```

dbolshak, 2022-05-11:
but empty output in 
`cat /boot/config-5.4.0-109-generic | grep -i sgx`

mythi, 2022-05-11:
> but empty output in `cat /boot/config-5.4.0-109-generic | grep -i sgx`

that's the problem. quick fix is to `kubectl edit nodefeaturerule intel-dp-devices` and drop that `kernel.config` match.

dbolshak, 2022-05-11:
Thanks for quick response and provided w/around, I will test it.

But I see that I need kernel version 5.11 or above, so after testing w/around I will check full fix

dbolshak, 2022-05-12:
@mythi 

Hello,

I feel myself very terrible for distrusting you again and again. But I am so close to success!

My current problem appears on the following step
```
kubectl apply -f https://raw.githubusercontent.com/intel/intel-device-plugins-for-kubernetes/main/deployments/operator/samples/deviceplugin_v1_sgxdeviceplugin.yaml -n sgx-ecdsa-quote
```
So, as the success of it, I expect to see `sgxdeviceplugin-sample` pod in `sgx-ecdsa-quote` namespace. But I don't.

Of course I have the desired namespace and all necessary labels on my node
```
kubectl describe node | grep -i sgx
                    feature.node.kubernetes.io/cpu-sgx.enabled=true
                    intel.feature.node.kubernetes.io/sgx=true
                    nfd.node.kubernetes.io/extended-resources: sgx.intel.com/epc
  sgx.intel.com/enclave:    10
  sgx.intel.com/epc:        2126512128
  sgx.intel.com/provision:  10
  sgx.intel.com/enclave:    10
  sgx.intel.com/epc:        2126512128
  sgx.intel.com/provision:  10
  inteldeviceplugins-system   intel-sgx-plugin-4pdrg                                    0 (0%)        0 (0%)      0 (0%)           0 (0%)         58m
  sgx.intel.com/enclave    0           0
  sgx.intel.com/epc        0           0
  sgx.intel.com/provision  0           0
```

What I've done to debug so far.
I checked the deployed sgxdeviceplugin, and it looks by the following way
```
kubectl describe sgxdeviceplugin
Name:         sgxdeviceplugin-sample
Namespace:    
Labels:       <none>
Annotations:  <none>
API Version:  deviceplugin.intel.com/v1
Kind:         SgxDevicePlugin
Metadata:
  Creation Timestamp:  2022-05-12T15:07:17Z
  Generation:          1
  Managed Fields:
    API Version:  deviceplugin.intel.com/v1
    Fields Type:  FieldsV1
    fieldsV1:
      f:metadata:
        f:annotations:
          .:
          f:kubectl.kubernetes.io/last-applied-configuration:
      f:spec:
        .:
        f:enclaveLimit:
        f:image:
        f:initImage:
        f:logLevel:
        f:nodeSelector:
          .:
          f:intel.feature.node.kubernetes.io/sgx:
        f:provisionLimit:
    Manager:      kubectl-client-side-apply
    Operation:    Update
    Time:         2022-05-12T15:07:17Z
    API Version:  deviceplugin.intel.com/v1
    Fields Type:  FieldsV1
    fieldsV1:
      f:status:
        .:
        f:controlledDaemonSet:
          .:
          f:apiVersion:
          f:kind:
          f:name:
          f:namespace:
          f:resourceVersion:
          f:uid:
        f:desiredNumberScheduled:
        f:nodeNames:
        f:numberReady:
    Manager:         intel_deviceplugin_operator
    Operation:       Update
    Subresource:     status
    Time:            2022-05-12T15:07:18Z
  Resource Version:  5710
  UID:               ceda107e-4f6b-44a0-8c38-13bc05e85567
Spec:
  Enclave Limit:  10
  Image:          intel/intel-sgx-plugin:0.24.0
  Init Image:     intel/intel-sgx-initcontainer:0.24.0
  Log Level:      4
  Node Selector:
    intel.feature.node.kubernetes.io/sgx:  true
  Provision Limit:                         10
Status:
  Controlled Daemon Set:
    API Version:             apps/v1
    Kind:                    DaemonSet
    Name:                    intel-sgx-plugin
    Namespace:               inteldeviceplugins-system
    Resource Version:        5688
    UID:                     98078e21-abf7-4059-b761-d91fb51412f6
  Desired Number Scheduled:  1
  Node Names:
    euclid-4
  Number Ready:  1
Events:          <none>
```
I see no events here, also there are no any issues with controlled daemon set. Pod `intel-sgx-plugin-4pdrg` in `inteldeviceplugins-system` namespace works fine.

But logs of an operator pod (pod `inteldeviceplugins-controller-manager` in `inteldeviceplugins-system` namespace) have the following lines
```
I0512 15:01:28.422447       1 reconciler.go:231] "intel-device-plugins-manager/controller/sgxdeviceplugin: " reconciler group="deviceplugin.intel.com" reconciler kind="SgxDevicePlugin" name="sgxdeviceplugin-sample" namespace="" ="(MISSING)"
I0512 15:01:28.503111       1 reconciler.go:231] "intel-device-plugins-manager/controller/sgxdeviceplugin: " reconciler group="deviceplugin.intel.com" reconciler kind="SgxDevicePlugin" name="sgxdeviceplugin-sample" namespace="" ="(MISSING)"
```
It's not clear why namespace is empty. And I see another issues with namespace (probably they are both related to each other), if I removed device plugin by `kubectl delete -f https://raw.githubusercontent.com/intel/intel-device-plugins-for-kubernetes/main/deployments/operator/samples/deviceplugin_v1_sgxdeviceplugin.yaml -n sgx-ecdsa-quote` I see the following warning `warning: deleting cluster-scoped resources, not scoped to the provided namespace`, but it does not effect result of the operation and plugin is going to be deleted without any issues.

Only hope on your help. 

I use v0.24.0 and my kuber cluster `v1.24.0`

mythi, 2022-05-12:
> It's not clear why namespace is empty

@dbolshak looks like you have everything correct. SgxDevicePlugin is a cluster scoped resource so just deploy it without the namespace (-n sgx-ecdsa-quote).

dbolshak, 2022-05-13:
Ok, I've deployed without specifying namespace, but it changes nothing, I still don't see `sgxdeviceplugin-sample` and sections with events of the following command is empty
```kubectl describe sgxdeviceplugin sgxdeviceplugin-sample```

```
...
....
Spec:
  Enclave Limit:  110
  Image:          intel/intel-sgx-plugin:0.24.0
  Init Image:     intel/intel-sgx-initcontainer:0.24.0
  Log Level:      4
  Node Selector:
    intel.feature.node.kubernetes.io/sgx:  true
  Provision Limit:                         110
Status:
  Controlled Daemon Set:
    API Version:             apps/v1
    Kind:                    DaemonSet
    Name:                    intel-sgx-plugin
    Namespace:               inteldeviceplugins-system
    Resource Version:        80318
    UID:                     e80dc977-8956-424f-9f27-151cd987a144
  Desired Number Scheduled:  1
  Node Names:
    euclid-4
  Number Ready:  1
Events:          <none>
```

mythi, 2022-05-13:
>  I still don't see `sgxdeviceplugin-sample` and sections with events of the following command is empty

I don't think we report events for `xDevicePlugins`  objects today so empty `Events` is expected AFAICS. Other than that, is there something else missing?

dbolshak, 2022-05-13:
Screencast shows that there is pod with name `sgxdeviceplugin-sample` should be appeared, I don't have it.

mythi, 2022-05-13:
> Screencast shows that there is pod with name `sgxdeviceplugin-sample` should be appeared, I don't have it.

I see. That posted video seems to be outdated but the latest "script" with up-to-date information is in our [demo](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/demo/screencast-sgx.sh) folder. With `./screencast-play.sh play` it runs you trough the steps (beware of some of the problems reported in this issue though).

The video shows the plugin running in that namespace when `SgxDevicePlugin` was still namespace scoped but we moved it to cluster scoped.

Apologies for some of the outdated information.

dbolshak, 2022-05-16:
Thanks for clarification!