intel/intel-device-plugins-for-kubernetes #972: installing package fails with unknown revision v0.0.0

DanielFroehlich, 2022-04-19:
Hi! 
I am trying to use the package to develop my own device plugin, following DEVEL instructions. 

using ` go version go1.16.15 darwin/amd64`, getting the package fails with:

```
% go get -v github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin
get "k8s.io/api": found meta tag vcs.metaImport{Prefix:"k8s.io/api", VCS:"git", RepoRoot:"https://github.com/kubernetes/api"} at //k8s.io/api?go-get=1
get "k8s.io/kubelet": found meta tag vcs.metaImport{Prefix:"k8s.io/kubelet", VCS:"git", RepoRoot:"https://github.com/kubernetes/kubelet"} at //k8s.io/kubelet?go-get=1
get "k8s.io/kube-proxy": found meta tag vcs.metaImport{Prefix:"k8s.io/kube-proxy", VCS:"git", RepoRoot:"https://github.com/kubernetes/kube-proxy"} at //k8s.io/kube-proxy?go-get=1
get "k8s.io/csi-translation-lib": found meta tag vcs.metaImport{Prefix:"k8s.io/csi-translation-lib", VCS:"git", RepoRoot:"https://github.com/kubernetes/csi-translation-lib"} at //k8s.io/csi-translation-lib?go-get=1
get "k8s.io/kubectl": found meta tag vcs.metaImport{Prefix:"k8s.io/kubectl", VCS:"git", RepoRoot:"https://github.com/kubernetes/kubectl"} at //k8s.io/kubectl?go-get=1
get "k8s.io/cluster-bootstrap": found meta tag vcs.metaImport{Prefix:"k8s.io/cluster-bootstrap", VCS:"git", RepoRoot:"https://github.com/kubernetes/cluster-bootstrap"} at //k8s.io/cluster-bootstrap?go-get=1
get "k8s.io/kube-aggregator": found meta tag vcs.metaImport{Prefix:"k8s.io/kube-aggregator", VCS:"git", RepoRoot:"https://github.com/kubernetes/kube-aggregator"} at //k8s.io/kube-aggregator?go-get=1
get "k8s.io/sample-apiserver": found meta tag vcs.metaImport{Prefix:"k8s.io/sample-apiserver", VCS:"git", RepoRoot:"https://github.com/kubernetes/sample-apiserver"} at //k8s.io/sample-apiserver?go-get=1
get "k8s.io/pod-security-admission": found meta tag vcs.metaImport{Prefix:"k8s.io/pod-security-admission", VCS:"git", RepoRoot:"https://github.com/kubernetes/pod-security-admission"} at //k8s.io/pod-security-admission?go-get=1
get "k8s.io/controller-manager": found meta tag vcs.metaImport{Prefix:"k8s.io/controller-manager", VCS:"git", RepoRoot:"https://github.com/kubernetes/controller-manager"} at //k8s.io/controller-manager?go-get=1
get "k8s.io/cloud-provider": found meta tag vcs.metaImport{Prefix:"k8s.io/cloud-provider", VCS:"git", RepoRoot:"https://github.com/kubernetes/cloud-provider"} at //k8s.io/cloud-provider?go-get=1
get "k8s.io/component-base": found meta tag vcs.metaImport{Prefix:"k8s.io/component-base", VCS:"git", RepoRoot:"https://github.com/kubernetes/component-base"} at //k8s.io/component-base?go-get=1
get "k8s.io/apiserver": found meta tag vcs.metaImport{Prefix:"k8s.io/apiserver", VCS:"git", RepoRoot:"https://github.com/kubernetes/apiserver"} at //k8s.io/apiserver?go-get=1
get "k8s.io/client-go": found meta tag vcs.metaImport{Prefix:"k8s.io/client-go", VCS:"git", RepoRoot:"https://github.com/kubernetes/client-go"} at //k8s.io/client-go?go-get=1
get "k8s.io/kube-controller-manager": found meta tag vcs.metaImport{Prefix:"k8s.io/kube-controller-manager", VCS:"git", RepoRoot:"https://github.com/kubernetes/kube-controller-manager"} at //k8s.io/kube-controller-manager?go-get=1
get "k8s.io/apimachinery": found meta tag vcs.metaImport{Prefix:"k8s.io/apimachinery", VCS:"git", RepoRoot:"https://github.com/kubernetes/apimachinery"} at //k8s.io/apimachinery?go-get=1
get "k8s.io/cri-api": found meta tag vcs.metaImport{Prefix:"k8s.io/cri-api", VCS:"git", RepoRoot:"https://github.com/kubernetes/cri-api"} at //k8s.io/cri-api?go-get=1
get "k8s.io/apiextensions-apiserver": found meta tag vcs.metaImport{Prefix:"k8s.io/apiextensions-apiserver", VCS:"git", RepoRoot:"https://github.com/kubernetes/apiextensions-apiserver"} at //k8s.io/apiextensions-apiserver?go-get=1
get "k8s.io/component-helpers": found meta tag vcs.metaImport{Prefix:"k8s.io/component-helpers", VCS:"git", RepoRoot:"https://github.com/kubernetes/component-helpers"} at //k8s.io/component-helpers?go-get=1
get "k8s.io/metrics": found meta tag vcs.metaImport{Prefix:"k8s.io/metrics", VCS:"git", RepoRoot:"https://github.com/kubernetes/metrics"} at //k8s.io/metrics?go-get=1
get "k8s.io/mount-utils": found meta tag vcs.metaImport{Prefix:"k8s.io/mount-utils", VCS:"git", RepoRoot:"https://github.com/kubernetes/mount-utils"} at //k8s.io/mount-utils?go-get=1
get "k8s.io/kube-scheduler": found meta tag vcs.metaImport{Prefix:"k8s.io/kube-scheduler", VCS:"git", RepoRoot:"https://github.com/kubernetes/kube-scheduler"} at //k8s.io/kube-scheduler?go-get=1
get "k8s.io/legacy-cloud-providers": found meta tag vcs.metaImport{Prefix:"k8s.io/legacy-cloud-providers", VCS:"git", RepoRoot:"https://github.com/kubernetes/legacy-cloud-providers"} at //k8s.io/legacy-cloud-providers?go-get=1
get "k8s.io/cli-runtime": found meta tag vcs.metaImport{Prefix:"k8s.io/cli-runtime", VCS:"git", RepoRoot:"https://github.com/kubernetes/cli-runtime"} at //k8s.io/cli-runtime?go-get=1
get "k8s.io/code-generator": found meta tag vcs.metaImport{Prefix:"k8s.io/code-generator", VCS:"git", RepoRoot:"https://github.com/kubernetes/code-generator"} at //k8s.io/code-generator?go-get=1
go get: github.com/intel/intel-device-plugins-for-kubernetes@v0.11.0 updating to
	github.com/intel/intel-device-plugins-for-kubernetes@v0.23.0 requires
	k8s.io/kubectl@v0.0.0: reading k8s.io/kubectl/go.mod at revision v0.0.0: unknown revision v0.0.0
```

What am I doing wrong?

mythi, 2022-04-20:
@DanielFroehlich I'm seeing this too. I'll take a look.

mythi, 2022-04-20:
FYI it works with Go 1.18:

```
$  mkdir -p $(go env GOPATH)/src/example.com/my-custom-plugin
$  cd $(go env GOPATH)/src/example.com/my-custom-plugin
$  go mod init
$  go get -v github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin@v0.23.0
```

mythi, 2022-04-20:
Go 1.17 is also fine so you should be able to use at least `v0.23.0`  (which sets `go 1.17`). For Go 1.16, `v0.22.0` should also work but it fails... any change you could upgrade to Go 1.17/18 and use the latest release?

bart0sh, 2022-04-20:
@mythi it would probably make sense to include recommended go versions to [this table](https://github.com/intel/intel-device-plugins-for-kubernetes#supported-kubernetes-versions) and cover this with test cases.

DanielFroehlich, 2022-04-20:
Thx for looking into this!

With `go version go1.17.9 darwin/amd64`
I do get 
```
% go get -v github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin@v0.23.0
github.com/intel/intel-device-plugins-for-kubernetes/pkg/topology
# github.com/intel/intel-device-plugins-for-kubernetes/pkg/topology
../../../go/pkg/mod/github.com/intel/intel-device-plugins-for-kubernetes@v0.23.0/pkg/topology/topology.go:244:21: cannot use rdev (type int32) as type uint64 in argument to unix.Major
../../../go/pkg/mod/github.com/intel/intel-device-plugins-for-kubernetes@v0.23.0/pkg/topology/topology.go:245:21: cannot use rdev (type int32) as type uint64 in argument to unix.Minor
```
(on a fresh dir with fresh `go init`)

I would prefer V0.22.0, as my target k8s cluster is k8s V1.22, but will try with go 1.18 now...

mythi, 2022-04-21:
That error is some `GOOS=darwin` issue (I can reproduce it here by setting that env for my `go get`). If you use `go get -d ...` this probably fails and then build the plugin for `GOOS=linux`? @bart0sh any thoughts on this?

> I would prefer V0.22.0, as my target k8s cluster is k8s V1.22

Looks like the releases before v0.23 have problems with `go.mod`. Our `tests/e2e` depends on `k8s.io/kubernetes` which is not preferred and that results in this error you reported originally. With v0.23 we moved to `go 1.17` and I think the *lazy module loading*  makes the problem go away.

DanielFroehlich, 2022-04-21:
I can confirm that V0.23 wit go 1.18 works, even against k8s V1.22. So from my perspective, this issue can be closed, thanks for your help!
But I leave that to you, if you want to update the docs as mention in a comment above, or even actually fix the issue in the previous versions.

mythi, 2022-04-21:
@DanielFroehlich thanks for confirming it works for you! 

I think we can close this

closed 2022-04-21 by mythi