intel/intel-device-plugins-for-kubernetes #1007: revisit set-version flows

mythi, 2022-05-10:
we have `make set-version TAG=<release>` which used to work OK for the purpose it was created but new functionality and usages have added corner cases that makes `set-version` fragile:

- [ ] `make test-with-kind` in `Jenkinsfile` is not covered
- [ ] `ImageMinVersion` is not covered
- [ ] e2e tests for pull requests expect to get the image version from that pull request so we are "patching" `test/e2e/*/*.go`.

This issue is to work on an improved version for setting the image tags in various places.

hj-johannes-lee, 2022-05-20:
- [ ] variable `OPERATOR_VERSION` in `Makefile`
- [ ] `version` field in `deployments/operator/manifests/bases/intel-device-plugins-operator.clusterserviceversion.yaml`

msivosuo, 2022-10-05:
Slow progress/no progress. Need to be pushed from September to October.

msivosuo, 2022-11-17:
Deprioritized -> moving to backlog.