intel/intel-device-plugins-for-kubernetes #1053: Use initcontainer to configure DLB devices

bart0sh, 2022-06-14:
Some of the DLB usage scenarios require additional setup, such as enabling VFs, binding them etc. This should be done in the initcontainer similarly to how it's done in the [QAT initcontainer](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/release-0.24/build/docker/intel-qat-initcontainer.Dockerfile#L58).

mythi, 2022-06-15:
> This should be done in the initcontainer

as part of this item plan for the image publishing parts

daixiang0, 2022-07-18:
I met the case that the number DLB device(`/dev/dlbN`) is random in the container, I use the first DLB device as default and support config, but it is impossible to confirm which random number would be used before config. Could we pin it in the initContainer as well? I really do not want to detect it with `ls`.

bart0sh, 2022-07-18:
@daixiang0 Let's discuss this on the following example:

Let's say we have 8 pf devices: /dev/dlb0 - /dev/dlb7. Pods request them as dlb.intel.com/pf resources, work with them and release them after their jobs are done. DLB plugin allocates devices in their order (dlb0, dlb1, dlb2 ... dlb7). At some point all devices are allocated to workloads. The workload that uses /dev/dlb5 terminates and dlb5 becomes allocatable again. When new workload requests dlb.intel.com/pf, /dev/dlb5 device will be allocated for it.
How would you propose to pin the devices?

daixiang0, 2022-07-19:
So now we can not get fake N in the container, right? In most cases, users do not care about the actual N as all they want is only one device. I know that keeps the actual N easy to debug, could we optimize it?

For now, I need to add detect logic to ensure the app work.

mythi, 2022-07-19:
> I know that keeps the actual N easy to debug, could we optimize it?

I think the best thing to do is to set the provided devices to env variables for the container, something like:

```
DLB0=/dev/dlbX
DLBx=/dev/dlbY
...
```

This is similar what we do with QAT. See: https://github.com/intel/intel-device-plugins-for-kubernetes/blob/3c948cc106914c8e7149b6b5e74e05490f4e8a7f/demo/crypto-perf/run-dpdk-test#L13

Would that work?

bart0sh, 2022-07-19:
Another way is to make one VF device and request it. This way you'll always get the same N.

bart0sh, 2022-07-19:
@mythi would it make sense to sort [this output](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/dlb_plugin/dlb_plugin.go#L81) to make N a bit more predictable?

mythi, 2022-07-19:
> @mythi would it make sense to sort [this output](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/dlb_plugin/dlb_plugin.go#L81) to make N a bit more predictable?

There's no predictability in how kubelet picks devices in Allocate because the registered device IDs are stored in Go maps and the order of elements in Go maps is random.

daixiang0, 2022-08-08:
Is there an example for C++ to detect it? I can not use bash script in initContainer since it is already a sidecar.

mythi, 2022-08-08:
> Is there an example for C++ to detect it? I can not use bash script in initContainer since it is already a sidecar.

Are you asking how the proposal I made would work or how to detect devices in `/dev/dlbX`?

daixiang0, 2022-08-09:
I ask how to detect devices in `/dev/dlbx` using c++ code?

mythi, 2022-08-09:
> I ask how to detect devices in `/dev/dlbx` using c++ code?

Before going into this question, let's try to conclude what the device plugin can do to make it as easy as possible. Can you comment on the proposal I made? (btw, maybe start with a new issue first because this conversation is not related to this issue)

daixiang0, 2022-08-09:
I agree.

daixiang0, 2022-08-09:
Is it possible to expose an ENV to make applications easily detect?

mythi, 2022-08-09:
Yes (see the QAT example)

daixiang0, 2022-08-10:
Could you give a link?

mythi, 2022-08-10:
The plugin does not support it yet.

daixiang0, 2022-08-15:
> The plugin does not support it yet.

Will it come in Sep.2022?

mythi, 2022-08-15:
> > The plugin does not support it yet.
> 
> Will it come in Sep.2022?

Most likely no, unless you are able to contribute the changes. The first thing is to submit an issue about what the ask is so we could get it to our Q4 planning.

There are ways to workaround the gap you are observing so not having it should not be a blocker.

PR #1088 (2022-10-07): dlb: add initcontainer to plugin
Closes: #1053 

initcontainer enables vfs and configures vfs
 - the number of pfs and vfs per pf can be configurable by users
add dlb-initcontainer kustomize overlay
update CRD to have initImage
implment operator to run initcontainer
update e2e test to run initcontainer overlay
pudate envtest to test initimage

---
TO BE DISCUSSED & IMPROVED:
1. Do we actually keep the variable "REMAINING_NUM_PFS"? Or find another way?
2. How to have no 'default' values for variables ("REMAINING_NUM_PFS", "NUM_VFS_PER_PF) in the script, yaml, etc.
3. This PR does not have any c [... cut]
closed 2022-10-07 by bart0sh