intel/intel-device-plugins-for-kubernetes #1059: GPU plugin README deployment links give 404

eero-t, 2022-06-22:
In GPU plugin README install section: https://github.com/intel/intel-device-plugins-for-kubernetes/tree/main/cmd/gpu_plugin#deploy-with-pre-built-container-image

Both links to releases give 404:
* https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/gpu_plugin?ref=v.0.24.0
* https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/gpu_plugin/overlays/nfd_labeled_nodes?ref=v0.24.0

For some reason "kubectl apply -k" works, but it's better if one can review what is to be installed, before actually doing it.

eero-t, 2022-07-20:
> `kubectl apply -k --dry-run` can be used for viewing what is going to be constructed as a result of kustomization of that directory, those links are not supposed to be viewed in browser.

That tells just type and name of the created objects:
```
$ kubectl apply --dry-run=client -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/gpu_plugin?ref=v0.24.0
daemonset.apps/intel-gpu-plugin created (dry run)
```

And nothing about their properties: what they require, where they are deployed, what accesses are granted to the pods etc, like link here would tell: https://github.com/intel/intel-device-plugins-for-kubernetes/tree/release-0.24/deployments

PS. "desired release version" could be a link to: https://github.com/intel/intel-device-plugins-for-kubernetes/releases

uniemimu, 2022-07-20:
@eero-t byako wasn't perhaps descriptive enough above. With `-o yaml` in the end you can see the content.

Try
`
kubectl apply --dry-run=client -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/gpu_plugin?ref=v0.24.0 -o yaml`

In any case the links weren't intended for browser viewing

mythi, 2022-07-20:
I'm always using 
```
kubectl kustomize https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/gpu_plugin?ref=v0.24.0
```

mythi, 2022-07-20:
> @mythi this is somewhat inconsistent.

yep, the docs used to follow the same structure but the installation instructions have diverged depending on what deployment options have been available at the time new plugins have been introduced.

PR #1384 (2023-04-26): Doc updates
Updates for documentation from #1045.

Fixes: #1381
Fixes: #1132 
Fixes: #1059
commit 2a365263 (2023-04-26): gpu: add note about dry-run and yaml output

Fixes: #1059