intel/intel-device-plugins-for-kubernetes #1069: NFD does not label the  arria10 fpga following the operator README

klin938, 2022-07-06:
Hi guys,

I am trying to provision fpga support on our K8s cluster. I mainly follow the operator README but no success

> https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/operator/README.md

After NFD and the NodeFeatureRules have been installed, I don't see my FPGA node gets labeled as expected.

```
I0706 02:13:11.213981       1 nfd-worker.go:155] Node Feature Discovery Worker v0.10.1
I0706 02:13:11.214076       1 nfd-worker.go:156] NodeName: 'r640-1-10.local'
I0706 02:13:11.214440       1 nfd-worker.go:423] configuration file "/etc/kubernetes/node-feature-discovery/nfd-worker.conf" parsed
I0706 02:13:11.214500       1 nfd-worker.go:461] worker (re-)configuration successfully completed
I0706 02:13:11.215027       1 base.go:126] connecting to nfd-master at nfd-master:8080 ...
I0706 02:13:11.215083       1 component.go:36] [core]parsed scheme: ""
I0706 02:13:11.215110       1 component.go:36] [core]scheme "" not registered, fallback to default scheme
I0706 02:13:11.215150       1 component.go:36] [core]ccResolverWrapper: sending update to cc: {[{nfd-master:8080  <nil> 0 <nil>}] <nil> <nil>}
I0706 02:13:11.215166       1 component.go:36] [core]ClientConn switching balancer to "pick_first"
I0706 02:13:11.215171       1 component.go:36] [core]Channel switches to new LB policy "pick_first"
I0706 02:13:11.215317       1 component.go:36] [core]Subchannel Connectivity change to CONNECTING
I0706 02:13:11.215451       1 component.go:36] [core]Subchannel picks a new address "nfd-master:8080" to connect
I0706 02:13:11.215567       1 component.go:36] [core]Channel Connectivity change to CONNECTING
W0706 02:13:12.219934       1 component.go:41] [core]grpc: addrConn.createTransport failed to connect to {nfd-master:8080 nfd-master:8080 <nil> 0 <nil>}. Err: connection error: desc = "transport: Error while dialing dial tcp 172.130.68.166:8080: connect: connection refused". Reconnecting...
I0706 02:13:12.219981       1 component.go:36] [core]Subchannel Connectivity change to TRANSIENT_FAILURE
I0706 02:13:12.220056       1 component.go:36] [core]Channel Connectivity change to TRANSIENT_FAILURE
I0706 02:13:13.220337       1 component.go:36] [core]Subchannel Connectivity change to CONNECTING
I0706 02:13:13.220438       1 component.go:36] [core]Subchannel picks a new address "nfd-master:8080" to connect
I0706 02:13:13.220516       1 component.go:36] [core]Channel Connectivity change to CONNECTING
W0706 02:13:14.223652       1 component.go:41] [core]grpc: addrConn.createTransport failed to connect to {nfd-master:8080 nfd-master:8080 <nil> 0 <nil>}. Err: connection error: desc = "transport: Error while dialing dial tcp 172.130.68.166:8080: connect: connection refused". Reconnecting...
I0706 02:13:14.223694       1 component.go:36] [core]Subchannel Connectivity change to TRANSIENT_FAILURE
I0706 02:13:14.223749       1 component.go:36] [core]Channel Connectivity change to TRANSIENT_FAILURE
I0706 02:13:16.141036       1 component.go:36] [core]Subchannel Connectivity change to CONNECTING
I0706 02:13:16.141096       1 component.go:36] [core]Subchannel picks a new address "nfd-master:8080" to connect
I0706 02:13:16.141230       1 component.go:36] [core]Channel Connectivity change to CONNECTING
W0706 02:13:17.144054       1 component.go:41] [core]grpc: addrConn.createTransport failed to connect to {nfd-master:8080 nfd-master:8080 <nil> 0 <nil>}. Err: connection error: desc = "transport: Error while dialing dial tcp 172.130.68.166:8080: connect: connection refused". Reconnecting...
I0706 02:13:17.144101       1 component.go:36] [core]Subchannel Connectivity change to TRANSIENT_FAILURE
I0706 02:13:17.144156       1 component.go:36] [core]Channel Connectivity change to TRANSIENT_FAILURE
I0706 02:13:20.157894       1 component.go:36] [core]Subchannel Connectivity change to CONNECTING
I0706 02:13:20.157957       1 component.go:36] [core]Subchannel picks a new address "nfd-master:8080" to connect
I0706 02:13:20.158158       1 component.go:36] [core]Channel Connectivity change to CONNECTING
I0706 02:13:21.163224       1 component.go:36] [core]Subchannel Connectivity change to READY
I0706 02:13:21.163282       1 component.go:36] [core]Channel Connectivity change to READY
E0706 02:13:21.182977       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em1/speed: invalid argument
E0706 02:13:21.183117       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em2/speed: invalid argument
E0706 02:13:21.183245       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em3/speed: invalid argument
I0706 02:13:21.258538       1 nfd-worker.go:472] starting feature discovery...
I0706 02:13:21.259297       1 nfd-worker.go:484] feature discovery completed
I0706 02:13:21.259320       1 nfd-worker.go:565] sending labeling request to nfd-master
E0706 02:14:21.322843       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em1/speed: invalid argument
E0706 02:14:21.323017       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em2/speed: invalid argument
E0706 02:14:21.323174       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em3/speed: invalid argument
I0706 02:14:21.398657       1 nfd-worker.go:472] starting feature discovery...
I0706 02:14:21.399335       1 nfd-worker.go:484] feature discovery completed
I0706 02:14:21.399375       1 nfd-worker.go:565] sending labeling request to nfd-master
E0706 02:15:21.443092       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em1/speed: invalid argument
E0706 02:15:21.443307       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em2/speed: invalid argument
E0706 02:15:21.443525       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em3/speed: invalid argument
I0706 02:15:21.511411       1 nfd-worker.go:472] starting feature discovery...
I0706 02:15:21.511724       1 nfd-worker.go:484] feature discovery completed
I0706 02:15:21.511736       1 nfd-worker.go:565] sending labeling request to nfd-master
E0706 02:16:21.552463       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em1/speed: invalid argument
E0706 02:16:21.552585       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em2/speed: invalid argument
E0706 02:16:21.552702       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em3/speed: invalid argument
I0706 02:16:21.619316       1 nfd-worker.go:472] starting feature discovery...
I0706 02:16:21.619634       1 nfd-worker.go:484] feature discovery completed
I0706 02:16:21.619647       1 nfd-worker.go:565] sending labeling request to nfd-master
E0706 02:17:21.669180       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em1/speed: invalid argument
E0706 02:17:21.669361       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em2/speed: invalid argument
E0706 02:17:21.669508       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em3/speed: invalid argument
I0706 02:17:21.737414       1 nfd-worker.go:472] starting feature discovery...
[... 42 lines cut ...]
I0706 02:24:22.556538       1 nfd-worker.go:484] feature discovery completed
I0706 02:24:22.556553       1 nfd-worker.go:565] sending labeling request to nfd-master
E0706 02:25:22.599371       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em1/speed: invalid argument
E0706 02:25:22.599496       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em2/speed: invalid argument
E0706 02:25:22.599620       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em3/speed: invalid argument
I0706 02:25:22.669471       1 nfd-worker.go:472] starting feature discovery...
I0706 02:25:22.669724       1 nfd-worker.go:484] feature discovery completed
I0706 02:25:22.669736       1 nfd-worker.go:565] sending labeling request to nfd-master
E0706 02:26:22.716209       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em1/speed: invalid argument
[... cut ...]
E0706 02:27:22.832923       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em2/speed: invalid argument
E0706 02:27:22.833120       1 network.go:145] failed to read net iface attribute speed: read /host-sys/class/net/em3/speed: invalid argument
I0706 02:27:22.899538       1 nfd-worker.go:472] starting feature discovery...
I0706 02:27:22.899851       1 nfd-worker.go:484] feature discovery completed
I0706 02:27:22.899863       1 nfd-worker.go:565] sending labeling request to nfd-master
```
We are running K8s 1.23.0 version and the FPGA node has IAS installed.

Cheers
Derrick

bart0sh, 2022-07-06:
Hi Derrick,

Thank you for reporting this issue.
You can see current labelling rules for Arria10 [here](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/deployments/nfd/overlays/node-feature-rules/node-feature-rules.yaml#L36:L44)

Can you show lspci output for your devices please?
Here is how it looks on our system with two Arria10 cards:
```
> sudo lspci -vn |grep -iA2 09c4
3b:00.0 1200: 8086:09c4
	Subsystem: 8086:0000
	Flags: bus master, fast devsel, latency 0, IRQ 62, NUMA node 0
--
5e:00.0 1200: 8086:09c4
	Subsystem: 8086:0000
	Flags: bus master, fast devsel, latency 0, IRQ 70, NUMA node 0
```

Another guess is that you're not using dfl driver.
In any case if you can describe your setup in more details we'll be happy to update the rules.

bart0sh, 2022-07-06:
yes, you're using 3rd party kernel module and that's the reason why that labelling rule didn't work.

Can you check if this one works for you:
```
    - name: "intel.fpga-arria10"
      labels:
        "intel.feature.node.kubernetes.io/fpga-arria10": "true"
      matchFeatures:
        - feature: pci.device
          matchExpressions:
            vendor: {op: In, value: ["8086"]}
            device: {op: In, value: ["09c4"]}
            class: {op: In, value: ["1200"]}
      matchAny:
        - feature: kernel.loadedmodule
            matchExpressions:
              dfl_pci: {op: Exists}
        - feature: kernel.loadedmodule
            matchExpressions:
              intel_fpga_pci: {op:Exists}
```

klin938, 2022-07-06:
Thanks for providing the insight! I gave your code a quick touchup and it works!
```
    - name: "intel.fpga-arria10"
      labels:
        "intel.feature.node.kubernetes.io/fpga-arria10": "true"
      matchFeatures:
        - feature: pci.device
          matchExpressions:
            vendor: {op: In, value: ["8086"]}
            device: {op: In, value: ["09c4"]}
            class: {op: In, value: ["1200"]}
      matchAny:
        - matchFeatures:
            - feature: kernel.loadedmodule
              matchExpressions:
                dfl_pci: {op: Exists}
        - matchFeatures:
            - feature: kernel.loadedmodule
              matchExpressions:
                intel_fpga_pci: {op: Exists}
```
I also want to mention that this documentation states both DFL and OPAE drivers are supported by the plugin:

> https://intel.github.io/intel-device-plugins-for-kubernetes/cmd/fpga_plugin/README.html?highlight=dfl+driver

The current code seems to be supporting DFL only? But your quick fix allows me to progress to the next step! Thanks!

bart0sh, 2022-07-06:
@klin938 thank you for confirming that updated rules work for you. I'll submit a PR today.

The current code should [support both drivers](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/fpga_plugin/fpga_plugin.go#L207:L209).

PR #1070 (2022-07-18): fpga: update NFD labelling rules
Added OPAE kernel module as an alternative to DFL
to the Arria10 labeling rules.

Fixes: #1069
commit 328e56e9 (2022-07-18): fpga: update NFD labelling rules

Added OPAE kernel module as an alternative to DFL
to the Arria10 labeling rules.

Fixes: #1069
closed 2022-07-18 by mythi