intel/intel-device-plugins-for-kubernetes #1074: Resources device name in Pod spec

klin938, 2022-07-06:
Hi again,

In the demo, it has:

```
> kubectl describe node  |grep -A3 Allocatable                                                                                                                                    
Allocatable:                                                                                                                                                                      
  cpu:                                                     32                                                                                                                     
  ephemeral-storage:                                       94636958359                                                                                                            
  fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6:  2   
```

But inside the test-fpga-region.yaml, the specified device name is

```
fpga.intel.com/arria10.dcp1.2-nlb3-orchestrated: 1
```

I have watched the demo few times, could not figure it out where and how the name changed. 

On my own K8s cluster environment, I replaced the device name in Pod spec with the one from the Allocatable output, the webhook refused the request:
```
Error from server: error when creating "test-fpga-orchestrated.yaml": admission webhook "fpga.mutator.webhooks.intel.com" denied the request: no such resource: "fpga.intel.com/region-38d782e3b6125343b9342433e348ac4c"
```

Did I miss something?

Cheers
Derrick

bart0sh, 2022-07-07:
It's changed by webhook according to the [mappings](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/deployments/fpga_admissionwebhook/mappings-collection.yaml#L115:L122).
You can change mappings to reflect your devices interface and AFU ids if they differ from those in the example collection.
Please find more details about mappings in the [webhook documentation](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/fpga_admissionwebhook/README.md#mappings).

Generally region(orchestration programmed) mode is much more difficult to setup. You can start from af(preprogrammed) mode if you just want to get familiar with the plugin.

klin938, 2022-07-07:
Hi! I did change the mapping in the test-fpga-region.yaml to reflect what is mapped on our system.

I redeployed the plugin in af mode, new mapping name is generated. But the webhook still deny the resource request:

```
[root@r640-1-10 demo]# kubectl describe node  |grep -A3 Allocatable
--
Allocatable:
  cpu:                                                                    80
  ephemeral-storage:                                                      354306354614
  fpga.intel.com/af-38d.d84.ONeC47YSU0O5NCQz40isTNhCTcSko8QT-J5DNoP5BAs:  1
[root@r640-1-10 demo]# kubectl create -f test-fpga-preprogrammed.yaml
Error from server: error when creating "test-fpga-preprogrammed.yaml": admission webhook "fpga.mutator.webhooks.intel.com" denied the request: no such resource: "fpga.intel.com/af-38d.d84.onec47ysu0o5ncqz40istnhctcsko8qt-j5dnop5bas"
```
My pod yaml looks like:
```
[root@r640-1-10 demo]# cat test-fpga-preprogrammed.yaml
apiVersion: v1
kind: Pod
metadata:
  name: test-fpga-preprogrammed
spec:
  containers:
  - name: test-container
    image: intel/opae-nlb-demo:devel
    imagePullPolicy: IfNotPresent
    securityContext:
      capabilities:
        add:
          [IPC_LOCK]
    resources:
      limits:
        fpga.intel.com/af-38d.d84.ONeC47YSU0O5NCQz40isTNhCTcSko8QT-J5DNoP5BAs: 1
        cpu: 1
        hugepages-2Mi: 20Mi

  restartPolicy: Never
```

bart0sh, 2022-07-07:
You should use mapped name in the pod description. That's actually the purpose of the mappings - to use human-readable resource names in yamls, e.g. fpga.intel.com/<name from the mappings> instead of fpga.intel.com/af-38d.d84.ONeC47YSU0O5NCQz40isTNhCTcSko8QT-J5DNoP5BAs.

klin938, 2022-07-07:
Hi, thanks for providing more information. I realized that I have missed the big part here:
`deployments/fpga_admissionwebhook/mappings-collection.yaml`
I think the documentation implies system admin may need to deploy their own mappings according to their need.

My FPGA card looks like
```
[root@r640-1-10 intel-device-plugins-for-kubernetes]# fpgainfo fme
Board Management Controller, microcontroller FW version 26895
Last Power Down Cause: POK_CORE
Last Reset Cause: None
//****** FME ******//
Object Id                     : 0xEB00000
PCIe s:b:d:f                  : 0000:5E:00:0
Device Id                     : 0x09C4
Socket Id                     : 0x00
Ports Num                     : 01
Bitstream Id                  : 0x124000200000367
Bitstream Version             : 1.2.4
Pr Interface Id               : 38d782e3-b612-5343-b934-2433e348ac4c
Boot Page                     : user
```
I am not sure what the correct DCP version for setting up the FpgaRegion object. Or maybe it does not matter at all?

Thanks

bart0sh, 2022-07-07:
> I think the documentation implies system admin may need to deploy their own mappings according to their need.

Yes, the documentation states that [example mappings collection can be used as an example for cluster mappings. This collection is not intended to be deployed as is, it should be used as a reference and example of your own cluster mappings](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/fpga_admissionwebhook/README.md#mappings-deployment)

klin938, 2022-07-07:
Hi! Yes, I managed to run the test pod and replicated the af demo! You may close this ticket.

thanks!

closed 2022-07-08 by bart0sh