intel/intel-device-plugins-for-kubernetes #1087: Failed to open Intel SGX device

astronaut0131, 2022-07-28:
I've follow the installation directions in README and verified the installation successfully
```
$ kubectl describe node zhenhui-control-plane | grep sgx.intel
                    sgx.intel.com/capable=true
                    nfd.node.kubernetes.io/extended-resources: sgx.intel.com/epc
  sgx.intel.com/enclave:    110
  sgx.intel.com/epc:        4261412864
  sgx.intel.com/provision:  110
  sgx.intel.com/enclave:    110
  sgx.intel.com/epc:        4261412864
  sgx.intel.com/provision:  110
  sgx.intel.com/enclave    1           1
  sgx.intel.com/epc        512Ki       512Ki
  sgx.intel.com/provision  0           0
```
Then got the following error when deploying TCS issuer (https://github.com/intel/trusted-certificate-issuer) in k8s cluster.
```
$ kubectl logs tcs-controller-79c499fb98-v5kv8
2022-07-28T14:13:05.734Z        INFO    controller-runtime.metrics      metrics server is starting to listen    {"addr": ":8082"}
[get_driver_type edmm_utility.cpp:111] Failed to open Intel SGX device.
[get_driver_type /home/sgx/jenkins/ubuntuServer2004-release-build-trunk-215/build_target/PROD/label/Builder-UbuntuSrv20/label_exp/ubuntu64/linux-trunk-opensource/psw/urts/linux/edmm_utility.cpp:111] Failed to open Intel SGX device.
2022-07-28T14:13:05.929Z        LEVEL(-2)       SGX     Failed to configure command
2022-07-28T14:13:05.929Z        ERROR   setup   SGX initialization      {"error": "failed to initialize PKCS#11 library: pkcs11: 0x30: CKR_DEVICE_ERROR", "errorVerbose": "pkcs11: 0x30: CKR_DEVICE_ERROR\nfailed to initialize PKCS#11 library"}
```

I don't know whether here is the right place to ask this question, anyone can help?
Thanks.

mythi, 2022-07-28:
> I don't know whether here is the right place to ask this question, anyone can help?

I think the TCS repo would be a better place for this issue.

astronaut0131, 2022-07-28:
> 

ok, thank you for help @mythi

closed 2022-07-28 by astronaut0131