intel/intel-device-plugins-for-kubernetes #1090: Fix dlb plugin logic about deciding if devices are pf/vf/neither.

hj-johannes-lee, 2022-08-03:
There is a problem for the logic of dlb plugin.

DLB plugin decides if it's pf or vf based on sriov_nuvmfs [[link](https://github.com/intel/intel-device-plugins-for-kubernetes/pull/733/files#diff-c157625489fd9d5964dfe23cb487e21e93850f847a0c767308ff8c103008bfd4R94)].

The problem is that if a device has the value sriov_numvfs >0, the device is regarded as 'non dlb' device no matter how many resources are left.

But, unless resources of a PF are all already consumed by vfs, shouldn't the PF be visible as a working PF?

bart0sh, 2022-08-03:
>  the device is regarded as 'non dlb' device

do you mean non PF device?

hj-johannes-lee, 2022-08-08:
@bart0sh Ah, sorry that I missed your comment again.. Idk why I am doing all the time..
But, I think we discussed about this enough and this issue is not valid any more.!

closed 2022-08-08 by hj-johannes-lee