intel/intel-device-plugins-for-kubernetes #1095: SgxDevicePlugin not ready

astronaut0131, 2022-08-11:
I follow the instructions in https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/operator/README.md
```
$ kubectl get SgxDevicePlugin
NAME                     DESIRED   READY   NODE SELECTOR                                     AGE
sgxdeviceplugin-sample   1         0       {"intel.feature.node.kubernetes.io/sgx":"true"}   69m
```
```
$ kubectl get no -o json | jq .items[].metadata.labels |grep intel.feature.node.kubernetes.io/sgx
  "intel.feature.node.kubernetes.io/sgx": "true",
```

Why there is always 0 READY sgxdeviceplugin? My node actually has the label `"intel.feature.node.kubernetes.io/sgx": "true"`
How can I get more debug information about what's going on? Thanks for help.

mythi, 2022-08-11:
> How can I get more debug information about what's going on? 

This will give you better status information about the object:
```
$ kubectl describe sgxdeviceplugin sgxdeviceplugin-sample
```

Especially, look for `Controlled Daemon Set` and, further, check how that daemonset is going (e.g., `kubectl describe ds sgxdeviceplugin-sample-<suffix> -n inteldeviceplugins-system`)

astronaut0131, 2022-08-12:
```
$ kubectl describe SgxDevicePlugin sgxdeviceplugin-sample
Name:         sgxdeviceplugin-sample
Namespace:
Labels:       <none>
Annotations:  <none>
API Version:  deviceplugin.intel.com/v1
Kind:         SgxDevicePlugin
Metadata:
  Creation Timestamp:  2022-08-11T14:17:29Z
  Generation:          1
  Managed Fields:
    API Version:  deviceplugin.intel.com/v1
    Fields Type:  FieldsV1
    fieldsV1:
      f:metadata:
        f:annotations:
          .:
          f:kubectl.kubernetes.io/last-applied-configuration:
      f:spec:
        .:
        f:enclaveLimit:
        f:image:
        f:initImage:
        f:logLevel:
        f:nodeSelector:
          .:
          f:intel.feature.node.kubernetes.io/sgx:
        f:provisionLimit:
    Manager:      kubectl-client-side-apply
    Operation:    Update
    Time:         2022-08-11T14:17:29Z
    API Version:  deviceplugin.intel.com/v1
    Fields Type:  FieldsV1
    fieldsV1:
      f:status:
        .:
        f:controlledDaemonSet:
        f:desiredNumberScheduled:
        f:nodeNames:
        f:numberReady:
    Manager:         intel_deviceplugin_operator
    Operation:       Update
    Subresource:     status
    Time:            2022-08-11T14:42:18Z
  Resource Version:  127952
  UID:               7fa44f44-ca5e-41f8-a246-2c097d8a8d48
Spec:
  Enclave Limit:  110
  Image:          intel/intel-sgx-plugin:0.24.0
  Init Image:     intel/intel-sgx-initcontainer:0.24.0
  Log Level:      4
  Node Selector:
    intel.feature.node.kubernetes.io/sgx:  true
  Provision Limit:                         110
Status:
  Controlled Daemon Set:
    API Version:             apps/v1
    Kind:                    DaemonSet
    Name:                    intel-sgx-plugin
    Namespace:               inteldeviceplugins-system
    Resource Version:        923
    UID:                     c2b71f99-05dd-48f4-93ed-8ecc29f16dd9
  Desired Number Scheduled:  1
  Node Names:
    zhenhui-control-plane
  Number Ready:  0
Events:          <none>
```
```
$ kubectl describe ds intel-sgx-plugin -n inteldeviceplugins-system
Name:           intel-sgx-plugin
Selector:       app=intel-sgx-plugin
Node-Selector:  intel.feature.node.kubernetes.io/sgx=true
Labels:         app=intel-sgx-plugin
Annotations:    deprecated.daemonset.template.generation: 1
Desired Number of Nodes Scheduled: 1
Current Number of Nodes Scheduled: 1
Number of Nodes Scheduled with Up-to-date Pods: 1
Number of Nodes Scheduled with Available Pods: 0
Number of Nodes Misscheduled: 0
Pods Status:  1 Running / 0 Waiting / 0 Succeeded / 0 Failed
Pod Template:
  Labels:  app=intel-sgx-plugin
  Init Containers:
   intel-sgx-initcontainer:
    Image:        intel/intel-sgx-initcontainer:0.24.0
    Port:         <none>
    Host Port:    <none>
    Environment:  <none>
    Mounts:
      /etc/kubernetes/node-feature-discovery/source.d/ from nfd-source-hooks (rw)
  Containers:
   intel-sgx-plugin:
    Image:      intel/intel-sgx-plugin:0.24.0
    Port:       <none>
    Host Port:  <none>
    Args:
      -v
      4
      -enclave-limit
      110
      -provision-limit
      110
    Environment:  <none>
    Mounts:
      /dev/sgx_enclave from sgx-enclave (ro)
      /dev/sgx_provision from sgx-provision (ro)
      /var/lib/kubelet/device-plugins from kubeletsockets (rw)
  Volumes:
   kubeletsockets:
    Type:          HostPath (bare host directory volume)
    Path:          /var/lib/kubelet/device-plugins
    HostPathType:
   sgx-enclave:
    Type:          HostPath (bare host directory volume)
    Path:          /dev/sgx_enclave
    HostPathType:  CharDevice
   sgx-provision:
    Type:          HostPath (bare host directory volume)
    Path:          /dev/sgx_provision
    HostPathType:  CharDevice
   nfd-source-hooks:
    Type:          HostPath (bare host directory volume)
    Path:          /etc/kubernetes/node-feature-discovery/source.d/
    HostPathType:  DirectoryOrCreate
Events:            <none>
```
Is there anything wrong?

mythi, 2022-08-12:
> Is there something wrong?

the device plugin seems to be running OK (`kubectl get pods -n inteldeviceplugins-system`). Not sure why `READY` gives wrong info.

astronaut0131, 2022-08-12:
> kubectl get pods -n inteldeviceplugins-system

This works, the plugin pod is crashed.
```
$ kubectl get pods -n inteldeviceplugins-system
NAME                                                     READY   STATUS             RESTARTS          AGE
intel-sgx-plugin-pnt26                                   0/1     CrashLoopBackOff   268 (2m48s ago)   22h
inteldeviceplugins-controller-manager-68c66b8b45-lttkz   2/2     Running
```
```
$ kubectl logs intel-sgx-plugin-pnt26 -n inteldeviceplugins-system
Defaulted container "intel-sgx-plugin" out of: intel-sgx-plugin, intel-sgx-initcontainer (init)
I0812 13:09:24.665489       1 sgx_plugin.go:137] SGX device plugin started with 110 "sgx.intel.com/enclave" resources and 110 "sgx.intel.com/provision" resources.
I0812 13:09:24.691611       1 manager.go:115] Received dev updates:{map[enclave:map[sgx-enclave-0:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164480 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-1:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001644a0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-10:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001645c0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-100:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000165100 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-101:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000165120 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-102:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000165140 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-103:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000165160 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-104:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000165180 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-105:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001651a0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-106:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001651c0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-107:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001651e0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-108:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000165200 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-109:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000165220 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-11:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001645e0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-12:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164600 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-13:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164620 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-14:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164640 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-15:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164660 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-16:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164680 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-17:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001646a0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-18:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001646c0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-19:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001646e0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-2:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001644c0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-20:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164700 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-21:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164720 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-22:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164740 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-23:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164760 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-24:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164780 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-25:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001647a0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-26:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001647c0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-27:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001647e0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-28:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164800 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-29:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164820 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-3:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc0001644e0 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-30:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164840 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enclave-31:{[{/dev/sgx /dev/sgx false {} 0}] map[] map[] 0xc000164860 Healthy [{/dev/sgx_enclave /dev/sgx_enclave rw {} 0}]} sgx-enc [... cut ...]
        /go/src/github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin/manager.go:144
runtime.goexit
        /usr/local/go/src/runtime/asm_amd64.s:1571
```
This is how the directory looks like on my host os
```
ll /var/lib/kubelet/device-plugins
total 12
drwxr-xr-x 2 root root 4096 8月   2 23:54 ./
drwx------ 8 root root 4096 8月   2 00:30 ../
-rw-r--r-- 1 root root    0 8月   2 00:30 DEPRECATION
-rw------- 1 root root   79 8月   2 23:54 kubelet_internal_checkpoint
srwxr-xr-x 1 root root    0 8月   2 00:30 kubelet.sock=
```
Why the sock is already in use?

mythi, 2022-08-12:
> It's very confusing, the intel-sgx-plugin in default namespace is RUNNING.

you have deployed the plugin twice. just delete the daemonset in `default` namespace and then it'll work

astronaut0131, 2022-08-12:
Thank you for you patience @mythi :)
There's another kind of problem, I will raise a new issuer.

closed 2022-08-12 by astronaut0131