intel/intel-device-plugins-for-kubernetes #1098: SGX DCAP Quote Generation Error

astronaut0131, 2022-08-12:
I'm trying to run SGX ECDSA Remote Attestation following the instructions in https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/sgx_plugin/README.md
```
$ kubectl describe node zhenhui-control-plane | grep sgx.intel.com
                    nfd.node.kubernetes.io/extended-resources: sgx.intel.com/epc
  sgx.intel.com/enclave:    110
  sgx.intel.com/epc:        4261412864
  sgx.intel.com/provision:  110
  sgx.intel.com/enclave:    110
  sgx.intel.com/epc:        4261412864
  sgx.intel.com/provision:  110
  sgx.intel.com/enclave    1           1
  sgx.intel.com/epc        1Mi         1Mi
  sgx.intel.com/provision  1           1
```
```
$ kubectl get po
NAME                                  READY   STATUS    RESTARTS   AGE
ecdsa-quote-intelsgx-demo-job-5jgjg   0/1     Error     0          5m23s
ecdsa-quote-intelsgx-demo-job-7wwpb   0/1     Error     0          5m10s
ecdsa-quote-intelsgx-demo-job-hfc6v   0/1     Error     0          5m48s
ecdsa-quote-intelsgx-demo-job-kdt6l   0/1     Error     0          5m44s
ecdsa-quote-intelsgx-demo-job-l4vxg   0/1     Error     0          5m27s
ecdsa-quote-intelsgx-demo-job-mt4kx   0/1     Error     0          5m31s
ecdsa-quote-intelsgx-demo-job-pg79b   0/1     Error     0          5m6s
intel-sgx-aesmd-ztlcx                 1/1     Running   0          6m28s
intel-sgx-plugin-748n9                1/1     Running   0          12m
```
```
$ kubectl logs ecdsa-quote-intelsgx-demo-job-pg79b

Step1: Call sgx_qe_get_target_info:Error in sgx_qe_get_target_info. 0xe001
```

But I can run DCAP QuoteGenerationSample successfully on my host os.
```
~/zhenhui/SGXDataCenterAttestationPrimitives/SampleCode/QuoteGenerationSample# ./app
sgx_qe_set_enclave_load_policy is valid in in-proc mode only and it is optional: the default enclave load policy is persistent:
set the enclave load policy as persistent:succeed!

Step1: Call sgx_qe_get_target_info:succeed!
Step2: Call create_app_report:succeed!
Step3: Call sgx_qe_get_quote_size:succeed!
Step4: Call sgx_qe_get_quote:succeed!cert_key_type = 0x5
sgx_qe_cleanup_by_policy is valid in in-proc mode only.

 Clean up the enclave load policy:succeed!
```
https://github.com/intel/SGXDataCenterAttestationPrimitives/issues?q=0xe001

`0xe001` looks like an unexpected error, how can I debug this problem? Thanks for help.

mythi, 2022-08-12:
> 0xe001 looks like an unexpected error, how can I debug this problem? Thanks for help.

Is your `aesmd` pod configured to access this PCC:
https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/deployments/sgx_aesmd/base/sgx_default_qcnl.conf

astronaut0131, 2022-08-12:
> Is your `aesmd` pod configured to access this PCCS: https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/deployments/sgx_aesmd/base/sgx_default_qcnl.conf

I just use  
`kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/sgx_aesmd?ref=main` to deploy `aesmd` pod, without changing any configuration.

mythi, 2022-08-12:
The comparison of running `./app` on the host vs `ecdsa-quote-intelsgx-demo-job` not apples to apples because the latter uses `aesmd`.

Two things to check:
```
# SGX_AESM_ADDR=1 ./app
```

Do you see the `/var/run/aesmd/aesm.sock`?

astronaut0131, 2022-08-12:
> ```
> # SGX_AESM_ADDR=1 ./app
> ```
```
~/zhenhui/SGXDataCenterAttestationPrimitives/SampleCode/QuoteGenerationSample# SGX_AESM_ADDR=1 ./app

Step1: Call sgx_qe_get_target_info:succeed!
Step2: Call create_app_report:succeed!
Step3: Call sgx_qe_get_quote_size:succeed!
Step4: Call sgx_qe_get_quote:succeed!cert_key_type = 0x5
```

> Do you see the `/var/run/aesmd/aesm.sock`?

```
~/zhenhui/SGXDataCenterAttestationPrimitives/SampleCode/QuoteGenerationSample# ls -lt /var/run/aesmd/
total 0
srwxrwxrwx 1 aesmd aesmd 0 8月  11 18:13 aesm.socket
```
the file name is not `aesm.sock` but `aesm.socket`

mythi, 2022-08-12:
> ´systemctl status aesmd`

you have `aesmd` running both on the host and in the container? only one is enough.

astronaut0131, 2022-08-12:
> you have `aesmd` running both on the host and in the container? only one is enough.

Yep, I tried to stop the one on host with `systemctl stop aesmd` , but it doesn't help, there must be other reasons for the error.

astronaut0131, 2022-08-12:
If `aesmd` pod works correctly, should there be a socket file in `/var/run/aesmd/` on host os?  
After I stop the `aesmd` service on host os, `/var/run/aesmd/` now contains no socket.
```
$ kubectl get po
NAME                                  READY   STATUS    RESTARTS   AGE
ecdsa-quote-intelsgx-demo-job-58mgf   0/1     Error     0          5m10s
ecdsa-quote-intelsgx-demo-job-68nwh   0/1     Error     0          5m2s
ecdsa-quote-intelsgx-demo-job-c22wt   0/1     Error     0          4m37s
ecdsa-quote-intelsgx-demo-job-h5gcl   0/1     Error     0          4m50s
ecdsa-quote-intelsgx-demo-job-wnw9t   0/1     Error     0          5m6s
ecdsa-quote-intelsgx-demo-job-x46q4   0/1     Error     0          4m54s
ecdsa-quote-intelsgx-demo-job-xrm7x   0/1     Error     0          4m58s
intel-sgx-aesmd-7d4s8                 1/1     Running   0          6m41s
intel-sgx-plugin-748n9                1/1     Running   0          135m
```
```
$ ls -l /var/run/aesmd/
total 0
```

mythi, 2022-08-12:
> If `aesmd` pod works correctly, should there be a socket file in `/var/run/aesmd/` on host os?

yes. perhaps the webhook is not mutating correctly for you. can you `describe` the demo job and aesmd pods' output

astronaut0131, 2022-08-12:
> yes. perhaps the webhook is not mutating correctly for you. can you `describe` the demo job and aesmd pods' output
```
$ kubectl describe po intel-sgx-aesmd-9898b
Name:         intel-sgx-aesmd-9898b
Namespace:    default
Priority:     0
Node:         zhenhui-control-plane/172.21.0.3
Start Time:   Sat, 13 Aug 2022 00:46:35 +0800
Labels:       app=intel-sgx-aesmd
              controller-revision-hash=647857d559
              pod-template-generation=1
Annotations:  sgx.intel.com/epc: 1Mi
              sgx.intel.com/quote-provider: aesmd
Status:       Running
IP:           172.21.0.3
IPs:
  IP:           172.21.0.3
Controlled By:  DaemonSet/intel-sgx-aesmd
Containers:
  aesmd:
    Container ID:   containerd://c23a6802da868a755d42fd6523f9a139014eec997878e14c30d801a9989aba59
    Image:          intel/sgx-aesmd-demo:devel
    Image ID:       docker.io/library/import-2022-08-12@sha256:aeb55ce8810b059d1adb902ade638fe57a3682200dcde3fbdf5550057489cf79
    Port:           <none>
    Host Port:      <none>
    State:          Running
      Started:      Sat, 13 Aug 2022 00:47:10 +0800
    Ready:          True
    Restart Count:  0
    Limits:
      sgx.intel.com/enclave:    1
      sgx.intel.com/epc:        1Mi
      sgx.intel.com/provision:  1
    Requests:
      sgx.intel.com/enclave:    1
      sgx.intel.com/epc:        1Mi
      sgx.intel.com/provision:  1
    Environment:
      SGX_AESM_ADDR:  1
    Mounts:
      /etc/aesmd.conf from aesmdconf (rw,path="aesmd.conf")
      /etc/sgx_default_qcnl.conf from qplconf (rw,path="sgx_default_qcnl.conf")
      /var/run/aesmd from aesmd-socket (rw)
      /var/run/secrets/kubernetes.io/serviceaccount from kube-api-access-w6g6r (ro)
Conditions:
  Type              Status
  Initialized       True
  Ready             True
  ContainersReady   True
  PodScheduled      True
Volumes:
  aesmdconf:
    Type:      ConfigMap (a volume populated by a ConfigMap)
    Name:      sgx-attestation-conf
    Optional:  false
  qplconf:
    Type:      ConfigMap (a volume populated by a ConfigMap)
    Name:      sgx-attestation-conf
    Optional:  false
  kube-api-access-w6g6r:
    Type:                    Projected (a volume that contains injected data from multiple sources)
    TokenExpirationSeconds:  3607
    ConfigMapName:           kube-root-ca.crt
    ConfigMapOptional:       <nil>
    DownwardAPI:             true
  aesmd-socket:
    Type:          HostPath (bare host directory volume)
    Path:          /var/run/aesmd
    HostPathType:  DirectoryOrCreate
QoS Class:         BestEffort
[... 107 lines cut ...]
    TokenExpirationSeconds:  3607
    ConfigMapName:           kube-root-ca.crt
    ConfigMapOptional:       <nil>
    DownwardAPI:             true
QoS Class:                   BestEffort
Node-Selectors:              kubernetes.io/arch=amd64
Tolerations:                 node.kubernetes.io/disk-pressure:NoSchedule op=Exists
                             node.kubernetes.io/memory-pressure:NoSchedule op=Exists
                             node.kubernetes.io/not-ready:NoExecute op=Exists
                             node.kubernetes.io/pid-pressure:NoSchedule op=Exists
                             node.kubernetes.io/unreachable:NoExecute op=Exists
                             node.kubernetes.io/unschedulable:NoSchedule op=Exists
Events:
  Type    Reason     Age   From               Message
  ----    ------     ----  ----               -------
  Normal  Scheduled  18m   default-scheduler  Successfully assigned default/intel-sgx-plugin-dqr6j to zhenhui-control-plane
  Normal  Pulling    18m   kubelet            Pulling image "intel/intel-sgx-initcontainer:devel"
  Normal  Pulled     18m   kubelet            Successfully pulled image "intel/intel-sgx-initcontainer:devel" in 25.720694391s
  Normal  Created    18m   kubelet            Created container intel-sgx-initcontainer
  Normal  Started    18m   kubelet            Started container intel-sgx-initcontainer
  Normal  Pulling    18m   kubelet            Pulling image "intel/intel-sgx-plugin:devel"
  Normal  Pulled     17m   kubelet            Successfully pulled image "intel/intel-sgx-plugin:devel" in 31.173730203s
  Normal  Created    17m   kubelet            Created container intel-sgx-plugin
  Normal  Started    17m   kubelet            Started container intel-sgx-plugin
```

mythi, 2022-08-12:
@astronaut0131 everything seems correct. you could always try running `aesmd` on the host and undeploy that `aesmd` container.

astronaut0131, 2022-08-12:
> @astronaut0131 everything seems correct. you could always try running `aesmd` on the host and undeploy that `aesmd` container.

Using aesmd on host only gives:
```
$ kubectl logs ecdsa-quote-intelsgx-demo-job-w2gbd

Step1: Call sgx_qe_get_target_info:Error in sgx_qe_get_target_info. 0xe040
[init ../../../psw/ae/aesm_service/source/core/ipc/UnixCommunicationSocket.cpp:225] Failed to connect to socket /var/run/aesmd/aesm.socket
```
The socket actually appears on the host
```
$ ls /var/run/aesmd/
aesm.socket
```
Sample works correctly
```
$ SGX_AESM_ADDR=1 ./app

Step1: Call sgx_qe_get_target_info:succeed!
Step2: Call create_app_report:succeed!
Step3: Call sgx_qe_get_quote_size:succeed!
Step4: Call sgx_qe_get_quote:succeed!cert_key_type = 0x5
```
Shall I mount this socket onto the job demo pod or something?

mythi, 2022-08-12:
> Shall I mount this socket onto the job demo pod or something?

It is mounted:
```
    Mounts:
      /var/run/aesmd from aesmd-socket (rw)
...
  aesmd-socket:
    Type:          HostPath (bare host directory volume)
    Path:          /var/run/aesmd
    HostPathType:  DirectoryOrCreate
```

mythi, 2022-08-12:
> `Image:          intel/intel-sgx-plugin:devel`

can you try with version `0.24.0`?

astronaut0131, 2022-08-12:
> > `Image:          intel/intel-sgx-plugin:devel`
> 
> can you try with version `0.24.0`?

same error
```
$ kubectl describe po intel-sgx-plugin-hckfz
Name:         intel-sgx-plugin-hckfz
Namespace:    default
Priority:     0
Node:         zhenhui-control-plane/172.21.0.2
Start Time:   Sat, 13 Aug 2022 03:04:23 +0800
Labels:       app=intel-sgx-plugin
              controller-revision-hash=79f8f5cd9d
              pod-template-generation=1
Annotations:  <none>
Status:       Running
IP:           10.244.0.11
IPs:
  IP:           10.244.0.11
Controlled By:  DaemonSet/intel-sgx-plugin
Init Containers:
  intel-sgx-initcontainer:
    Container ID:   containerd://edace9707de247e6787c14edc28c098c4a1fcae1e9b46f3f669a004dfd9d0157
    Image:          intel/intel-sgx-initcontainer:0.24.0
    Image ID:       docker.io/intel/intel-sgx-initcontainer@sha256:da7f4ad1c2d0b9690744841a6b228ce33efe06087a8027c945e06353d5416fb9
    Port:           <none>
    Host Port:      <none>
    State:          Terminated
      Reason:       Completed
      Exit Code:    0
      Started:      Sat, 13 Aug 2022 03:04:57 +0800
      Finished:     Sat, 13 Aug 2022 03:04:57 +0800
    Ready:          True
    Restart Count:  0
    Environment:    <none>
    Mounts:
      /etc/kubernetes/node-feature-discovery/source.d/ from nfd-source-hooks (rw)
      /var/run/secrets/kubernetes.io/serviceaccount from kube-api-access-927lg (ro)
Containers:
  intel-sgx-plugin:
    Container ID:   containerd://078879155314cd6ca4a07c17ddfe2b10db1262ad98987df2fa51883555aaee25
    Image:          intel/intel-sgx-plugin:0.24.0
    Image ID:       docker.io/intel/intel-sgx-plugin@sha256:1fc6756cc3423ae05df07605cb30e2615725c2cb0192805d7da2bf67eb1b678c
    Port:           <none>
    Host Port:      <none>
    State:          Running
      Started:      Sat, 13 Aug 2022 03:05:18 +0800
    Ready:          True
    Restart Count:  0
    Environment:    <none>
    Mounts:
      /dev/sgx from sgxdevices (ro)
      /dev/sgx_enclave from sgx-enclave (ro)
      /dev/sgx_provision from sgx-provision (ro)
      /var/lib/kubelet/device-plugins from kubeletsockets (rw)
      /var/run/secrets/kubernetes.io/serviceaccount from kube-api-access-927lg (ro)
Conditions:
  Type              Status
  Initialized       True
  Ready             True
  ContainersReady   True
  PodScheduled      True
Volumes:
  nfd-source-hooks:
    Type:          HostPath (bare host directory volume)
    Path:          /etc/kubernetes/node-feature-discovery/source.d/
    HostPathType:  DirectoryOrCreate
  kubeletsockets:
    Type:          HostPath (bare host directory volume)
    Path:          /var/lib/kubelet/device-plugins
    HostPathType:
  sgxdevices:
    Type:          HostPath (bare host directory volume)
    Path:          /dev/sgx
    HostPathType:  DirectoryOrCreate
  sgx-enclave:
    Type:          HostPath (bare host directory volume)
    Path:          /dev/sgx_enclave
    HostPathType:  CharDevice
  sgx-provision:
    Type:          HostPath (bare host directory volume)
    Path:          /dev/sgx_provision
    HostPathType:  CharDevice
  kube-api-access-927lg:
    Type:                    Projected (a volume that contains injected data from multiple sources)
    TokenExpirationSeconds:  3607
    ConfigMapName:           kube-root-ca.crt
    ConfigMapOptional:       <nil>
    DownwardAPI:             true
QoS Class:                   BestEffort
Node-Selectors:              kubernetes.io/arch=amd64
Tolerations:                 node.kubernetes.io/disk-pressure:NoSchedule op=Exists
                             node.kubernetes.io/memory-pressure:NoSchedule op=Exists
                             node.kubernetes.io/not-ready:NoExecute op=Exists
                             node.kubernetes.io/pid-pressure:NoSchedule op=Exists
                             node.kubernetes.io/unreachable:NoExecute op=Exists
                             node.kubernetes.io/unschedulable:NoSchedule op=Exists
Events:
  Type     Reason       Age    From               Message
  ----     ------       ----   ----               -------
  Normal   Scheduled    2m44s  default-scheduler  Successfully assigned default/intel-sgx-plugin-hckfz to zhenhui-control-plane
  Warning  FailedMount  2m42s  kubelet            MountVolume.SetUp failed for volume "kube-api-access-927lg" : failed to sync configmap cache: timed out waiting for the condition
  Normal   Pulling      2m41s  kubelet            Pulling image "intel/intel-sgx-initcontainer:0.24.0"
  Normal   Pulled       2m10s  kubelet            Successfully pulled image "intel/intel-sgx-initcontainer:0.24.0" in 31.333289507s
  Normal   Created      2m10s  kubelet            Created container intel-sgx-initcontainer
  Normal   Started      2m10s  kubelet            Started container intel-sgx-initcontainer
  Normal   Pulling      2m9s   kubelet            Pulling image "intel/intel-sgx-plugin:0.24.0"
  Normal   Pulled       109s   kubelet            Successfully pulled image "intel/intel-sgx-plugin:0.24.0" in 19.330088346s
  Normal   Created      109s   kubelet            Created container intel-sgx-plugin
  Normal   Started      109s   kubelet            Started container intel-sgx-plugin
```
```
$ kubectl logs ecdsa-quote-intelsgx-demo-job-mjxl4
[init ../../../psw/ae/aesm_service/source/core/ipc/UnixCommunicationSocket.cpp:225] Failed to connect to socket /var/run/aesmd/aesm.socket

Step1: Call sgx_qe_get_target_info:Error in sgx_qe_get_target_info. 0xe040
```

astronaut0131, 2022-08-12:
Could this problem relate to linux permission?
```
ll /var/run/aesmd/
total 0
drwxr-xr-x  2 aesmd aesmd   60 8月  13 01:46 ./
drwxr-xr-x 41 root  root  1420 8月  13 02:57 ../
srwxrwxrwx  1 aesmd aesmd    0 8月  13 01:46 aesm.socket=
```

closed 2022-08-12 by astronaut0131

mythi, 2022-08-12:
@astronaut0131 good to hear it works! 

> but I ignored that user should mount volumn manually from host to kind node on bootstrap.

maybe share what you learned here so that if someone hits the same problem they will find it here

astronaut0131, 2022-08-12:
Sure, I've made a pull request to add deployment notes to avoid this problem.
https://github.com/intel/intel-device-plugins-for-kubernetes/pull/1099