intel/intel-device-plugins-for-kubernetes #1109: Intel arc support

atotalnoob, 2022-08-24:
Hi, 

Since the arc `a380`s are now available, if/when will arc be supported for kubernetes workloads? Does it follow the normal Intel discrete GPU setup?

uniemimu, 2022-08-24:
From the point of view of this project, a380 should already work with the latest Intel GPU-plugin. Workload containers can get the GPU device, if you add the GPU resource request to the workload POD specification.

eero-t, 2022-08-24:
@atotalnoob If your kernel supports / exposes given (Intel GPU) device to user-space, GPU plugin supports it.

Whether you can then use the GPU exposed to your k8s container workload by the GPU plugin, depends on the (Intel GPU) driver(s) in that container.

---

Intel provides pre-built driver packages: https://dgpu-docs.intel.com/installation-guides/index.html

I think kernel / i915 driver packages in those repos have more complete support than upstream kernel [1].

---

If you want to build drivers from upstream sources...

[1] In case of current upstream kernel, Intel dGPU support is not enabled by default quite yet, you need to use kernel "i915.force_probe=GPU-PCI-ID" probe option to enable it (PCI ID is listed e.g. in "lspci" output).  This is because things may not work optimally yet, due to some final parts listed here still missing from upstream: https://www.kernel.org/doc/html/latest/gpu/rfc/index.html

Upstream user-space drivers do have support for the new HW, but you need to check release notes for level of HW support included to given release (and whether it's enabled in given project release binaries):
* https://github.com/intel/media-driver/releases
* https://github.com/intel/compute-runtime/releases
* https://gitlab.freedesktop.org/mesa/mesa/-/blob/main/include/pci_ids/iris_pci_ids.h#L224

mythi, 2022-08-31:
labeled as a 'question' which also got answered.

closed 2022-08-31 by mythi