intel/intel-device-plugins-for-kubernetes #1158: k8s-at-home deprecation for helm chart

leopck, 2022-09-15:
There's a deprecation notice from k8s-at-home [https://github.com/k8s-at-home/charts/issues/1761](https://github.com/k8s-at-home/charts/issues/1761), one of the listed Charts there is Intel GPU Plugin [https://github.com/k8s-at-home/charts/tree/master/charts/stable/intel-gpu-plugin](https://github.com/k8s-at-home/charts/tree/master/charts/stable/intel-gpu-plugin)

Is there any plans for adopting this Charts here and re-hosting under Intel Org inside ArtifactHub? [https://artifacthub.io/packages/search?org=intel&sort=relevance&page=1](https://artifacthub.io/packages/search?org=intel&sort=relevance&page=1)

leopck, 2022-09-15:
I just noticed that this repo already uses github pages to publish some helm but I don't see the Intel GPU Plugin inside?

mythi, 2022-09-15:
We are moving our helm charts to intel/helm-charts. They will be operator based so the GPU one should be pretty easy to add if needed.

leopck, 2022-09-16:
Thank you @mythi ! Just noticed the PR: https://github.com/intel/helm-charts/pull/2. Looking forward to it!

closed 2022-09-16 by leopck