intel/intel-device-plugins-for-kubernetes #1173: GPU demo error - intel/intel-opencl-icd:devel not found

yockgen, 2022-09-28:
Trying to run GPU demo (./demo/intelgpu-job.yaml), however, encountering error as below:

```
Warning  Failed          23s (x2 over 39s)  kubelet            Failed to pull image "intel/intel-opencl-icd:devel": rpc error: code = Unknown desc = Error response from daemon: pull access denied for intel/intel-opencl-icd, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
  Warning  Failed          23s (x2 over 39s)  kubelet            Error: ErrImagePull
  Normal   BackOff         9s (x6 over 36s)   kubelet            Back-off pulling image "intel/intel-opencl-icd:devel"
  Warning  Failed          9s (x6 over 36s)   kubelet            Error: ImagePullBackOff
```

Trying to build the image locally:
intel-device-plugins-for-kubernetes/demo/intel-opencl-icd# docker build . -t yockgen/intel-opencl-icd

However, error as below:
```
root@KBL02:/data/intel-device-plugins-for-kubernetes/demo/intel-opencl-icd# docker build . -t yockgen/intel-opencl-icd
Sending build context to Docker daemon   2.56kB
Step 1/3 : FROM ubuntu:20.04
 ---> a0ce5a295b63
Step 2/3 : ARG APT="env DEBIAN_FRONTEND=noninteractive apt"
 ---> Using cache
 ---> 9f302e46f20a
Step 3/3 : RUN ${APT} update && ${APT} install -y curl gpg-agent     && echo 'deb [arch=amd64 signed-by=/usr/share/keyrings/intel-graphics.gpg] https://repositories.intel.com/graphics/ubuntu focal main' |        tee -a /etc/apt/sources.list.d/intel.list     && curl -s https://repositories.intel.com/graphics/intel-graphics.key |        gpg --dearmor --output /usr/share/keyrings/intel-graphics.gpg     && ${APT} update     && ${APT} install -y --no-install-recommends        intel-opencl-icd        clinfo     && ${APT} remove -y curl gpg-agent     && ${APT} autoremove -y
 ---> Running in fa3482307290

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Err:1 http://security.ubuntu.com/ubuntu focal-security InRelease
  Temporary failure resolving 'security.ubuntu.com'
Err:2 http://archive.ubuntu.com/ubuntu focal InRelease
  Temporary failure resolving 'archive.ubuntu.com'
Err:3 http://archive.ubuntu.com/ubuntu focal-updates InRelease
  Temporary failure resolving 'archive.ubuntu.com'
Err:4 http://archive.ubuntu.com/ubuntu focal-backports InRelease
  Temporary failure resolving 'archive.ubuntu.com'
Reading package lists...
Building dependency tree...
Reading state information...
All packages are up to date.
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/focal/InRelease  Temporary failure resolving 'archive.ubuntu.com'
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/focal-updates/InRelease  Temporary failure resolving 'archive.ubuntu.com'
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/focal-backports/InRelease  Temporary failure resolving 'archive.ubuntu.com'
W: Failed to fetch http://security.ubuntu.com/ubuntu/dists/focal-security/InRelease  Temporary failure resolving 'security.ubuntu.com'
W: Some index files failed to download. They have been ignored, or old ones used instead.

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package curl
E: Unable to locate package gpg-agent
The command '/bin/sh -c ${APT} update && ${APT} install -y curl gpg-agent     && echo 'deb [arch=amd64 signed-by=/usr/share/keyrings/intel-graphics.gpg] https://repositories.intel.com/graphics/ubuntu focal main' |        tee -a /etc/apt/sources.list.d/intel.list     && curl -s https://repositories.intel.com/graphics/intel-graphics.key |        gpg --dearmor --output /usr/share/keyrings/intel-graphics.gpg     && ${APT} update     && ${APT} install -y --no-install-recommends        intel-opencl-icd        clinfo     && ${APT} remove -y curl gpg-agent     && ${APT} autoremove -y' returned a non-zero code: 100
```

Please advise any other ways to run the GPU demo?

eero-t, 2022-09-28:
@yockgen If your internet access needs to go through proxy, you need to specify that proxy for Docker.  Typically it looks like this:
```
export http_proxy=<your proxy address>
docker build --build-arg http_proxy=$http_proxy ...
```
(Depending on docker file content, you may need to pass also https proxy address.)

tkatila, 2022-09-29:
Our documentation had a bug saying the intel-opencl-icd image is available from dockerhub. It isn't and has to be manually compiled. Build issue was fixed offline.

eero-t, 2022-09-29:
@tkatila I think this should be re-opened (I cannot).  There's no documentation saying that "intel/intel-opencl-icd:devel" image does not exist / that it's just a place holder for the registry URL of an image (that user must build & push to a local registry to which s/he has access).

@mythi While GPU plugin README mentions briefly needing to built the image, demo/ folder README does not say anything about a need to build things under it, or how to do it.

Neither README says anything about how to set proxy variables to get demo images builds to work, e.g. with "make <image name>".

Project demo image making does not have any support for passing proxy variables either:
* https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/Makefile#L220
* https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/demo/build-image.sh#L20

(At least it allows overriding container builder and defaults to docker, as that's most common one.)

tkatila, 2022-09-29:
Sure, we could improve the documentation so it underlines the need to build the image, push it to a known registry and changing the yaml according. But I wouldn't maybe start adding workarounds for various docker build issues (proxies, DNS etc.). We'd never get it to a perfect state.

But as discussed offline, we should to put effort into either pushing this opencl-icd image to dockerhub (or other registry) or finding a different demo image. As it is current, the demo isn't really spectacular. :)

eero-t, 2022-09-29:
The main points of the "demo" are to validate that device plugin and device drivers work, and act as an example in using given Intel device with k8s:
* verify: host kernel driver has enabled support for user's  Intel device
* verify: device plugin exposes device to container
* verify: container process can access the device (it's run as root, with same group as host device file, or user's containerd config enables option for automatically setting container's device file access rights according to container security context)
* example: How to install Intel's latest user-space device driver(s) to container (until distro drivers provide that support)
* verify: user-space drivers from Intel's public repository have enabled support for user's GPU
* info: what features underlying HW + drivers support (= output of `vainfo` / `vpl-inspect` / `clinfo` / etc)

IMHO anything further than that, belongs to some other repo.  Device plugins project wants as little responsibility for doing containers updates to fix some complex demo app drivers to latest versions, verify that demo still works on all relevant HW with that driver, do security updates for it etc.

Device plugin README of course can contain links to such projects.

mythi, 2022-09-29:
> Project demo image making does not have any support for passing proxy variables either:

It's not needed: [see docs](https://docs.docker.com/network/proxy/#configure-the-docker-client).

tkatila, 2022-09-29:
Re-opening for the documentation updates.

eero-t, 2022-09-29:
> It's not needed: [see docs](https://docs.docker.com/network/proxy/#configure-the-docker-client).

If you read the original bug description, it seems that documentation should at least include the link you provided. :-)

PR #1183 (2022-10-07): gpu: improve demo run instructions
Include the image building and pushing in the steps. Also notes about the need to modify the job yaml.

Closes: #1173
closed 2022-10-07 by mythi