intel/intel-device-plugins-for-kubernetes #1181: Move from NFD hook usage to NFD feature-file usage

uniemimu, 2022-10-05:
Intel device plugins for kubernetes, particularly GPU and SGX initcontainers, should align with the NFD hooks becoming deprecated. Ref https://github.com/kubernetes-sigs/node-feature-discovery/issues/855

NFD feature-files can be used instead of binary hooks. Potentially a sidecar container which produces the NFD feature-file would work. The possibility of cleaning up the feature-file at plugin POD deletion time should be investigated. The current initcontainer-based hook-approach leaves the hosts polluted even after the plugin is removed, since there is no cleanup currently.

eero-t, 2022-10-13:
> The current initcontainer-based hook-approach leaves the hosts polluted even after the plugin is removed, since there is no cleanup currently.

NFD hook will fail if there are no GPUs exposed by the kernel, so it will not provide labels to nfd-worker. nfd-worker always checks all sources before sending update to nfd-master, and that update replaces all earlier labels for given node. I.e. use of NFD hooks is not a labeling problem, it will "just" fill nfd-worker log with failure warnings, and is otherwise icky.

With feature files, you *do* need to remove the file [1] when GPU plugin goes down, in case all Intel GPUs get removed from node, and (another) GPU plugin instance won't be there to update the feature file. This is trivial though, as k8s sends termination signal to containers before forcefully terminating them, and you can specify in pod spec how long grace period you need for termination.

[1] I have ticket for NFD to automate feature files life-cycle handling (add expiration date): https://github.com/kubernetes-sigs/node-feature-discovery/issues/857

mythi, 2023-04-26:
dropped `sgx` label due to #1392

eero-t, 2023-06-22:
Hooks are now disabled by default in NFD upstream: https://github.com/kubernetes-sigs/node-feature-discovery/pull/1182

That change will be part of next (v0.14) NFD release.

PR #1492 (2023-09-12): gpu: copy nfdhook functionality to gpu-plugin
In order to support v0.14+ NFD and extended resources, write node labels into a file and store it for NFD's features.d directory.

The old functionality still exists in the initcontainer, but if it's used NFD has to be configured to allow it.

Fixes #1181
closed 2023-09-12 by uniemimu