intel/intel-device-plugins-for-kubernetes #1207: lint warnings

bart0sh, 2022-10-14:
There are 61 lint warnings in the [CI job output](https://github.com/intel/intel-device-plugins-for-kubernetes/actions/runs/3248409090)
60 warnings are about 'set-output' deprecation and one is about 'save-state'.
Here are some examples:
```
[lint](https://github.com/intel/intel-device-plugins-for-kubernetes/actions/runs/3248409090/jobs/5330874205#step:3:26)
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/

[lint](https://github.com/intel/intel-device-plugins-for-kubernetes/actions/runs/3248409090/jobs/5330874205#step:5:19)
The `save-state` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/

[Build image (intel-fpga-plugin, docker)](https://github.com/intel/intel-device-plugins-for-kubernetes/actions/runs/3248409090/jobs/5330875070#step:4:15)
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
```

Most of warnings come from image building, as the last example.

mythi, 2022-10-14:
@bart0sh hmm I can't find them. The blog post suggests to at least check the github runner version. What else could we do? Our actions are automatically updated.

mythi, 2023-01-12:
it looks they are gone now

closed 2023-01-12 by mythi