intel/intel-device-plugins-for-kubernetes #1252: e2e: request to change current approach to get logs

bart0sh, 2022-11-17:
e2e tests use this approach to get pod logs:
```
		ginkgo.By("waiting for the IAA demo to succeed")
		f.PodClient().WaitForSuccess(podName, 300*time.Second)

		ginkgo.By("getting workload log")
```

Which is almost useless as nobody is interested in getting logs of succeeded workloads.
We should change this in a way that even if workload fails we can see the logs.

mythi, 2022-11-17:
Agree. I have asked @hj-johannes-lee to check a better way to get logs while he works on getting #1188 working

hj-johannes-lee, 2022-11-22:
Yup!

hj-johannes-lee, 2023-01-24:
@bart0sh @mythi I want your opinions about this idea!
https://github.com/intel/intel-device-plugins-for-kubernetes/blob/3ec51928abd56ec78dbf1483f6d75e736e7d0b1e/test/e2e/dlb/dlb.go#L94-L108

PR #1470 (2023-08-14): e2e: make demo pod logged only when failed running
fixes: #1252