intel/intel-device-plugins-for-kubernetes #1269: Failed to open Intel SGX device

xxinran, 2022-12-12:
Hi, I am trying to run sgx demo in my local env, and it occurs `Failed to open Intel SGX device` error, the logs are the following:
```
$ k logs intelsgx-demo-job-t5qkh
[get_driver_type edmm_utility.cpp:111] Failed to open Intel SGX device.
[open_se_device edmm_utility.cpp:168] Failed to open Intel SGX device.
[get_driver_type /home/sgx/jenkins/ubuntuServer2004-release-build-trunk-218/build_target/PROD/label/Builder-UbuntuSrv20/label_exp/ubuntu64/linux-trunk-opensource/psw/urts/linux/edmm_utility.cpp:111] Failed to open Intel SGX device.
Info: Please make sure SGX module is enabled in the BIOS, and install SGX driver afterwards.
Error: Invalid SGX device.
Enter a character before exit ...
```

I have my sgx enabled platform, and I successfully run sgx quote generation on the host. 

I have checked the resource allocation of the node, and the output is :
```
$ k get  node -o json | jq .items[].status.allocatable
{
  "cpu": "8",
  "ephemeral-storage": "451743804693",
  "hugepages-1Gi": "0",
  "hugepages-2Mi": "0",
  "memory": "6997304Ki",
  "pods": "110",
  "sgx.intel.com/enclave": "110",
  "sgx.intel.com/epc": "197132288",
  "sgx.intel.com/provision": "110"
}
```
And I checked the existance of sgx device by
```
$ ll /dev/sgx
total 0
./
../
enclave -> ../sgx_enclave
provision -> ../sgx_provision
```

Can anyone help with this? Thanks in advance.

mythi, 2022-12-12:
@xxinran are you building from `main` branch or using some of the release images (and release branch)?

xxinran, 2022-12-12:
I am building from commit `c1422ea3401d65786b108373a33f5bf44f1f1233`

mythi, 2022-12-12:
can you try to build `sgx-sdk-demo` from main `HEAD`. The plugin images you don't need to build.

xxinran, 2022-12-13:
> can you try to build `sgx-sdk-demo` from main `HEAD`. The plugin images you don't need to build.

I have rebuild `sgx-sdk-demo` with `main`, it occurs the same error

xxinran, 2022-12-13:
I debugged in the pod and only found `sgx_enclave` under `/dev`. There is no `/dev/sgx` either '/dev/isgx'.

mythi, 2022-12-13:
> I debugged in the pod and only found `sgx_enclave` under `/dev`. There is no `/dev/sgx` either `/dev/isgx`.

This is expected and OK. One more question that it not clear from the original report: how do you deploy the demo app?

xxinran, 2022-12-13:
Thanks @mythi for helping, it is caused by a bug in sgx sdk https://github.com/intel/linux-sgx/issues/920

closed 2022-12-13 by xxinran