intel/intel-device-plugins-for-kubernetes #1293: CLI argument for dlb-dpdk-demo has been changed

vranar-int, 2023-01-02:
DPDK demo for DLB plugin is using dpdk-test-eventdev application to demonstrate and test if the device is working and is properly configured. The DLB device is passed into the application via `--vdev` parameter with an argument **dev_id** which value is an index to the device. If none is provided the argument defaults to `/dev/dlb0`.

The argument however was changed to **hwdev_id** (confirmed by DLB team) so the dev_id argument is ignored and the application always defaults to dlb0 even if the actual device may be, for example, dlb3 (likely case when deploying in containers).

Changing the parameter in `test.sh` for dlb-dpdk-demo should resolve the issue and the application opens the correct device.

hj-johannes-lee, 2023-01-18:
@vranar-int Hi. I am thinking about this matter, but could you let me know how you concluded that the change of argument would make other devices than dlb0 can be used?
If I understood the script correctly,
```bash
dlb_dev=$(ls /dev/dlb* | sed 's/\/dev\/dlb//' | head -1)
```
https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/demo/dlb-dpdk-demo/test.sh#L2

the script always uses the first device among listed devices.

hj-johannes-lee, 2023-01-18:
@vranar-int Thank you for the issue. I was confused about what you said. We use the first device among listed ones, and I thought you missed it.
I understood what you said and the pr #1307 will fix that!

PR #1307 (2023-01-18): fix dlb-dpdk-demo's argument 'dev_id' to 'hwdev_id'
Fixes: #1293
closed 2023-01-18 by bart0sh