intel/intel-device-plugins-for-kubernetes #1301: install accel-config from Debian

mythi, 2023-01-13:
from `debian:unstable-slim`:
```
# apt search accel-config
Sorting... Done
Full Text Search... Done
accel-config/unstable 3.5.0-2 amd64
  utility for configuring the DSA subsystem

accel-config-test/unstable 3.5.0-2 amd64
  utility to test the DSA subsystem

libaccel-config-dev/unstable 3.5.0-2 amd64
  development files for libaccel-config

libaccel-config1/unstable 3.5.0-2 amd64
  utility library wrapping the Intel DSA sysfs ABI
```

tkatila, 2023-06-09:
This happened for idxd initcontainers but accel-config-demo is still using accel-config from sources. Is the intent to also the demo to debian packages? We are modifying the accel-config content in the demo.

mythi, 2023-06-09:
That's the intent. We have issues submitted in the idxd to get rid of our patches

PR #1618 (2023-12-14): demo: update idxd-config to v4.1.4
Fixes: #1301
closed 2023-12-14 by tkatila