intel/intel-device-plugins-for-kubernetes #1303: Failed to deploy SGX device plugin

graziadonghia, 2023-01-14:
Hello, I'm trying to deploy Intel SGX device plugin on my Ubuntu 22.04 machine. Since my actual intention is to just make some tests I created the Kuberentes cluster with kind and everything is fine so far. However, when I try to run the [ECDSA Remote Attestation demo](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/v0.23.0/cmd/sgx_plugin/README.md#sgx-ecdsa-remote-attestation) the `aesmd` pod has `ImagePullBackOff` error.

This is the output I receive:
aesmd-demo image was compiled [here](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/v0.23.0/cmd/sgx_plugin/README.md#build-the-image)
````
~$ sudo docker images
REPOSITORY                            TAG                  IMAGE ID       CREATED             SIZE
intel/sgx-sdk-demo                    devel                d69a2a252b8c   About an hour ago   143MB
intel/sgx-aesmd-demo                  devel                d4f27b015268   About an hour ago   143MB
localhost:5000/intel/sgx-aesmd-demo   latest               d4f27b015268   About an hour ago   143MB
intel/intel-sgx-initcontainer         devel                de1924ff361d   About an hour ago   33.1MB
intel/intel-sgx-plugin                devel                83514db242de   About an hour ago   12.3MB
````

```
 $ k get po
NAME                                  READY   STATUS             RESTARTS   AGE
intel-sgx-aesmd-wlbpx                 0/1     ImagePullBackOff   0          5m55s
intel-sgx-plugin-pmh62                1/1     Running            0          70m
```

``` 
$ k describe po intel-sgx-aesmd-wlbpx
Events:
  Type     Reason     Age                     From               Message
  ----     ------     ----                    ----               -------
  Normal   Scheduled  7m58s                   default-scheduler  Successfully assigned default/intel-sgx-aesmd-wlbpx to test-control-plane
  Normal   Pulling    6m23s (x4 over 7m58s)   kubelet            Pulling image "intel/sgx-aesmd-demo:devel"
  Warning  Failed     6m22s (x4 over 7m57s)   kubelet            Failed to pull image "intel/sgx-aesmd-demo:devel": rpc error: code = Unknown desc = failed to pull and unpack image "docker.io/intel/sgx-aesmd-demo:devel": failed to resolve reference "docker.io/intel/sgx-aesmd-demo:devel": pull access denied, repository does not exist or may require authorization: server message: insufficient_scope: authorization failed
  Warning  Failed     6m22s (x4 over 7m57s)   kubelet            Error: ErrImagePull
  Warning  Failed     6m8s (x6 over 7m56s)    kubelet            Error: ImagePullBackOff
  Normal   BackOff    2m57s (x20 over 7m56s)  kubelet            Back-off pulling image "intel/sgx-aesmd-demo:devel"
```

What does it mean "repository does not exist or may require authorization"?

mythi, 2023-01-15:
we don't publish `intel/sgx-aesmd-demo:devel`. If you build this locally, it needs to be made available to your "localhost" registry and `kind` must be configured to use it. (the deployment might also need changing to point to your localhost registry.

mythi, 2023-02-01:
@graziadonghia do you have any outstanding opens or can we close?

closed 2023-02-17 by mythi