intel/intel-device-plugins-for-kubernetes #1304: Add QuoteVerification for SGX device plugin

graziadonghia, 2023-01-14:
Hello, I'm trying to do ECDSA Remote Attestation demo on my Kubernetes cluster. I see that the documentation stops at quote generation, do you think it is possible to do a similar demo for quote verification, so that ECDSA remote attestation process would be covered as a whole.

Another thing, is it possible to exploit this device plugin to build a system that periodically attests containers with the ecdsa pod?

mythi, 2023-01-15:
> do you think it is possible to do a similar demo for quote verification, so that ECDSA remote attestation process would be covered as a whole.

it would be possible but I never implemented that part and now it's a low-priority for me. 

>  this device plugin to build a system that periodically attests containers with the ecdsa pod?

The device plugin would not be the right place for this. I guess you could have a "cronjob" based app deployed in your cluster if you have a use case that needs to get regular quotes from services/apps.

mythi, 2023-02-01:
@graziadonghia do you have any outstanding opens or can we close?

PR #1513 (2023-09-04): k8s.io 1.28.1 update and SGX rework
Fixes: #1304 

The goal is to rework the deployments so that sgx* do not use `hostNetwork: true` to make IAS scanners happier.
closed 2023-09-04 by hj-johannes-lee