intel/intel-device-plugins-for-kubernetes #1305: Allow Disabling Image Name Validation

DWSR, 2023-01-14:
[This check](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/974c8e40bc24e3117f6bff35387c39823f9c2a7f/pkg/apis/deviceplugin/v1/webhook_common.go#L27) has the potential to break some workflows for mirroring images to private registries such as [kbld](https://carvel.dev/kbld).

It would be helpful to have an option to disable the image name validation since this it's only a sanity check and not a functional test.

mythi, 2023-01-15:
> potential to break some workflows for mirroring images to private registries such as [kbld](https://carvel.dev/kbld).

the registry / org are not checked by the validation. the image name validation checks the images matches  `<image-name>.Dockerfile` (or `make <image-name>`) from this repo.

DWSR, 2023-01-15:
@mythi Yes, exactly. Checking the name of the image isn't exactly particularly effective ([Rook](https://rook.io), by contrast, checks that the supplied Ceph image actually has Ceph installed and of a sufficient version), IMO.

I'm not asking for the default behaviour to change, but rather a way to turn off this validation with a CLI arg or environment variable (or both).

mythi, 2023-01-16:
We may need to revisit the validation also for other reasons (e.g., to allow users to use image sha's as versions) at some point. Is there anything that's blocking you right now?

DWSR, 2023-01-16:
@mythi Yes, currently this blocks me from air-gapping my cluster using my current workflows (via kbld and imgpkg).

Skaronator, 2025-09-21:
It would be great if both an image tag and its corresponding sha256 digest could be supported at the same time. That way we’d get the best of both worlds:

- a human-readable tag name, and
- a pinned version via the exact digest.

This is especially useful with Renovate, since it updates both the tag and the digest. That way, I can clearly see the version bump Renovate is applying while still ensuring the image is pinned.