intel/intel-device-plugins-for-kubernetes #1312: Using the GPU plugin - How to actually use the shared-dev-num option?

projx, 2023-01-20:
Hi

I've got the GPU plugin deployed using the NFD instructions,  but on the same page here:

- https://intel.github.io/intel-device-plugins-for-kubernetes/cmd/gpu_plugin/README.html

It mentions under the "Modes and Configuration Options" section, that the "shared-dev-num" can be used to allow this, can you provide actual details how this can be utilised, especially if using the supplied deployment instructions?

The only details I could find on this, were outline in this issue here:

- https://github.com/intel/intel-device-plugins-for-kubernetes/pull/88

Which state, the actual deamonset needs to be updated, adding in the following, to the Pod section:

```
          command: ["/usr/local/bin/intel_gpu_device_plugin"]
          args: ["-shared-dev-num", "5"]
```

I can confirm this works, but the information was difficult to find, and manually inserting this into a vendor supplied deployment-manifest is precarious at best.

Is there a better way of achieving? Such as setting a CRD value?

tkatila, 2023-01-20:
Hi projx,

In addition to the vanilla NFD assisted installation, we just added a NFD + monitoring + shared-dev-num deployment:
https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/gpu_plugin/README.md#install-to-nodes-with-nfd-monitoring-and-shared-dev

But the values are still hardcoded so you'd have to edit them manually or with kustomize + patch.

We also support an operator assisted deployment method. The basic instructions are here:
https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/operator/README.md#installation
NOTE: In the instructions, use "gpu" instead of "sgx" for the NFD's node-feature-rules.

With operator, GPU-plugin is deployed with a CRD and it's easier to customize the features.

Operator + GPU plugin can also be installed with our helm charts:
https://github.com/intel/helm-charts/tree/main/charts/device-plugin-operator
https://github.com/intel/helm-charts/tree/main/charts/gpu-device-plugin

Helm chart installation instructions are a bit lacking. I'll add something to our README.

tkatila, 2023-02-13:
@projx can we close this or is there something open from point of view?

closed 2023-02-17 by tkatila