intel/intel-device-plugins-for-kubernetes #1333: Race condition when installing using kubectl

Setomidor, 2023-02-16:
Using this command causes a race condition for the intel-dp-services object:

kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/sgx_plugin/overlays/epc-nfd/

Running it the first time:
```kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/sgx_plugin/overlays/epc-nfd/
namespace/intelsgxwebhook-system created
namespace/node-feature-discovery created
customresourcedefinition.apiextensions.k8s.io/nodefeaturerules.nfd.k8s-sigs.io created
customresourcedefinition.apiextensions.k8s.io/nodefeatures.nfd.k8s-sigs.io created
serviceaccount/nfd-master created
serviceaccount/nfd-worker created
role.rbac.authorization.k8s.io/nfd-worker created
clusterrole.rbac.authorization.k8s.io/nfd-master created
rolebinding.rbac.authorization.k8s.io/nfd-worker created
clusterrolebinding.rbac.authorization.k8s.io/nfd-master created
configmap/nfd-worker-conf created
service/intelsgxwebhook-webhook-service created
service/nfd-master created
deployment.apps/intelsgxwebhook-webhook created
deployment.apps/nfd-master created
daemonset.apps/nfd-worker created
daemonset.apps/intel-sgx-plugin created
certificate.cert-manager.io/intelsgxwebhook-serving-cert created
issuer.cert-manager.io/intelsgxwebhook-selfsigned-issuer created
mutatingwebhookconfiguration.admissionregistration.k8s.io/intelsgxwebhook-mutating-webhook-configuration created
error: resource mapping not found for name: "intel-dp-devices" namespace: "" from "https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/sgx_plugin/overlays/epc-nfd/": no matches for kind "NodeFeatureRule" in version "nfd.k8s-sigs.io/v1alpha1"
ensure CRDs are installed first
```

Running it again:

```kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/sgx_plugin/overlays/epc-nfd/
namespace/intelsgxwebhook-system unchanged
namespace/node-feature-discovery unchanged
customresourcedefinition.apiextensions.k8s.io/nodefeaturerules.nfd.k8s-sigs.io configured
customresourcedefinition.apiextensions.k8s.io/nodefeatures.nfd.k8s-sigs.io configured
serviceaccount/nfd-master unchanged
serviceaccount/nfd-worker unchanged
role.rbac.authorization.k8s.io/nfd-worker unchanged
clusterrole.rbac.authorization.k8s.io/nfd-master unchanged
rolebinding.rbac.authorization.k8s.io/nfd-worker unchanged
clusterrolebinding.rbac.authorization.k8s.io/nfd-master unchanged
configmap/nfd-worker-conf unchanged
service/intelsgxwebhook-webhook-service unchanged
service/nfd-master unchanged
deployment.apps/intelsgxwebhook-webhook unchanged
deployment.apps/nfd-master configured
daemonset.apps/nfd-worker unchanged
daemonset.apps/intel-sgx-plugin unchanged
certificate.cert-manager.io/intelsgxwebhook-serving-cert unchanged
issuer.cert-manager.io/intelsgxwebhook-selfsigned-issuer unchanged
nodefeaturerule.nfd.k8s-sigs.io/intel-dp-devices created
mutatingwebhookconfiguration.admissionregistration.k8s.io/intelsgxwebhook-mutating-webhook-configuration configured
```

mythi, 2023-02-16:
> Using this command causes a race condition for the intel-dp-services object:
> 
> kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/sgx_plugin/overlays/epc-nfd/
> 

Thanks. Yeah it's obvious.

mythi, 2023-02-23:
reopening to get it properly closed when #1340 gets merged.

Setomidor, 2023-02-23:
Sorry!

mythi, 2023-02-23:
@Setomidor no prob, you reported a valid issue so thank you!

PR #1340 (2023-03-07): more little fixes
Fixes: #1325
Fixes: #1333
Fixes: #1347
closed 2023-03-07 by tkatila