intel/intel-device-plugins-for-kubernetes #1341: ECDSA Quote example in 0.26.0 not getting a sgx_provision device

Setomidor, 2023-02-23:
Running a fresh cluster (single node kubeadm), installing the sgx components using the single-line kustomization:

`kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/sgx_plugin/overlays/epc-nfd/`

Installed aesmd, pods look fine:

```
intel-sgx-aesmd-czxk8                 1/1     Running       0             5m40s
intel-sgx-plugin-m98b6                1/1     Running       4 (47m ago)   2d23h
```

... but when running the ecdsa quote example, the resulting pod does not get an sgx_provision device assigned and fails.

```
kubectl apply -k 'https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/sgx_enclave_apps/overlays/sgx_ecdsa_aesmd_quote?ref=v0.26.0'
...
ecdsa-quote-intelsgx-demo-job-5b5gg   0/1     Error         0             5m
> kubectl logs ecdsa-quote-intelsgx-demo-job-5b5gg

Step1: Call sgx_qe_get_target_info:Error in sgx_qe_get_target_info. 0xe001
...
spec:
  containers:
  - command:
    - /opt/intel/sgx-quote-generation/sgx-quote-generation
    env:
    - name: SGX_AESM_ADDR
      value: "1"
    image: intel/sgx-sdk-demo:devel
    imagePullPolicy: IfNotPresent
    name: intelsgx-demo-job-1
    resources:
      limits:
        sgx.intel.com/enclave: "1"
        sgx.intel.com/epc: 512Ki
      requests:
        sgx.intel.com/enclave: "1"
        sgx.intel.com/epc: 512Ki
```
        
if I create the same container image with my own job-yaml and add the provisioning resource manually the example works:

```
apiVersion: batch/v1
kind: Job
metadata:
  name: own-sgx
  namespace: default
spec:
  backoffLimit: 6
  template:
    spec:
      containers:
      - command: ["/opt/intel/sgx-quote-generation/sgx-quote-generation"]
        image: intel/sgx-sdk-demo:devel
        imagePullPolicy: IfNotPresent
        name: own-sgx-1
        resources:
          limits:
            sgx.intel.com/epc: 512Ki
            sgx.intel.com/provision: 1
            sgx.intel.com/enclave: 1
        securityContext:
          allowPrivilegeEscalation: false
          readOnlyRootFilesystem: true
        workingDir: /opt/intel/sgx-quote-generation/
      dnsPolicy: ClusterFirst
      restartPolicy: Never
      schedulerName: default-scheduler
      securityContext: {}
      terminationGracePeriodSeconds: 30
  ...
own-sgx-b6bwz                         0/1     Completed     0             45s 
```

mythi, 2023-02-23:
> but when running the ecdsa quote example, the resulting pod does not get an sgx_provision device assigned and fails.

`overlays/sgx_ecdsa_aesmd_quote` gets quotes from `aesmd` so it's not supposed to get `sgx_provision` device. do you have `aesmd` correctly configure to talk to your PCCS?

Setomidor, 2023-02-28:
Thanks for the clarification -- there was indeed an issue with conflicting packages in the aesmd that was the issue.

For anyone who may find this in the future; the az-dcap-client package used in Azure is overshadowed by the libsgx-dcap-default-qpl installed by default in the images used in the demo. If you'd like to use the Azure one, the default-qpl package needs to be removed from the image or the Azure one will not take effect.

closed 2023-02-28 by Setomidor

mythi, 2023-02-28:
> If you'd like to use the Azure one, the default-qpl package needs to be removed from the image or the Azure one will not take effect.

Azure [provides their own `aesmd` package](https://learn.microsoft.com/en-us/azure/confidential-computing/confidential-nodes-aks-addon#psw-with-sgx-quote-helper) and that can be used when setting up an AKS cluster.