intel/intel-device-plugins-for-kubernetes #1342: monitoring_shared-dev_nfd fails to apply with "error: evalsymlink failure"

elapse2039, 2023-02-25:
Following the instructions to include monitoring overlay at the following documentation:

https://intel.github.io/intel-device-plugins-for-kubernetes/cmd/gpu_plugin/README.html#install-to-nodes-with-nfd-monitoring-and-shared-dev

Gives

```
kubectl apply -k 'https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/gpu_plugin/overlays/monitoring_shared-dev_nfd/?ref=v0.25.1'
error: evalsymlink failure on '/tmp/kustomize-579034736/deployments/gpu_plugin/overlays/monitoring_shared-dev_nfd' : lstat /tmp/kustomize-579034736/deployments/gpu_plugin/overlays/monitoring_shared-dev_nfd: no such file or directory
```

Other instructions to install device plugins and overlays work successfully.

Any advice would be very welcome on how to resolve this kustomize error.

Other questions:
1. Is there a Grafana dashboard that is tailored to showing Intel GPU metrics. (I've found plenty for NIVIDA.)
2. Is there a set of helm chart that I can use instead of these kustomize scripts?

Thx.

mythi, 2023-02-28:
cc @eero-t @uniemimu

eero-t, 2023-03-01:
> ```
> kubectl apply -k 'https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/gpu_plugin/overlays/monitoring_shared-dev_nfd/?ref=v0.25.1'
> error: evalsymlink failure on [...]

The documentation you were following is for the latest v0.26.0 release, and that overlay was added in latest release:
https://github.com/intel/intel-device-plugins-for-kubernetes/tree/v0.26.0/deployments/gpu_plugin/overlays/

There is no such overlay in previous v0.25.1 release:
https://github.com/intel/intel-device-plugins-for-kubernetes/tree/v0.25.1/deployments/gpu_plugin/overlays/

@mythi / @tkatila When release documentation is generated, would it make sense to replace all the "<RELEASE_VERSION>" strings with the corresponding release tag (and preferably also test those URLs working, with a kustomize dry-run)?

>     1. Is there a Grafana dashboard that is tailored to showing Intel GPU metrics. (I've found plenty for NIVIDA.)

Intel GPU metrics are provided by XPU manager: https://github.com/intel/xpumanager

And its k8s integration: https://github.com/intel/xpumanager/tree/master/deployment/kubernetes

NOTE: you need to enable GPU plugin monitoring resource for this.

I would also recommend reviewing its deployment and disable e.g. host mounts for features you're not interested (XPU manager has lot of features, intended for large cluster maintenance automation, which need all kind of accesses and privileges from the system).  [1]

> Following the instructions to include monitoring overlay at the following documentation:
> 
> https://intel.github.io/intel-device-plugins-for-kubernetes/cmd/gpu_plugin/README.html#install-to-nodes-with-nfd-monitoring-and-shared-dev
...
>     2. Is there a set of helm chart that I can use instead of these kustomize scripts?

See Helm docs in the repo itself: https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/INSTALL.md

I think the doc site content is generated only on releases, and improved Helm documentation was added after that.  @tkatila / @mythi?

----
[1] Another alternative is _collectd_ `gpu_sysman` plugin, which offers just GPU metrics: https://github.com/collectd/collectd/pull/3968

However, you would need to build _collectd_ with it from (v6 branch) sources: https://github.com/collectd/collectd/tree/collectd-6.0

With OneAPI Level-Zero frontend + backend libraries installed, either from from latest distro versions (= Debian Testing / Fedora 38+), or from Intel package repository:
https://dgpu-docs.intel.com/installation-guides/index.html

Using e.g. the following `configure` options:
`--disable-all-plugins --enable-write_prometheus --with-sysman=yes --enable-gpu_sysman`

Here's Grafana dashboard example supporting both collectd Sysman plugin and XPU manager: https://github.com/intel/fakedev-exporter/blob/main/grafana/gpu-metrics.json

(If you're interested on this option, I can help in getting it working.)

elapse2039, 2023-03-01:
Thanks so much for this detailed response.  I will re-read and digest and pick a path.  Thanks for the offer of help.
I believe I also tried applying 0.26.0 and got the same error but I didn't realise it was only available 0.26.0 originally.  

Do the 0.25, 0.26 intentionally map to k8s minor release numbers or is that just coincidence?

What is the relationship between XPU Manager and the Intel Device Plugin and its overlays?

I originally started down the route of Intel Device Plugins in k8s to expose GPUs to applications and I believe I have this successfully configured using the NFD labeling.  I did not come across any guides/docs/suggestions about XPU Manager.  Should I have just used that instead or is the XPU Manager a cluster application that takes advantage of the Intel Device Plugins?

eero-t, 2023-03-01:
> Do the 0.25, 0.26 intentionally map to k8s minor release numbers or is that just coincidence?

@mythi ?

> What is the relationship between XPU Manager and the Intel Device Plugin and its overlays?

XPUM (XPU Manager) project deployment file requests `i915_monitoring` k8s resource to get access to all Intel GPU devices on the node.

`i915_monitoring` resource is provided by GPU plugin, when its started with option for enabling that [1].   Otherwise these Intel projects are not related.

[1] Kustomize + overlay is just one way to add this option GPU plugin arguments.  Another is Intel devices operator, and its CRD option for monitoring.  Third one is the option in Helm chart (which I think to use the operator).

> I originally started down the route of Intel Device Plugins in k8s to expose GPUs to applications and I believe I have this successfully configured using the NFD labeling.  I did not come across any [...]

I'm not sure I understand your question.  NFD and XPUM are completely different things.

NFD is an official Kubernetes project, used to automatically add HW capability labels to all nodes, so that Pods using specific HW feature, can use corresponding label in their `nodeSelector`: https://kubernetes-sigs.github.io/node-feature-discovery/stable/get-started/introduction

(Nowadays NFD can also be configured to automatically taint nodes having specific HW capability, to avoid unrelated pods being scheduled on those same nodes.)

Whereas XPUM is Intel specific project used to get Intel GPU information[1], export GPU metrics to Prometheus (for things like Grafana GPU dashboards), and e.g. GPU management capabilities to Intel Data Center Manager [2]: https://www.intel.com/content/www/us/en/software/intel-dcm-technical-information.html

[1] There are some CLI tools included with XPUM container: https://github.com/intel/xpumanager/blob/master/doc/CLI_user_guide.md

Which you can run e.g. with: `kubectl -n NAMESPACE exec -it XPUM_POD -c xpumd -- xpumcli`

[2] After one enables REST API from the XPUM deployment file.  By default it exposes only Prometheus metrics API.

eero-t, 2023-03-01:
Regarding the things enabled in XPUM deployment: https://github.com/intel/xpumanager/blob/master/deployment/kubernetes/daemonset-intel-xpum.yaml

And how you could isolate the container better...

You can drop `host` networking as it's only relevant for GPU fabric _XeLink_ ports which are only in PVC HW (kernel provides _XeLink_ data through _netlink_ interface, which requires container to use `host` networking).

If you do not have device plugin for `smarter-devices` resources, you can drop also volumes related to PCIe reads/writes, and `SYS_RAWIO` capability.

If you are running a recent distro (kernel) and use _containerD_ or _CRI-O_ as container engine instead of _Docker_, you could try changing (very wide) `SYS_ADMIN` capability to newer and much more narrow `PERFMON` capability (no `SYS_` prefix).   It's a bit hard to debug issues with (new) capabilities not being supported by (old) container engines, so check first what metrics you get with `SYS_ADMIN`, and whether any (e.g. engine utilization) disappears when changing to  `PERFMON` capability. For reasons to do this change, see `man capabilities`.

mythi, 2023-03-02:
> > 3\. Is there a set of helm chart that I can use instead of these kustomize scripts?
> 
> See Helm docs in the repo itself: https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/INSTALL.md
> 
> I think the doc site content is generated only on releases, and improved Helm documentation was added after that. @tkatila / @mythi?

Actually, that link is for the docs generated from `main` after each PR. Docs from the releases have their own sub-folders (the the link from the top-level README)..

> > Do the 0.25, 0.26 intentionally map to k8s minor release numbers or is that just coincidence?

Before we make a release, we update our Go dependencies to the "matching" k8s Go modules. We always release our 0.xx after a new 1.xx k8s release.

tkatila, 2023-03-07:
@elapse2039 is there anything else we could help you with?

eero-t, 2023-03-07:
Added ticket for XPUM to support kustomization of its deployment privileges vs. features: https://github.com/intel/xpumanager/issues/32

closed 2023-03-07 by elapse2039