intel/intel-device-plugins-for-kubernetes #1344: SGX device not ready after device plugin installed.

Chen-Xintong, 2023-02-28:
I'm installing SGX device-plugin by this guide:https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/sgx_plugin/README.md#installation-using-the-operator

But after installed, I can get the sgx related resource by this 

```
kubectl describe node <node name> | grep sgx.intel.com

                    nfd.node.kubernetes.io/extended-resources: sgx.intel.com/epc
  sgx.intel.com/enclave:    110
  sgx.intel.com/epc:        98041856
  sgx.intel.com/provision:  110
  sgx.intel.com/enclave:    110
  sgx.intel.com/epc:        98041856
  sgx.intel.com/provision:  110
  sgx.intel.com/enclave    2           2
  sgx.intel.com/epc        2Mi         2Mi
  sgx.intel.com/provision  0           0
```

But when I start a pod and allocate resources, sgx still doesn't work since the intel SGX device cannot be opened, and the device is not ready yet: 

```
[get_driver_type /home/sgx/jenkins/ubuntuServer2204-release-build-trunk-218/build_target/PROD/label/Builder-UbuntuSrv2204/label_exp/ubuntu64/linux-trunk-opensource/psw/urts/linux/edmm_utility.cpp:111] Failed to open Intel SGX device.
Main C_Initialize(NULL) rv:CKR_DEVICE_ERROR
```

```
ll /dev/sgx_enclave
crw-rw-rw-1 root root 10, 125 Feb 28 09:01 /dev/sgx_enclave
```

Seems device plugin not assign the sgx device to the pod, how should I solve this problem??

Chen-Xintong, 2023-03-01:
> > Seems device plugin not assign the sgx device to the pod, how should I solve this problem??
> 
> is your build based on SGX SDK 2.18.1?

Is there any dependency between sgx device pulgin and sgxsdk? The sdk version on my host should be 2.14 or earlier. Is it possible to use the 1.26 version of the device plugin?

Chen-Xintong, 2023-03-01:
And I found an error log in device plugin manager:

`1 controller.go:329] "intel-device-plugins-manager: Reconciler error" err="Operation cannot be fulfilled on daemonsets.apps \"intel-sgx-plugin\": the object has been modified; please apply your changes to the latest version and try again" controller="sgxdeviceplugin" controllerGroup="deviceplugin.intel.com" controllerKind="SgxDevicePlugin" SgxDevicePlugin="sgxdeviceplugin-sample" namespace="" name="sgxdeviceplugin-sample" reconcileID=a23beec4-71f8-4f61-846f-9bddfc29bdd2`

mythi, 2023-03-01:
> Is there any dependency between sgx device pulgin and sgxsdk? The sdk version on my host should be 2.14 or earlier. Is it possible to use the 1.26 version of the device plugin?

The SGX setup does not need any SDK components on the host. We are talking about the version you have in your container you are getting the error from.

The latest device plugin versions do not make `/dev/sgx/*` symlinks available to containers (only `/dev/sgx_*`) device nodes. SGX SDK 2.18 has a bug that it only tries to open `/dev/sgx/enclave`  which is not available. Based on the error you copied it looks you might be using 2.18. The fix is to update to 2.18.1. (2.17 is also known to work).

mythi, 2023-03-01:
> And If I assign epc resource to any pod, there is a warning

your cluster does not have enough EPC available. From the log you copied it looks your node has only ~94Mb EPC.

Chen-Xintong, 2023-03-03:
> > And If I assign epc resource to any pod, there is a warning
> 
> your cluster does not have enough EPC available. From the log you copied it looks your node has only ~94Mb EPC.

Thanks, seems there is not enough EPC resources on my host, the node is assigned enclave and provision but can't grep epc:
````
kubectl describe node <> | grep sgx.intel.com
  sgx.intel.com/enclave:    110
  sgx.intel.com/provision:  110
  sgx.intel.com/enclave:    110
  sgx.intel.com/provision:  110
  sgx.intel.com/enclave    0           0
  sgx.intel.com/provision  0           0
````

mythi, 2023-03-03:
> the node is assigned enclave and provision but can't grep epc:

in the first message `epc` was available. what changed?

Chen-Xintong, 2023-03-03:
> > the node is assigned enclave and provision but can't grep epc:
> 
> in the first message `epc` was available. what changed?

I used device-plugin v0.26.0 in the first message, and successfully assigned epc to Node. Pod can run normally, but when I use SGX in pod, the `CKR_Device_ERROR `will occur.

Now, I am using the device-plugin v0.25.0, and I cannot get epc from the node. If I restore epc to any Pod, the Pod is always in the state of pending, and warning with `default-scheduler  0/1 nodes are available: 1 Insufficient sgx.intel.com/epc. preemption: 0/1 nodes are available: 1 No preemption victims found for incoming pod.`

But no matter which version Iuse, I think the root cause is that the epc resources are not enough.

mythi, 2023-03-03:
> Pod can run normally, but when I use SGX in pod, the `CKR_Device_ERROR` will occur.

the device plugin version has no effect in this. the container needs to use `/dev/sgx_enclave` to get rid of the error.

> `Insufficient sgx.intel.com/epc`

It's not clear to me if the resource is completely missing or your pod tries to request more than the node(s) can provide. `sgx.intel.com/epc` registration comes from Node Feature Discovery so that needs to be deployed as well.

Chen-Xintong, 2023-03-06:
Hi @mythi, I reinstalled the nfd/operator/device plugin again and it can works now😂, don't know why. But anyway, thank you very much for your reply.