intel/intel-device-plugins-for-kubernetes #1347: Certificate and Issuer missing from "Second Approach" SGX Webhook installation files?

Setomidor, 2023-03-03:
Looking at his page:

https://intel.github.io/intel-device-plugins-for-kubernetes/cmd/sgx_plugin/README.html

The "second approach" kustomization files installs a webhook deployment that reference the Secret 'webhook-server-cert', but this secret does not exist and is not created in this bundle:

```
$ kubectl kustomize https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/sgx_admissionwebhook/
```

Looking at the "first approach", "install everything" kustomization file on the same page a custom issuer and certificate are created that corresponds to this secret:

```
$ kubectl kustomize https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/sgx_plugin/overlays/epc-nfd/
...

apiVersion: cert-manager.io/v1
kind: Certificate
metadata:
  name: intelsgxwebhook-serving-cert
  namespace: intelsgxwebhook-system
spec:
  dnsNames:
  - intelsgxwebhook-webhook-service.intelsgxwebhook-system.svc
  - intelsgxwebhook-webhook-service.intelsgxwebhook-system.svc.cluster.local
  issuerRef:
    kind: Issuer
    name: intelsgxwebhook-selfsigned-issuer
  secretName: webhook-server-cert
---
apiVersion: cert-manager.io/v1
kind: Issuer
metadata:
  name: intelsgxwebhook-selfsigned-issuer
  namespace: intelsgxwebhook-system
spec:
  selfSigned: {}
```

mythi, 2023-03-03:
> The "second approach" kustomization files installs a webhook deployment that reference the Secret 'webhook-server-cert', but this secret does not exist and is not created in this bundle:
>
> ```
> $ [...]

Looks like a glitch in the doc. There's an overlay that is maintained for this purpose:

```
kubectl kustomize https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/sgx_admissionwebhook/overlays/default-with-certmanager
```

Setomidor, 2023-03-03:
Confirmed working, thank you!

PR #1340 (2023-03-07): more little fixes
Fixes: #1325
Fixes: #1333
Fixes: #1347
closed 2023-03-07 by tkatila